iiab/iiab #3544: Downloading Spanish Wikipedia ZIM puts an English description on your IIAB home page

holta, 2023-04-21:
A Spanish description of Spanish Wikipedia should probably be the default?

@tim-moody yesterday mentioned good ideas towards fixing this.

The root cause appears to be that both files below ( within /library/www/html/js-menu/menu-files/menu-defs/ )

- en-wikipedia_es_all_maxi.json
- es-wikipedia_es_all_maxi.json

...currently contain:

```
"lang": "es",
```

ASIDE: Fully testing this is somewhat challenging, as the ~26 GB http://download.kiwix.org/zim/wikipedia/wikipedia_es_all_maxi_2023-01.zim needs to be re-downloaded each time.

tim-moody, 2023-04-21:
to test could try removing menu def from menu in Adm Cons and click Update Home Menu

tim-moody, 2023-04-21:
N.B. https://github.com/iiab/iiab/wiki/IIAB-Menuing

- lang - 2 or 3 char code of the language of the menu item, not necessarily the target content

so lang should be en, not es in en-wikipedia_es_all_maxi.json

holta, 2023-04-21:
> to test could try removing menu def from menu in Adm Cons and click Update Home Menu

Very useful: similar to /usr/bin/iiab-update-menus, this is far faster than waiting for (and paying for) another 26G download! 👍

> so lang should be en, not es in en-wikipedia_es_all_maxi.json

Bad News: Changing `"lang": "es",` to `"lang": "en",` within `/library/www/html/js-menu/menu-files/menu-defs/en-wikipedia_es_all_maxi.json` does not help 🤔

The result is that the English description of Spanish Wikipedia keeps being added back (after I remove it from the IIAB home page).

Any other ideas?

tim-moody, 2023-04-21:
this will cause a lot of problems if people start removing things manually from json files.

What is the value of "menu_lang": "en" in /library/www/html/home/menu.json?

tim-moody, 2023-04-21:
https://github.com/iiab/iiab-admin-console/blob/0e59715d797df0e079b758334b79c78121d7f5eb/roles/common/templates/adm_lib.py#L594

name will be None if you remove it and I think cause another one to be generated

holta, 2023-04-21:
> What is the value of "menu_lang": "en" in /library/www/html/home/menu.json?

I tried both `"menu_lang": "es",` and `"menu_lang": "es",`

Either way, the English description continues to be auto-added to the IIAB home page, unfortunately 🤔

> name will be None if you remove it and I think cause another one to be generated

What is the purpose of the `name:` field?  Is it supposed to be identical to the JSON's filename (i.e. somewhat gratuitous duplication?)  It doesn't seem to be documented here:

https://github.com/iiab/iiab/wiki/IIAB-Menuing#menu-item-definition-fields

Certainly the `name:` field clearly has major consequences.  (Is one option to perhaps remove the `name:` field across the board, if it turns out these are causing trouble without much wider purpose?)

<br>

COMPLETE ASIDE: Is it currently possible to set /library/www/html/home/menu.json 's `menu_lang` in advance from within /etc/iiab/local_vars.yml — to help Spanish-centric (or any!) linguistic groups gets started on the right foot?

Of course almost everyone wants a bit of English/ETC, and/or including indigenous/local languages is very common too.

i.e. we really should try to soon clarify elegant ways to help everyone think through... how best to organize/display their favorite things across 2-3 languages!

tim-moody, 2023-04-21:
>COMPLETE ASIDE: Is it currently possible to set /library/www/html/home/menu.json 's menu_lang in advance from within /etc/iiab/local_vars.yml

I was thinking the same. If you install a preset you will get the proper lang code, I think.

We sort of stopped work on automating the menu when we could get one more or less menu item there. The nuances of the case where there is more than one candidate menu def were considered, but never really designed.

name is there because I don't like to derive data from file names; same for lang

holta, 2023-04-21:
I'll hold off on fixing the mainline/official version of en-wikipedia_es_all_maxi.json for now.

(So that this troublemaker continues to be a useful test case 🧪 !)

tim-moody, 2023-04-22:
After some thought, I think that menu update  should pick a menu def where the target = the permaref and the language of the menu def and of the zim are the same, rather than following some preferred language. If you can't read the menu item you probably can't read the zim either.

The question is what to do if such a menu def doesn't exist. I think the answer is generate one for the language in question rather than falling back to EN.

Secondarily, if autoupdate is False the code does nothing. I think if you are installing content it should continue to behave that way, but if you click the Update Home Menu it should honor the request regardless of that state of autoupdate.

holta, 2023-04-22:
> After some thought, I think that menu update should pick a menu def where the target = the permaref and the language of the menu def and of the zim are the same, rather than following some preferred language. If you can't read the menu item you probably can't read the zim either.

Seems reasonable to me, as multi-lingual IIAB's (with 2 or 3 languages) are essentially the norm.

Which suggests that the `menu_lang` variable in `/library/www/html/home/menu.json` is currently not really used?

> The question is what to do if such a menu def doesn't exist. I think the answer is generate one for the language in question rather than falling back to EN.

Seems reasonable.  Especially in `menu_lang` is effectively now deprecated ??

> Secondarily, if autoupdate is False the code does nothing. I think if you are installing content it should continue to behave that way, but if you click the Update Home Menu it should honor the request regardless of that state of autoupdate.

Sounds very reasonable!

holta, 2023-04-22:
Just FYI a Data Point... indicated the original "bug" is a bit more serious than I imagined:

- `es-wikipedia_es_all_maxi.json` seems to "never" appear when auto-updating your IIAB home page, when other JSON files refer to the same ZIM file 😢
- Earlier, I had naively (falsely!) assumed that `en-wikipedia_es_all_maxi.json` was predominating as a result of its filename being ahead of `es-...` in alphabetical order.
- That alphabetical order theory turns out to be entirely bogus, as confirmed when I moved `es-wikipedia_es_all_maxi.json` to `fr-wikipedia_es_all_maxi.json` and readjusted its internal fields to be "fr" — causing FRENCH to be auto-updated onto the IIAB home page 🤔

tim-moody, 2023-04-22:
>Especially in menu_lang is effectively now deprecated ??

we are only talking about zims here. there can still be other cases where we would like a default language.

is the name attribute still removed from es-wikipedia_es_all_maxi.json ?

holta, 2023-04-22:
> we are only talking about zims here. there can still be other cases where we would like a default language

Right.  I realized that later (e.g. most "IIAB Apps" like Calibre-Web, ETC) have many descriptions in many different languages.  And usually only 1 or 2 of those are desired on IIAB's home page, for any particular app.

While politically extremely uncomfortable (and dangerous in some countries) — ranking 2-or-3 languages in order of prominence (e.g. how near the top of IIAB home page, etc) — is is in effect what pedagogical design (a.k.a. curricular alignment) is all about.

_(As Tony Anderson explained very well to the OLPC community around 2010... this tension between 2-or-3 dominant languages is almost always... THE central learning design question...)_

> is the name attribute still removed from es-wikipedia_es_all_maxi.json ?

No.  That experiment was yesterday (i.e. a very separate but important finding: removing the `name` field from files like `en-wikipedia_es_all_maxi.json` and `fr-wikipedia_es_all_maxi.json` is the only way that I've found... so far anyway... to force auto-adding of Spanish to IIAB home page, FWIW!)

tim-moody, 2023-04-24:
A couple more wrinkles.

- If there is already a menu item in the menu that points to the perm_ref of the zim in question then do nothing.
- There can be multiple menu defs for a given perma_ref, where they have been created for different locales, such as es wikipedia with different home pages for Mexico or Peru. The preferred auto update is the closest to language+perma_ref as that should be the most generic.

holta, 2023-04-24:
> There can be multiple menu defs for a given perma_ref, where they have been created for different locales, such as es wikipedia with different home pages for Mexico or Peru. The preferred auto update is the closest to language+perma_ref as that should be the most generic.

Are you saying that be automated... approximately?
(e.g. occasionally getting the wrong locale/country/etc but close enough?)

tim-moody, 2023-04-24:
>Are you saying that be automated... approximately?
(e.g. occasionally getting the wrong locale/country/etc but close enough?)

no. I'm now thinking always use <zim lang>-<zim perma_ref> and if it doesn't exist create it. but the first point still holds that any item already in the menu with the right permaref stands and is sufficient.

holta, 2023-04-25:
Sounds exactly right.

It might seem off-topic but an associated painful/eternal reality is that... otherwise intelligent people (eternally!) struggle badly[1] with the idea that EduPacks & EduApps... *do not always come with a decent/minimal home page description/title/logo:*

Are there some specific areas where ChatGPT[2] could in future write... the needed/missing minimal stub IIAB home pages descriptions? ✍️

<br>

[1] Sometimes drowning in **Admin Console > Content Menus** adding a flurry of menu-defs... thinking each EduPack or EduApp will be downloaded/backfilled automatically... then getting very frustrated... never quite realizing they're supposed to start with **Admin Console > Install Content** 🧭

[2] It's claimed ChatGPT  & All are great at boiling down (summarizing) text... 🤖

tim-moody, 2023-04-25:
These menu defs all break the pattern of the being lang-permaref.
(first item is the name and second the  lang-permaref)

es-phet_html es-phet_es
pt-wikipedia_pt_medicine_maxi pt-wikipedia_pt_medicine
en-phet_zim en-phet_en
bh-wikipedia_bh_all_maxi bh-wikipedia_bho_all_maxi
en-wikihow_zim en-wikihow_en
en-wikipedia_es_all_maxi es-wikipedia_es_all_maxi
en-chemistry-stack-exchange en-chemistry_stackexchange_com_eng_all
en-wikipedia_en_for_schools en-wikipedia_en_for_schools_opt
fr-wikipedia_fr_all_haiti fr-wikipedia_fr_all
es-wikipedia_es_medicine_has_vids es-wikipedia_es_medicine_novid
en-wikipedia_kn_medicine kn-wikipedia_kn_medicine
hi-wikipedia_hi_medicine kn-wikipedia_hi_medicine
en-ray_charles_sample en-wikipedia_en_ray_charles
es-wikipedia_es_all_mx es-wikipedia_es_all
qu-wikipedia_es_all_maxi qu-wikipedia_qu_all_maxi
fr-wikistage_multi_all fr-wikistage_mul_all
en-wikipedia_chr_all chr-wikipedia_chr_all_novid
en-wikipedia_chr_all_maxi chr-wikipedia_chr_all__maxi
en-wikipedia_en_wp1-08 en-wikipedia_en_wp1-0_8
en-wikipedia_en_computer_maxi en-wikipedia_en_wikipedia_en_computer_maxi
en-ted_en_top-35-from-2017-2019 en-ted_en_playlist-curator-s-picks-top-10-ted-talks-of-2019
kok-wikipedia_gom_all_maxi gom-wikipedia_gom_all_maxi
es-stackoverflow_es_all es-es_stackoverflow_com_es_all
en-test_zim en-tes
en-raspberry-pi-stack-exchange en-raspberrypi_stackexchange_com_eng_all
ay-wikipedia_es_all_maxi ay-wikipedia_ay_all_maxi
en-wikistage_multi_all en-wikistage_mul_all
en-wikihow en-en-wikihow

qu-wikipedia_es_all_maxi qu-wikipedia_qu_all_maxi stands out as it should be the Quechua menu def for Quechua , but is the Quechua menu def for Spanish

Many reflect changes to Kiwix's naming conventions

tim-moody, 2023-04-26:
@holta the above list of non-canonical menu defs should be cleaned up, but even still they may cause problems for the rules for adding to home menu above.

holta, 2023-04-26:
Is some kind of mandatory auto-validation perhaps possible?

So everyone knows right away when filenames in `/library/www/html/js-menu/menu-files/menu-defs/` are structured incorrectly?

tim-moody, 2023-04-26:
the problem is only legacy. any new menu def created through adm cons will be canonical, but the current logic will automatically create new menu defs with different names if the old name is not canonical, which may require fixing up.

another problem is Kiwix changing names. phet_es becomes phet_es_all and both are in the catalog, so both can be installed, but only the old one has a menu def, so a new menu def will get created.

I'm wondering if we should fix all the legacy listed above.

tim-moody, 2023-04-26:
the zims phet_<lang> are now phet_<lang>_all

holta, 2023-04-26:
PhET ZIM filenames are suddenly getting longer, ah ok.

(With most languages having 3 files available for download at the moment, during Kiwix's transition, e.g. ...)

> [phet_en_2022-12.zim](https://download.kiwix.org/zim/phet/phet_en_2022-12.zim)      2022-12-03 13:13   65M   
[phet_en_2023-01.zim](https://download.kiwix.org/zim/phet/phet_en_2023-01.zim)      2023-01-01 02:01   66M   
[phet_en_all_2023-04.zim](https://download.kiwix.org/zim/phet/phet_en_all_2023-04.zim)  2023-04-02 15:37   68M

holta, 2023-04-26:
> I'm wondering if we should fix all the legacy listed above.

Maybe stash them in some well-documented place?

Is a README ideally allowed in this directory?

tim-moody, 2023-04-26:
let's not remove them until the zims they point to go away

holta, 2023-04-26:
@tim-moody's PR is tested/confirmed to solve the original problem:

- PR #iiab/iiab-admin-console#538

In short: the presence file of `/library/zims/content/wikipedia_es_all_maxi_2023-01.zim` leads to line `"es-wikipedia_es_all_maxi",` (great news, instead of line `"en-wikipedia_es_all_maxi",\` or French etc!) ...being added to /`library/www/html/home/menu.json`

tim-moody, 2023-04-26:
I'll manually fix the broken menudefs I've studies and we'll see where we are

holta, 2023-04-27:
Related improvements:

- https://github.com/iiab-share/js-menu-files/commit/680a5f427b6d482c07943103073c17a2f466d646

tim-moody, 2023-04-27:
Summary: https://docs.google.com/spreadsheets/d/1JE3j0TL_0iW-37fRSjKyOb8hq2sb94P_UX44TfuE_dU/edit?usp=sharing

holta, 2023-04-28:
3 news descriptions / menu-defs added:

- https://github.com/iiab-share/js-menu-files/pull/9/commits/85e5dbddf0625b8b958acf74c6be4be1c2aa7867

All together merged here:

- PR iiab-share/js-menu-files#9