iiab/iiab #3545: Multilingual Characters mangled, perhaps due to recent Kiwix catalog changes?

holta, 2023-04-21:
@tim-moody is on top of this.  Possibly arising from new Mac-like editing of character sets in the catalog itself?  So this ticket is just a place marker until workarounds and/or longer-term solutions emerge.

(Here's just one example of mangled characters in **Admin Console > Install Content > Get ZIM Files from Kiwix** :)

[image]

Related:

- iiab/iiab-admin-console#530

tim-moody, 2023-04-21:
see my comment on https://github.com/kiwix/k8s/issues/30

holta, 2023-04-22:
Looks like Emmanuel Engelhart found the root cause:

https://github.com/kiwix/libkiwix/issues/941

Interestingly the new file itself was fully correct — any idea why IIAB is acting like a browser[*] i.e. requiring http header *content-type: charset=utf-8* ?

[*] Rightly or wrongly 😄

holta, 2023-04-24:
Workaround installed:

- PR iiab/iiab-admin-console#537

holta, 2023-04-25:
Also now being solved upstream:

- PR kiwix/libkiwix#942

closed 2023-04-25 by holta