iiab/iiab #3547: 2 bits of bad news just for the record:

jvonau, 2023-04-24:
2 bits of bad news just for the record:

1. Latest screenshot from https://www.piwheels.org/project/cryptography/ indicates cryptography 40.0.2 (released 2023-04-14) fails on both Debian 10 (Buster) and 11 (Bullseye):

   [image]

2. "any wheels built with Rust will only work on Pi 2/3/4 not 1/Zero" for now... possibly 32-bit ARM issues might never be fully resolved in a stable way... as the world increasingly focuses on 64-bit:

   - https://github.com/piwheels/piwheels/pull/328#issuecomment-1518606329
   - https://github.com/piwheels/packages/issues/348

_Originally posted by @holta in https://github.com/iiab/iiab/issues/3459#issuecomment-1518810705_

holta, 2023-04-24:
> I suggest that an install be booted on a rpi-w and see if ansible runs.

Running Ansible of 32-bit Raspberry Pi Zero W might possibly be too much to ask for and not really needed.

In 2023 especially.  But in any case, here's a test, in case there's something to learn:

https://github.com/iiab/iiab/pull/3550#issuecomment-1520567117

holta, 2023-07-12:
Just FYI `cryptography` 41.0.2 was released yesterday and still fails on Bullseye / Python 3.9, according to https://www.piwheels.org/project/cryptography/ :

[image]

Background:

- #3459
- https://github.com/iiab/iiab/blob/975ce2f54e1fee82ba6412641b47a51ac078921d/scripts/ansible#L215-L217

holta, 2023-08-05:
Same story with `cryptography` 41.0.3 just released earlier this week:

[image]

holta, 2023-09-07:
Great news [piwheels](https://www.piwheels.org/project/cryptography/)  appears fixed across the board:

(Ideally piwheels might begin testing/publishing for imminBookworm Python 3.11 too?)

[image]

Related:

- https://github.com/iiab/iiab/issues/3526#issuecomment-1710649096

holta, 2023-09-08:
Ongoing [CFFI errors](https://github.com/iiab/iiab/issues/3526#issuecomment-1711559669) are serious (while testing PRs #3632 and #3634) but on the bright side piwheels-on-Bookworm efforts are clearly now moving forward!

- https://github.com/piwheels/packages/issues/388

holta, 2023-09-11:
This issues would appear to all be resolved, allowing this ticket to be closed soon in coming weeks, thanks to:

- PR #3632

closed 2025-04-06 by jvonau