iiab/iiab #3552: Test ansible-core 2.15 Release Candidate(s)

holta, 2023-04-25:
pip installing ansible-core 2.15 Release Candidate 1 is now easy:

https://pypi.org/project/ansible-core/2.15.0rc1/

https://github.com/ansible/ansible/releases/tag/v2.15.0rc1

Final release is expected in less than 3 weeks, on 2023-05-15.

holta, 2023-05-01:
After running `scripts/ansible` (or on any recent IIAB install) please run the following to test the latest:

```
/usr/local/ansible/bin/python3 -m pip install --upgrade ansible-core==2.15.0rc2
```

Details::

- https://pypi.org/project/ansible-core/2.15.0rc2/
- https://github.com/ansible/ansible/releases/tag/v2.15.0rc2

jvonau, 2023-05-31:
Since this issue mentions "Release Candidate(s)" and 2.15.0 has been released shouldn't this issue be closed now?

holta, 2023-06-01:
Not a bad idea.

But with ansible 2.15.0 having a serious flaw, maybe let's leave this open for just a bit longer, as a reminder that ansible-core 2.15.1 should be tested very soon when it's released likely in early June:

- #3588

And if absolutely necessary (if Ansible is unable to fix their regression) we can consider a workaround at that time.

jvonau, 2023-06-01:
I suggest editing the title to suggest testing 2.15.1 when that becomes available.

holta, 2023-10-18:
Good enough!

ansible-core 2.16 is imminent in early November, and testing looks good:

- #3661

closed 2023-10-18 by holta