iiab/iiab #3553: jupyterhub error

jvonau, 2023-04-26:
Noted jupyterhub error open an issue or I will.

_Originally posted by @jvonau in https://github.com/iiab/iiab/issues/3549#issuecomment-1523219241_

jvonau, 2023-04-26:
> psutil could not be installed from sources

Long story short is that this error is very similar to #3538 solved with #3503 with PR coming

holta, 2023-04-26:
Not sure if this applies to OS's beyond Ubuntu 23.04 and/or Python 3.11+ ?

In any case here's an example, excerpted from http://sprunge.us/r0BpIy?en :

> 2296 /opt/iiab/iiab/iiab-install.log-2023-04-25 20:12:41,271 p=5540 u=root n=ansible | TASK [jupyterhub : Break up jupyterhub/jupyterlab pip installs into 2 parts (3 packages above + 4 packages here) due to mutual dependency deadlock on some OS's] ***
2297 /opt/iiab/iiab/iiab-install.log-2023-04-25 20:13:28,310 p=5540 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/opt/iiab/jupyterhub/bin/pip3", "install", "--no-cache-dir", "jupyterlab", "jupyterhub_firstuseauthenticator", "jupyterhub-systemdspawner", "ipywidgets"], "msg": "stdout: Collecting jupyterlab\n  Downloading jupyterlab-3.6.3-py3-none-any.whl (8.9 MB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 8.9/8.9 MB 4.5 MB/s eta 0:00:00\nCollecting jupyterhub_firstuseauthenticator\n  Downloading jupyterhub_firstuseauthenticator-1.0.0-py3-none-any.whl (9.6 kB)\nCollecting jupyterhub-systemdspawner\n  Downloading jupyterhub_systemdspawner-0.17.0-py3-none-any.whl (14 kB)\nCollecting ipywidgets\n  Downloading ipywidgets-8.0.6-py3-none-any.whl (138 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 138.3/138.3 kB 6.4 MB/s eta 0:00:00\nCollecting ipython\n  Downloading ipython-8.12.0-py3-none-any.whl (796 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 796.4/796.4 kB 5.6 MB/s eta 0:00:00\nRequirement already satisfied: packaging in /opt/iiab/jupyterhub/lib/python3.11/site-packages (from jupyterlab) (23.1)\nRequirement already satisfied: tornado>=6.1.0 in /opt/iiab/jupyterhub/lib/python3.11/site-packages (from jupyterlab) (6.3.1)\nCollecting jupyter-core\n  Downloading jupyter_core-5.3.0-py3-none-any.whl (93 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 93.2/93.2 kB 6.1 MB/s eta 0:00:00\nCollecting jupyterlab-server~=2.19\n  Downloading jupyterlab_server-2.22.1-py3-none-any.whl (57 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 57.0/57.0 kB 5.9 MB/s eta 0:00:00\nCollecting jupyter-server<3,>=1.16.0\n  Downloading jupyter_server-2.5.0-py3-none-any.whl (366 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 366.8/366.8 kB 5.7 MB/s eta 0:00:00\nCollecting jupyter-ydoc~=0.2.3\n  Downloading jupyter_ydoc-0.2.4-py3-none-any.whl (5.9 kB)\nCollecting jupyter-server-ydoc~=0.8.0\n  Downloading jupyter_server_ydoc-0.8.0-py3-none-any.whl (11 kB)\nCollecting nbclassic\n  Downloading nbclassic-0.5.5-py3-none-any.whl (10.0 MB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 10.0/10.0 MB 5.3 MB/s eta 0:00:00\nCollecting notebook<7\n  Downloading notebook-6.5.4-py3-none-any.whl (529 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 529.8/529.8 kB 5.8 MB/s eta 0:00:00\nRequirement already satisfied: jinja2>=2.1 in /opt/iiab/jupyterhub/lib/python3.11/site-packages (from jupyterlab) (3.1.2)\nCollecting bcrypt\n  Downloading bcrypt-4.0.1-cp36-abi3-manylinux_2_28_aarch64.whl (583 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 583.6/583.6 kB 5.6 MB/s eta 0:00:00\nRequirement already satisfied: jupyterhub>=1.3 in /opt/iiab/jupyterhub/lib/python3.11/site-packages (from jupyterhub_firstuseauthenticator) (4.0.0)\nCollecting ipykernel>=4.5.1\n  Downloading ipykernel-6.22.0-py3-none-any.whl (149 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 150.0/150.0 kB 6.9 MB/s eta 0:00:00\nRequirement already satisfied: traitlets>=4.3.1 in /opt/iiab/jupyterhub/lib/python3.11/site-packages (from ipywidgets) (5.9.0)\nCollecting widgetsnbextension~=4.0.7\n  Downloading widgetsnbextension-4.0.7-py3-none-any.whl (2.1 MB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.1/2.1 MB 5.4 MB/s eta 0:00:00\nCollecting jupyterlab-widgets~=3.0.7\n  Downloading jupyterlab_widgets-3.0.7-py3-none-any.whl (198 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 198.2/198.2 kB 6.2 MB/s eta 0:00:00\nCollecting comm>=0.1.1\n  Downloading comm-0.1.3-py3-none-any.whl (6.6 kB)\nCollecting debugpy>=1.6.5\n  Downloading debugpy-1.6.7-py2.py3-none-any.whl (4.9 MB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.9/4.9 MB 5.3 MB/s eta 0:00:00\nCollecting jupyter-client>=6.1.12\n  Downloading jupyter_client-8.2.0-py3-none-any.whl (103 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 103.2/103.2 kB 6.2 MB/s eta 0:00:00\nCollecting matplotlib-inline>=0.1\n  Downloading matplotlib_inline-0.1.6-py3-none-any.whl (9.4 kB)\nCollecting nest-asyncio\n  Downloading nest_asyncio-1.5.6-py3-none-any.whl (5.2 kB)\nCollecting psutil\n  Downloading psutil-5.9.5.tar.gz (493 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 493.5/493.5 kB 5.6 MB/s eta 0:00:00\n  Installing build dependencies: started\n  Installing build dependencies: finished with status 'done'\n  Getting requirements to build wheel: started\n  Getting requirements to build wheel: finished with status 'done'\n  Preparing metadata (pyproject.toml): started\n  Preparing metadata (pyproject.toml): finished with status 'done'\nCollecting pyzmq>=20\n  Downloading pyzmq-25.0.2-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (865 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 865.6/865.6 kB 5.6 MB/s eta 0:00:00\nCollecting backcall\n  Downloading backcall-0.2.0-py2.py3-none-any.whl (11 kB)\nCollecting decorator\n  Downloading decorator-5.1.1-py3-none-any.whl (9.1 kB)\nCollecting jedi>=0.16\n  Downloading jedi-0.18.2-py2.py3-none-any.whl (1.6 MB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.6/1.6 MB 5.4 MB/s eta 0:00:00\nCollecting pickleshare\n  Downloading picklesh [... cut ...] or: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1\n      [end of output]\n  \n  note: This error originates from a subprocess, and is likely not a problem with pip.\n  ERROR: Failed building wheel for psutil\nERROR: Could not build wheels for psutil, which is required to install pyproject.toml-based projects\n"}
2298 /opt/iiab/iiab/iiab-install.log:2023-04-25 20:13:28,373 p=5540 u=root n=ansible | TASK [jupyterhub : SEE ERROR ABOVE (skip_role_on_error: False)] ****************

jvonau, 2023-04-26:
side note nodejs repo should be removed if the curl install fails to install.

jvonau, 2023-04-26:
https://pypi.org/project/psutil/#files

jvonau, 2023-04-26:
https://github.com/giampaolo/psutil/issues/2206
https://github.com/giampaolo/psutil/issues/2163

holta, 2023-04-27:
Resolved by:

- PR #3554

closed 2023-04-27 by holta

whomobile, 2024-04-18:
In case any comes here to fix "#include <Python.h>" not found issue with psutils
you will need to install correct python dev lib for your python. (error message is not wrong but not correct " sudo apt-get install gcc python3-dev" if you have more than one python verion.) 

e.g. use one of following
```
apt-get install -y python3.11-dev
apt-get install -y python3.10-dev
apt-get install -y python3.12-dev
apt-get install -y python3.13-dev
```