iiab/iiab #3556: Will FreePBX 17 support PHP 8 ?

holta, 2023-04-29:
Much speculation (a lot it dark) from October 2022 here:

- https://www.voip-info.org/forum/threads/the-future-of-freepbx.26709/
- https://community.freepbx.org/t/install-error/86309/7

Related:

- PR #3229
- PR #3508
- PR #3523

holta, 2023-08-11:
@EMG70

New gossip in recent weeks claims FreePBX 17 might actually happen, working with PHP 8.1+ — i.e. modern Linux OS's:

https://community.freepbx.org/t/freepbx-16-php-8-support/91444

holta, 2023-10-03:
@EMG70 FreePBX 17 progress is not public, if there is any:
https://github.com/FreePBX/core/tree/release/17.0
https://github.com/FreePBX/framework/tree/release/17.0

However just FYI Asterisk 21 Release Candidate 1 was posted 06-Sep-2023:
https://downloads.asterisk.org/pub/telephony/asterisk/
https://downloads.asterisk.org/pub/telephony/asterisk/asterisk-21.0.0-rc1.tar.gz

And explanations of Asterisk 21 are starting to appear...
https://www.asterisk.org/asterisk-21-module-removal/ (Asterisk 21 Module Removal)

holta, 2023-11-19:
FreePBX 17 pre-releases appear imminent based on these recent announcements:

- https://community.freepbx.org/t/freepbx-v17-update/92619
- https://community.freepbx.org/t/freepbx-v17-update/92474
- https://community.freepbx.org/t/freepbx-17-on-ubunutu-22/92649/5

> [Nenad Corbic](https://community.freepbx.org/u/ncorbic)
CTO Sangoma
>
> Hi All,
>
> FreePBX 17 based Debian 12 effort is making great progress.
We are getting close to releasing an ISO that one can just install.
>
> As part of move to Debian 12
>
> We again had to make a LOT all updates based on latest PHP 8 version.
Same as we did on FreePBX 16 with php 7
Scripts updates to support Debian
Repo updates to maintain sync with Debian versions
Build updates to create Asterisk, Wanpipe and other deb packages
More to come but a lot of progress is being made.
>
> Nenad

holta, 2023-11-19:
I'm not sure we should get involved just yet (maybe best to wait for Q1 2024?) but FYI pre-releases of Asterisk 21 and FreePBX 17 are both starting to circulate:

- 2023-10-18: https://downloads.asterisk.org/pub/telephony/asterisk/
- 2023-11-08: https://github.com/FreePBX/framework/commits/release/17.0

EMG70, 2023-11-19:
Thank you for the update.Hopefully coming soon.

holta, 2024-02-25:
Ugly hack but this appears to work on Debian 12 (PHP 8.2) and Ubuntu 24.04 (PHP 8.3):

- PR #3675

PS Raspberry Pi OS "Bookworm" (PHP 8.2) and Ubuntu 22.04 (PHP 8.1) probably work too, if @EMG70 has time to try?

closed 2024-02-25 by holta