iiab/iiab #3571: CUPS warning on Ubuntu 23.10: "Python crypt module is deprecated and will be removed from Python 3.13. Install the passlib library for continued encryption functionality"

holta, 2023-05-11:
> TASK [cups : CUPS web administration: Create Linux username 'Admin' with password 'changeme' in Linux group 'lpadmin' (shell: /usr/sbin/nologin, create_home: no)] ***
[DEPRECATION WARNING]: Encryption using the Python crypt module is deprecated.
The Python crypt module is deprecated and will be removed from Python 3.13.
Install the passlib library for continued encryption functionality. This feature will be removed in version 2.17. Deprecation warnings can be disabled by setting deprecation_warnings=False in ansible.cfg.

jvonau, 2023-05-15:
warning or error?

holta, 2023-05-15:
> warning or error?

Warning.

And it might possibly be solved upstream by Ansible, prior to the release of Python 3.13 (hopefully!)