iiab/iiab #3572: Lokole fails to install on Ubuntu 23.10: "General requirements to build wheel did not run successfully"

holta, 2023-05-11:
iiab-diagnostics: http://sprunge.us/6wHhhO?en

holta, 2023-05-11:
Short excerpt:

> error in mkwvconf setup command: 'python_requires' must be a string containing valid version specifiers; Invalid specifier: '>=2.7.*'\n [end of output]\n \n note: This error originates from a subprocess, and is likely not a problem with pip.\nerror: subprocess-exited-with-error\n\n× Getting requirements to build wheel did not run successfully.

Long excerpt:

> 1313 2023-05-10 20:22:59,706 p=2536 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/library/lokole/venv/bin/pip3", "install", "--no-cache-dir", "opwen_email_client"], "msg": "stdout: Collecting opwen_email_client\n  Downloading opwen_email_client-0.8.2.tar.gz (644 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 644.8/644.8 kB 26.1 MB/s eta 0:00:00\n  Installing build dependencies: started\n  Installing build dependencies: finished with status 'done'\n  Getting requirements to build wheel: started\n  Getting requirements to build wheel: finished with status 'done'\n  Preparing metadata (pyproject.toml): started\n  Preparing metadata (pyproject.toml): finished with status 'done'\nCollecting Babel==2.10.3 (from opwen_email_client)\n  Downloading Babel-2.10.3-py3-none-any.whl (9.5 MB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 9.5/9.5 MB 95.6 MB/s eta 0:00:00\nCollecting Flask-BabelEx==0.9.4 (from opwen_email_client)\n  Downloading Flask_BabelEx-0.9.4-py3-none-any.whl (8.7 kB)\nCollecting Flask-Caching==2.0.1 (from opwen_email_client)\n  Downloading Flask_Caching-2.0.1-py3-none-any.whl (28 kB)\nCollecting Flask-Cors==3.0.10 (from opwen_email_client)\n  Downloading Flask_Cors-3.0.10-py2.py3-none-any.whl (14 kB)\nCollecting Flask-Migrate==3.1.0 (from opwen_email_client)\n  Downloading Flask_Migrate-3.1.0-py3-none-any.whl (20 kB)\nCollecting Flask-SQLAlchemy==2.5.1 (from opwen_email_client)\n  Downloading Flask_SQLAlchemy-2.5.1-py2.py3-none-any.whl (17 kB)\nCollecting Flask-Script==2.0.6 (from opwen_email_client)\n  Downloading Flask-Script-2.0.6.tar.gz (43 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 43.1/43.1 kB 109.5 MB/s eta 0:00:00\n  Installing build dependencies: started\n  Installing build dependencies: finished with status 'done'\n  Getting requirements to build wheel: started\n  Getting requirements to build wheel: finished with status 'done'\n  Preparing metadata (pyproject.toml): started\n  Preparing metadata (pyproject.toml): finished with status 'done'\nCollecting Flask-Security==3.0.0 (from opwen_email_client)\n  Downloading Flask_Security-3.0.0-py2.py3-none-any.whl (68 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 68.9/68.9 kB 240.7 MB/s eta 0:00:00\nCollecting Flask-Testing==0.8.1 (from opwen_email_client)\n  Downloading Flask-Testing-0.8.1.tar.gz (45 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 45.2/45.2 kB 143.7 MB/s eta 0:00:00\n  Installing build dependencies: started\n  Installing build dependencies: finished with status 'done'\n  Getting requirements to build wheel: started\n  Getting requirements to build wheel: finished with status 'done'\n  Preparing metadata (pyproject.toml): started\n  Preparing metadata (pyproject.toml): finished with status 'done'\nCollecting Flask-WTF==1.0.1 (from opwen_email_client)\n  Downloading Flask_WTF-1.0.1-py3-none-any.whl (12 kB)\nCollecting Flask==2.2.1 (from opwen_email_client)\n  Downloading Flask-2.2.1-py3-none-any.whl (101 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 101.5/101.5 kB 263.6 MB/s eta 0:00:00\nCollecting SQLAlchemy==1.4.39 (from opwen_email_client)\n  Downloading SQLAlchemy-1.4.39.tar.gz (8.2 MB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 8.2/8.2 MB 26.2 MB/s eta 0:00:00\n  Installing build dependencies: started\n  Installing build dependencies: finished with status 'done'\n  Getting requirements to build wheel: started\n  Getting requirements to build wheel: finished with status 'done'\n  Preparing metadata (pyproject.toml): started\n  Preparing metadata (pyproject.toml): finished with status 'done'\nCollecting WTForms==3.0.1 (from opwen_email_client)\n  Downloading WTForms-3.0.1-py3-none-any.whl (136 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 136.5/136.5 kB 99.8 MB/s eta 0:00:00\nCollecting email-validator==1.2.1 (from opwen_email_client)\n  Downloading email_validator-1.2.1-py2.py3-none-any.whl (22 kB)\nCollecting Werkzeug==2.2.1 (from opwen_email_client)\n  Downloading Werkzeug-2.2.1-py3-none-any.whl (232 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 232.4/232.4 kB 161.0 MB/s eta 0:00:00\nCollecting fasteners==0.17.3 (from opwen_email_client)\n  Downloading fasteners-0.17.3-py3-none-any.whl (18 kB)\nCollecting apache-libcloud==3.6.0 (from opwen_email_client)\n  Downloading apache_libcloud-3.6.0-py2.py3-none-any.whl (4.0 MB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.0/4.0 MB 49.5 MB/s eta 0:00:00\nCollecting bcrypt==3.2.2 (from opwen_email_client)\n  Downloading bcrypt-3.2.2-cp36-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.manylinux_2_24_x86_64.whl (62 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.3/62.3 kB 304.8 MB/s eta 0:00:00\nCollecting beautifulsoup4==4.11.1 (from opwen_email_client)\n  Downloading beautifulsoup4-4.11.1-py3-none-any.whl (128 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 128.2/128.2 kB 349.7 MB/s eta 0:00:00\nCollecting cached-property==1.5.2 (from opwen_email_client)\n  Downloading cached_property-1.5.2-py2.py3-none-any.whl (7.6 kB)\nCollecting celery[sqlalchemy]==5.2.7 (from opwen_email_client)\n  Downloading celery-5.2.7-py3-none-any.whl (405 kB) [... cut ...] tput]\n      error in mkwvconf setup command: 'python_requires' must be a string containing valid version specifiers; Invalid specifier: '>=2.7.*'\n      [end of output]\n  \n  note: This error originates from a subprocess, and is likely not a problem with pip.\nerror: subprocess-exited-with-error\n\n× Getting requirements to build wheel did not run successfully.\n│ exit code: 1\n╰─> See above for output.\n\nnote: This error originates from a subprocess, and is likely not a problem with pip.\n"}

holta, 2023-05-11:
FWIW

```
root@box:~# grep -rn python_requires /library/lokole/ | grep mkwv
grep: /library/lokole/venv/lib/python3.11/site-packages/setuptools/
config/__pycache__/setupcfg.cpython-311.pyc: binary file matches
grep: /library/lokole/venv/lib/python3.11/site-packages/setuptools/
config/__pycache__/pyprojecttoml.cpython-311.pyc: binary file match
es
grep: /library/lokole/venv/lib/python3.11/site-packages/setuptools/
config/__pycache__/_apply_pyprojecttoml.cpython-311.pyc: binary fil
e matches
grep: /library/lokole/venv/lib/python3.11/site-packages/setuptools/
__pycache__/dist.cpython-311.pyc: binary file matches

jvonau, 2023-05-15:
Lack of available wheel for mkwvconf

holta, 2023-09-06:
LARGE-sized IIAB installs now avoid installing Lokole for this reason:

- PR #3630