iiab/iiab #3573: KA Lite fails to install on Ubuntu 23.10: "failed to find interpreter for Builtin discover of python_spec='python2.7'"

holta, 2023-05-11:
> TASK [kalite : Use pip to pin setuptools to 44 in /usr/local/kalite/venv] ******
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/usr/bin/virtualenv", "-ppython2.7", "/usr/local/kalite/venv"], "msg": "stdout: RuntimeError: failed to find interpreter for Builtin discover of python_spec='python2.7'\n"}

iiab-diagnostics: http://sprunge.us/6wHhhO?en

Related:

- #3528

holta, 2023-05-14:
A bit more detail here: (on another Ubuntu 23.10 attempt)

> 2023-05-13 10:52:36,470 p=44727 u=root n=ansible | TASK [kalite : Use pip to pin setuptools to 44 in /usr/local/kalite/venv] ******
2023-05-13 10:52:37,464 p=44727 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/usr/bin/virtualenv", "-ppython2.7", "/usr/local/kalite/venv"], "msg": "stdout: RuntimeError: failed to find interpreter for B\
uiltin discover of python_spec='python2.7'\n"}

holta, 2023-05-14:
> root@box:~# virtualenv -p python2.7 /usr/local/kalite/venv
RuntimeError: failed to find interpreter for Builtin discover of python_spec='python2.7'

FWIW specifying the full path `/usr/bin/python2.7` certainly doesn't help! 😒 

> root@box:~# virtualenv -p /usr/bin/python2.7 /usr/local/kalite/venv
RuntimeError: failed to query /usr/bin/python2.7 with code 1 err: '  File "/usr/lib/python3/dist-packages/virtualenv/discovery/py_info.py", line 152\n    os.path.join(base_dir, exe) for exe in (f"python{major}", f"python{major}.{minor}")\n                                                           ^\nSyntaxError: invalid syntax\n'

(In any case, explicitly specifying the path to `python2.7` should not be necessary...)

```
root@box:~# which python2.7
/usr/bin/python2.7

root@box:~# echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
```

tim-moody, 2023-05-14:
could you be using a python 3 virtualenv to start a 2.7 venv? i.e. do you need a python 2 virtual env executable?

holta, 2023-05-14:
> could you be using a python 3 virtualenv to start a 2.7 venv? i.e. do you need a python 2 virtual env executable?

Something like that sounds incredibly likely.

(Strange that the problem does not occur on Debian 12 and Ubuntu 23.04 — something clearly changed "upstream" very recently, right after Ubuntu 23.04 was released on April 20th...)

holta, 2023-05-14:
> (Strange that the problem does not occur on Debian 12 and Ubuntu 23.04 — something clearly changed "upstream" very recently, right after Ubuntu 23.04 was released on April 20th...)

I'd be very surprised if the jump from Python 3.11.2 (in Debian 12 and Ubuntu 23.04) to Python 3.11.3 (in Ubuntu 23.10) is causing this serious problem with `virtualenv` ?

(But I might very well be wrong...)

holta, 2023-05-14:
If @jvonau has time in the coming week, maybe he can take a look?

(Any chance this upstream change... effectively blocking Python 2... might be intentional?)

jvonau, 2023-05-14:
> I'd be very surprised if the jump from Python 3.11.2 (in Debian 12 and Ubuntu 23.04) to Python 3.11.3 (in Ubuntu 23.10) is causing this serious problem with `virtualenv` ?
>
> (But I might very well be [...]

lets start with `apt list python2*`

holta, 2023-05-15:
> lets start with `apt list python2*`

```
root@box:~# apt list python2*
Listing... Done
python2-minimal/now 2.7.18-3 amd64 [installed,local]
python2.7-minimal/now 2.7.18-13ubuntu1.1 amd64 [installed,local]
python2.7/now 2.7.18-13ubuntu1.1 amd64 [installed,local]
python2/now 2.7.18-3 amd64 [installed,local]

root@box:~# apt list virtualenv
Listing... Done
virtualenv/mantic,now 20.23.0+ds-1 all [installed]

root@box:~# which virtualenv
/usr/bin/virtualenv

root@box:~# virtualenv --version
virtualenv 20.23.0+ds from /usr/lib/python3/dist-packages/virtualenv/__init__.py
```

Latest iiab-diagnostics: http://sprunge.us/P5tNcW?en

jvonau, 2023-05-15:
There should be 6 apt packages installed from my notes at https://github.com/iiab/iiab/blob/master/scripts/install_python2.sh#L8-L14. How about `apt list *python2*`

Side note with ansible https://github.com/ansible/ansible/issues/77604

holta, 2023-05-15:
> There should be 6 apt packages installed from my notes at https://github.com/iiab/iiab/blob/master/scripts/install_python2.sh#L8-L14. How about `apt list *python2*`

These 7 are certainly present:

```
root@box:~# apt list *python2*
Listing... Done
libpython2-stdlib/now 2.7.18-3 amd64 [installed,local]
libpython2.7-minimal/now 2.7.18-13ubuntu1.1 amd64 [installed,local]
libpython2.7-stdlib/now 2.7.18-13ubuntu1.1 amd64 [installed,local]
python2-minimal/now 2.7.18-3 amd64 [installed,local]
python2.7-minimal/now 2.7.18-13ubuntu1.1 amd64 [installed,local]
python2.7/now 2.7.18-13ubuntu1.1 amd64 [installed,local]
python2/now 2.7.18-3 amd64 [installed,local]
```

jvonau, 2023-05-17:
Next would be the version of virtualenv installed on the non-working/working environments and the change log from the U23.10 version.

holta, 2023-05-17:
Ubuntu 23.04 (what works...)

```
root@box:~# virtualenv --version
virtualenv 20.19.0+ds from /usr/lib/python3/dist-packages/virtualenv/__init__.py
```

Ubuntu 23.10 pre-releases (don't work...)

```
root@box:~# virtualenv --version
virtualenv 20.23.0+ds from /usr/lib/python3/dist-packages/virtualenv/__init__.py
```

```
root@box:~# apt changelog virtualenv

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Get:1 https://changelogs.ubuntu.com python-virtualenv 20.23.0+ds-1 Changelog [35.1 kB]
python-virtualenv (20.23.0+ds-1) experimental; urgency=medium

  * New upstream release.

 -- Stefano Rivera <stefanor@debian.org>  Thu, 04 May 2023 15:36:09 -0400

python-virtualenv (20.21.0+ds-1) experimental; urgency=medium

  * New upstream release.
  * Refresh patches.

 -- Stefano Rivera <stefanor@debian.org>  Sat, 25 Mar 2023 21:12:22 -0400

python-virtualenv (20.19.0+ds-1) experimental; urgency=medium

  * New upstream release.
  * Refresh patches.
  * Bump Standards-Version to 4.6.2, no changes needed.
  * Bump copyright years.
  * Upstream migrated to hatchling, update Build-Depends.
  * Upstream stopped bundling docs in the source distribution, drop them.

 -- Stefano Rivera <stefanor@debian.org>  Sun, 12 Feb 2023 15:42:42 -0400

[...]
```

The more detailed page https://virtualenv.pypa.io/en/latest/changelog.html#v20-22-0-2023-04-19 might explain this:

> v20.22.0 (2023-04-19)
Features - 20.22.0
Drop support for creating Python <=3.6 (including 2) interpreters. Removed pip of 20.3.4, 21.3.1; wheel of 0.37.1; setuptools of 59.6.0, 44.1.1, 50.3.2- by @gaborbernat. ([#](https://virtualenv.pypa.io/en/latest/changelog.html#v20-22-0-2023-04-19)2548)

- https://github.com/pypa/virtualenv/issues/2548

( Should we try to install an older version of `virtualenv` like 20.21.1 for another year or so — for those unable to run memory-intensive Kolibri ?)

jvonau, 2023-05-18:
That might be viable option, if you can find the older version and stash the deb package on d.iiab.org. Be on the watch for any updates to virtualenv that might be introduces on the currently working platforms.

holta, 2023-05-19:
> stash the deb package

I could tried that and could not get it to work.  Details in the PR below.

In short, @neomatrixcode suggested using pip instead, which indeed appears to work: (with all the obvious risks that pip brings!)

- PR #3582

holta, 2023-05-22:
Solved / mitigated by:

- PR #3582 
- PR #3587

closed 2023-05-22 by holta