iiab/iiab #3574: Fresh Install. Home page links not working

stebbo63, 2023-05-13:
Hello,
I am completely new to iiab.  please excuse any errors on my part.

I have installed iiab on a new raspberry pi.  I selected "m" during the initial install process.  I then went to "Install Content" and selected "Medical Content for English Speakers".

Both the initial install and the install content seemed to go well.

However, many of the home screen menu options are not working.  I keep getting "502 Bad Gateway" on many of the options.

For example, the following options (not a complete list) produce a 502 Bad Gateway

- MDWiki Medical Encyclopedia with Videos
- StatsPearls Article Library
- NIH Genetic and Rare Diseases
and more.

Some menu options do work, for example

- The Global Emergency Medicine Wiki
- Healthphone Medical Video Collection in English

This is the URL for the MDWiki option which doesn't work 
http://box.lan/kiwix/viewer#mdwiki_en_all_2023-03/

Any assistance would be greatly appreciated.

Regards,
Chris.

[stebbo.txt](https://github.com/iiab/iiab/files/11469261/stebbo.txt)

holta, 2023-05-13:
Thanks @stebbo63 for the report.

We'll look into it as soon as possible in coming days trying to reproduce what you're describing.

> the install content seemed to go well.

Have enough hours gone by for the content downloads to complete?

Is there any indication if you look around **Admin Console (http://box.lan/admin) > Utilities > Display Job Status** ?

tim-moody, 2023-05-13:
What happens if you go to url http://box.lan/kiwix/ Let's make sure kiwix is installed and working.

tim-moody, 2023-05-13:
just noticed your stebbo.txt above which is iiab-diagnostics and shows kiwix running, but still good to check.

holta, 2023-05-13:
Indeed /etc/iiab/iiab_state.yml shows:

`kiwix_installed: True`

Just FYI kiwix-tools 3.5.0 is very new as of recent days, so this line of inquiry is a good one, towards verifying that.

tim-moody, 2023-05-13:
not sure what to make of 

```
May 13 18:42:44 box systemd[1]: Stopped Provides the Kiwix ZIM File Server.
May 13 18:42:44 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
May 13 18:42:44 box kiwix-serve[3750]: Loading the library from the following files:
May 13 18:42:44 box kiwix-serve[3750]: #011/library/zims/library.xml
May 13 18:42:44 box kiwix-serve[3750]: The library was successfully loaded.
May 13 18:42:44 box systemd[1]: Started Provides the Kiwix ZIM File Server.
May 13 18:42:44 box kiwix-serve[3751]: The Kiwix server is running and can be accessed in the local network at: http://192.168.20.127:3000/kiwix
May 13 18:42:53 box systemd[1]: systemd-timedated.service: Succeeded.
May 13 18:43:39 box IIAB-CMDSRV[1211]: IIAB-CMDSRV : Received CMD Message GET-MENU-ITEM-DEF-LIST.
May 13 18:43:42 box IIAB-CMDSRV[1211]: IIAB-CMDSRV : Received CMD Message UPDATE-HOME-MENU {}.
May 13 18:43:54 box dbus-daemon[418]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service' requested by ':1.62' (uid=0 pid=3772 comm="/usr/bin/timedatectl ")
May 13 18:43:54 box systemd[1]: Starting Time & Date Service...
May 13 18:43:54 box dbus-daemon[418]: [system] Successfully activated service 'org.freedesktop.timedate1'
May 13 18:43:54 box systemd[1]: Started Time & Date Service.
May 13 18:44:00 box systemd[1]: kiwix-serve.service: Main process exited, code=killed, status=11/SEGV
May 13 18:44:00 box systemd[1]: kiwix-serve.service: Failed with result 'signal'.
May 13 18:44:05 box systemd[1]: kiwix-serve.service: Scheduled restart job, restart counter is at 3.
May 13 18:44:05 box systemd[1]: Stopped Provides the Kiwix ZIM File Server.
May 13 18:44:05 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
```

is kiwix failing or are you restarting it?

tim-moody, 2023-05-13:
The only other fails I see are
```

May 13 18:34:45 box udisksd[479]: failed to load module mdraid: libbd_mdraid.so.2: cannot open shared object file: No such file or directory
May 13 18:34:45 box udisksd[479]: Failed to load the 'mdraid' libblockdev plugin

```

holta, 2023-05-13:
Just FYI 64-bit kiwix-tools for ARM is now available:
https://download.kiwix.org/release/kiwix-tools/kiwix-tools_linux-aarch64-3.5.0.tar.gz

CLARIF: IIAB should experiment with this very shortly (regardless whether or not helpful to this particular issue here!)

holta, 2023-05-13:
FWIW my experience on Ubuntu 23.10 is a bit different.

Every Content Pack (EduPack) is clickable from IIAB's home page except for:

- Khan Academy Lite: Health and Medicine: http://10.8.0.18:8008/learn/khan/science/health-and-medicine/
- TEDMED Medical TED Talks: http://10.8.0.18/js-menu/menu-files/html/undefined.html?menu_item=en-tedmed_en_all&zim_name=tedmed_en_all — later after reboot this one worked: http://10.8.0.18/kiwix/viewer#tedmed_en_all_2023-03/

holta, 2023-05-13:
"medical-preset-ubuntu-23.10" iiab-diagostics: http://sprunge.us/ojExP0?en

tim-moody, 2023-05-13:
is it possible that kalite was installed by iiab without the language pack? I'm not sure that the preset will pick up that step.

holta, 2023-05-13:
> is it possible that kalite was installed by iiab without the language pack? I'm not sure that the preset will pick up that step.

Oops it turns out I forgot that KA Lite fails to install on Ubuntu 23.10 — again demonstrated in the [iiab-diagnostics](http://sprunge.us/ojExP0?en) above:

```
2023-05-13 10:51:45,114 p=44727 u=root n=ansible | TASK [kalite : Use scripts/install_python2.sh to install python2 and virtualenv] ***
2023-05-13 10:52:36,453 p=44727 u=root n=ansible | changed: [127.0.0.1]
2023-05-13 10:52:36,470 p=44727 u=root n=ansible | TASK [kalite : Use pip to pin setuptools to 44 in /usr/local/kalite/venv] ******
2023-05-13 10:52:37,464 p=44727 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/usr/bin/virtualenv", "-ppython2.7", "/usr/local/kalite/venv"], "msg": "stdout: RuntimeError: failed to find interpreter for Builtin discover of python_spec='python2.7'\n"}
2023-05-13 10:52:37,476 p=44727 u=root n=ansible | TASK [kalite : SEE ERROR ABOVE (skip_role_on_error: True)] *********************
2023-05-13 10:52:37,498 p=44727 u=root n=ansible | skipping: [127.0.0.1]
2023-05-13 10:52:37,544 p=44727 u=root n=ansible | PLAY RECAP *********************************************************************
2023-05-13 10:52:37,544 p=44727 u=root n=ansible | 127.0.0.1                  : ok=26   changed=4    unreachable=0    failed=0    skipped=6    rescued=1    ignored=0
```

We should either (1) try to fix this in coming weeks if possible — or (2) begin the move away from KA Lite to Kolibri:

- #3573

stebbo63, 2023-05-13:
> What happens if you go to url http://box.lan/kiwix/ Let's make sure kiwix is installed and working.

Hi Tim,

thanks for your response.

if I go to that URL, I get a 502 bad gateway.  Upon checking the processes, I do see kiwix running, as shown from the output of a ps -eaf.
/opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml --urlRootLocation=/kiwix/ --threads 4

This does look to be the likely culprit at this stage go.

stebbo63, 2023-05-13:
Hi All,

thanks to those who responded, I will try to answer all your questions.  I responded to Tim above.

> is kiwix failing or am I restarting it?
I wasn't restarting it, so it must be failing.

> mdraid
I don't know what this is.  It looks like it's something to do with RAID and as I'm running on an SDCard on a raspberry pi I don't think it's relevant?

So it does seem that any menu option related to kiwix fails with a 502, whereas all the other menu options are working.

Regards,
Chris.

stebbo63, 2023-05-13:
Hi All,

one last answer.  With regards the content install, I had waited until the status of all jobs was "Succeeded".  I have looked through the status page and can't see anything obvious.

Regards,
Chris.

holta, 2023-05-13:
REASON: The `armhf` version of kiwix-tools 3.5.0 fails on 64-bit Raspberry Pi OS.

This is a sudden change (since kiwix-tools 3.5.0 was released 2023-04-28) due to their new compilation tooling upstream.

A workaround is now part of IIAB here:

- PR #3575

If you want to install the workaround, run:

```
cd /opt/iiab/iiab
sudo git pull
sudo ./runrole --reinstall kiwix
```

(Thanks @stebbo63 for letting us know!)

holta, 2023-05-13:
Sadly this issue turns out to be even more serious — affecting almost everyone using Raspberry Pi OS 32-bit or 64-bit:

Specifically, the `armhf` version of kiwix-tools 3.5.0 *also* fails[1] on 32-bit Raspberry Pi OS on Raspberry Pi 4 — even when it's running its 32-bit kernel.

[1] `kiwix-serve` and its systemd service run, but no website appears, which is what leads IIAB's proxy'd display of kiwix-tools 3.5.0's `armhf` to always show **502 Bad Gateway**.

This suggests that the `armhf` release of kixix-tools 3.5.0 has very serious problems — _and possibly needs to be avoided entirely, until legit solution(s) appear?_

(@mgautierfr might know more, if he has time in coming days?)

Tangentially related background:

- #3516
  - as explained at https://github.com/iiab/iiab/pull/3422#issuecomment-1533441463

holta, 2023-05-13:
REQUEST: If others can help, the problematic https://download.kiwix.org/release/kiwix-tools/kiwix-tools_linux-armhf-3.5.0.tar.gz should also be tested on Raspberry Pi 3 — and on Raspberry Pi Zero W — _to see if it works at all (on any hardware?)_

RECAP: So far kiwix-tools_linux-armhf-3.5.0.tar.gz seems to fail in all conditions (true 64-bit or true 32-bit) on Raspberry Pi 4.

holta, 2023-05-14:
Ticket subject line updated to reflect these unfortunate-but-real testing results:

- `armhf` version of kiwix-tools 3.5.0 appears to fail on all Raspberry Pi's — RPi Zero W, RPi 3, RPi 4 (and probably others too)
- `armhf` version of kiwix-tools 3.5.0 appears to fail regardless of OS — Raspberry Pi OS or Ubuntu

**FYI the situation appears even more serious on Raspberry Pi Zero W** (running a fully-updated version of 32-bit Raspberry Pi OS Lite).  In this case, the `armhf` version of kiwix-serve 3.5.0 unfortunately doesn't run at all:

```
root@box:~# /opt/iiab/kiwix/bin/kiwix-serve --version
Illegal instruction

root@box:~# systemctl start kiwix-serve
Job for kiwix-serve.service failed because a fatal signal was delivered to the control process.
See "systemctl status kiwix-serve.service" and "journalctl -xe" for details.

root@box:~# systemctl status kiwix-serve.service
● kiwix-serve.service - Provides the Kiwix ZIM File Server
     Loaded: loaded (/etc/systemd/system/kiwix-serve.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: signal) since Sat 2023-05-13 21:04:03 EDT; 4s ago
    Process: 1383 ExecStart=/opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml --urlRootLocation=/kiwix/ --threads 4 (code=killed, signal=ILL)
        CPU: 10ms

root@box:~# journalctl -u kiwix-serve.service | tail
May 13 21:04:35 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
May 13 21:04:35 box systemd[1]: kiwix-serve.service: Control process exited, code=killed, status=4/ILL
May 13 21:04:35 box systemd[1]: kiwix-serve.service: Failed with result 'signal'.
May 13 21:04:35 box systemd[1]: Failed to start Provides the Kiwix ZIM File Server.
May 13 21:04:40 box systemd[1]: kiwix-serve.service: Scheduled restart job, restart counter is at 69.
May 13 21:04:40 box systemd[1]: Stopped Provides the Kiwix ZIM File Server.
May 13 21:04:40 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
May 13 21:04:40 box systemd[1]: kiwix-serve.service: Control process exited, code=killed, status=4/ILL
May 13 21:04:40 box systemd[1]: kiwix-serve.service: Failed with result 'signal'.
May 13 21:04:40 box systemd[1]: Failed to start Provides the Kiwix ZIM File Server.

root@box:~# uname -a
Linux box 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux

root@box:~# dpkg --print-architecture
armhf
```

**In contrast on most other Raspberry Pi's (arbitrary example below is Ubuntu Server 23.04 on Raspberry Pi 4) the `armhf` version of kiwix-serve 3.5.0 at least tries to start:** (before failing to display any web page on port 3000)

```
root@box:~# /opt/iiab/kiwix/bin/kiwix-serve --version
kiwix-tools 3.5.0

libkiwix 12.0.0
+ libzim 8.2.0
+ libxapian 1.4.22
+ libcurl 7.67.0
+ libmicrohttpd 0.9.76
+ libz 1.2.12
+ libicu 58.2.0
+ libpugixml 0.12.0

libzim 8.2.0
+ libzstd 1.5.2
+ liblzma 5.2.6
+ libxapian 1.4.22
+ libicu 58.2.0

root@box:~# uname -a
Linux box 5.15.0-1028-raspi #30-Ubuntu SMP PREEMPT Thu Apr 20 01:21:34 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux

root@box:~# dpkg --print-architecture
arm64

root@box:~# systemctl status kiwix-serve.service
● kiwix-serve.service - Provides the Kiwix ZIM File Server
     Loaded: loaded (/etc/systemd/system/kiwix-serve.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2023-05-13 21:20:12 EDT; 2min 40s ago
    Process: 20011 ExecStart=/opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml --urlRootLocation=/kiwix/ --threads 4 (code=exited, status=0/SUCCESS)
   Main PID: 20012 (kiwix-serve)
      Tasks: 5 (limit: 2082)
     Memory: 7.8M
        CPU: 107ms
     CGroup: /system.slice/kiwix-serve.service
             └─20012 /opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml --urlRootLocation=/kiwix/ --threads 4

May 13 21:20:11 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
May 13 21:20:11 box kiwix-serve[20011]: Loading the library from the following files:
May 13 21:20:11 box kiwix-serve[20011]:         /library/zims/library.xml
May 13 21:20:12 box kiwix-serve[20011]: The library was successfully loaded.
May 13 21:20:12 box systemd[1]: Started Provides the Kiwix ZIM File Server.
May 13 21:20:12 box kiwix-serve[20012]: The Kiwix server is running and can be accessed in the local network at: http://192.168.0.181:3000/kiwix

root@box:~# journalctl -u kiwix-serve.service | tail
May 13 20:55:02 box systemd[1]: Started Provides the Kiwix ZIM File Server.
May 13 21:20:11 box systemd[1]: Stopping Provides the Kiwix ZIM File Server...
May 13 21:20:11 box systemd[1]: kiwix-serve.service: Deactivated successfully.
May 13 21:20:11 box systemd[1]: Stopped Provides the Kiwix ZIM File Server.
May 13 21:20:11 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
May 13 21:20:11 box kiwix-serve[20011]: Loading the library from the following files:
May 13 21:20:11 box kiwix-serve[20011]:         /library/zims/library.xml
May 13 21:20:12 box kiwix-serve[20011]: The library was successfully loaded.
May 13 21:20:12 box systemd[1]: Started Provides the Kiwix ZIM File Server.
May 13 21:20:12 box kiwix-serve[20012]: The Kiwix server is running and can be accessed in the local network at: http://192.168.0.181:3000/kiwix
```

**Unfortunately however, the URL announced above (http://192.168.0.181:3000/kiwix which is unproxied) never displays an actual web page — e.g. Chrome browser oscillates between these 2 error messages:

[image]

[image]

holta, 2023-05-14:
As mentioned above, the `armhf` version of kiwix-tools 3.5.0 appears to fail across the board.

So here's just 1 more data point for the record — in this case using a fully updated version of 32-bit Raspberry Pi OS Lite on Raspberry Pi 3:

```
root@box:~# uname -a
Linux box 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

root@box:~# dpkg --print-architecture
armhf

root@box:~# tail -1 /proc/cpuinfo
Model           : Raspberry Pi 3 Model B Rev 1.2

root@box:~# /opt/iiab/kiwix/bin/kiwix-serve --version
kiwix-tools 3.5.0

libkiwix 12.0.0
+ libzim 8.2.0
+ libxapian 1.4.22
+ libcurl 7.67.0
+ libmicrohttpd 0.9.76
+ libz 1.2.12
+ libicu 58.2.0
+ libpugixml 0.12.0

libzim 8.2.0
+ libzstd 1.5.2
+ liblzma 5.2.6
+ libxapian 1.4.22
+ libicu 58.2.0

root@box:~# systemctl status kiwix-serve.service
● kiwix-serve.service - Provides the Kiwix ZIM File Server
     Loaded: loaded (/etc/systemd/system/kiwix-serve.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2023-05-13 22:01:11 EDT; 9min ago
    Process: 550 ExecStart=/opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml --urlRootLocation=/kiwix/ --threads 4 (code=exited, status=0/SUCCESS)
   Main PID: 643 (kiwix-serve)
      Tasks: 5 (limit: 1595)
        CPU: 178ms
     CGroup: /system.slice/kiwix-serve.service
             └─643 /opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml --urlRootLocation=/kiwix/ --threads 4

May 13 22:01:10 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
May 13 22:01:11 box kiwix-serve[550]: Loading the library from the following files:
May 13 22:01:11 box kiwix-serve[550]:         /library/zims/library.xml
May 13 22:01:11 box kiwix-serve[550]: The library was successfully loaded.
May 13 22:01:11 box kiwix-serve[643]: The Kiwix server is running and can be accessed in the local network at: http://127.0.0.1:3000/kiwix

root@box:~# journalctl -u kiwix-serve.service | tail -20
May 13 22:00:50 box systemd[1]: kiwix-serve.service: Scheduled restart job, restart counter is at 711.
May 13 22:00:50 box systemd[1]: Stopped Provides the Kiwix ZIM File Server.
May 13 22:00:50 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
May 13 22:00:50 box systemd[1]: kiwix-serve.service: Control process exited, code=killed, status=4/ILL
May 13 22:00:50 box systemd[1]: kiwix-serve.service: Failed with result 'signal'.
May 13 22:00:50 box systemd[1]: Failed to start Provides the Kiwix ZIM File Server.
May 13 22:00:55 0I systemd[1]: kiwix-serve.service: Scheduled restart job, restart counter is at 712.
May 13 22:00:55 0I systemd[1]: Stopped Provides the Kiwix ZIM File Server.
May 13 22:00:55 0I systemd[1]: Starting Provides the Kiwix ZIM File Server...
May 13 22:00:55 0I systemd[1]: kiwix-serve.service: Control process exited, code=killed, status=4/ILL
May 13 22:00:55 0I systemd[1]: kiwix-serve.service: Failed with result 'signal'.
May 13 22:00:55 0I systemd[1]: Failed to start Provides the Kiwix ZIM File Server.
May 13 22:00:59 0I systemd[1]: Stopped Provides the Kiwix ZIM File Server.
-- Boot c2f823b101f84c81868f35abb1ba6b5c --
May 13 22:01:10 box systemd[1]: Starting Provides the Kiwix ZIM File Server...
May 13 22:01:11 box kiwix-serve[550]: Loading the library from the following files:
May 13 22:01:11 box kiwix-serve[550]:         /library/zims/library.xml
May 13 22:01:11 box kiwix-serve[550]: The library was successfully loaded.
May 13 22:01:11 box kiwix-serve[643]: The Kiwix server is running and can be accessed in the local network at: http://127.0.0.1:3000/kiwix
May 13 22:01:11 box systemd[1]: Started Provides the Kiwix ZIM File Server.

root@box:~# hostname -I
192.168.0.131 10.10.10.10

root@box:~# curl http://127.0.0.1:3000/kiwix
curl: (52) Empty reply from server

root@box:~# wget http://127.0.0.1:3000/kiwix
--2023-05-13 22:13:48--  http://127.0.0.1:3000/kiwix
Connecting to 127.0.0.1:3000... connected.
HTTP request sent, awaiting response... No data received.
Retrying.

--2023-05-13 22:13:49--  (try: 2)  http://127.0.0.1:3000/kiwix
Connecting to 127.0.0.1:3000... failed: Connection refused.
```

stebbo63, 2023-05-14:
> cd /opt/iiab/iiab
sudo git pull
sudo ./runrole --reinstall kiwix

I can confirm that after running the above, all the kiwix home page links are working correctly.

If anyone would like me to do a test of a re-install, let me know. I'm happy to reformat the SD drive and do a complete install again.

Thanks everyone for the assistance.

Regards,
Chris.

holta, 2023-05-14:
> If anyone would like me to do a test of a re-install, let me know. I'm happy to reformat the SD drive and do a complete install again.

Not necessary for now (Many Thanks Chris!)

PS An additional patch is now merged to help 32-bit Raspberry Pi Zero W communities here:

- PR #3576