iiab/iiab #3581: Search not working for one of the modules

stebbo63, 2023-05-17:
Hi All,

I'm not sure if this is the right place to ask this or not.  I have setup a medical internet-in-a-box and have downloaded all the content (a few days ago).

I am going through all the menu items to make sure they all work, and all are working except for the search function for "The Global Emergency Medicine Wiki".  When I try to search on this page I get a return page with a "File not found" message.

All the other modules search function appears to be working fine.

Regards,
Chris.

Version:	master
Date Installed:	2023-05-14T01:06:49Z
Commit ID:	f8a543c4b855a458070cdfdd644086a394bab955

stebbo63, 2023-05-18:
@tim-moody   

Thanks for your response.

I am a member of the Rotary Club of Warrnambool.  Our club, plus a few others, have a long term project installing water pumps to villages in PNG.  Over the course of a few years, they've had the occasional scrape and illness, and have had to avail themselves of the medical facilities over there.  I found out about IIAB many years ago and have always thought there would be a good opportunity for it over there.  One of the other clubs has a project for the knowledge based IIAB, so we have decided to see if there is a need / use case for the medical IAAB.  I am currently setting up a full kit for the team to take over there later this year as a POC, and if it is successful I will be trying to deploy a number each year.

I have some good computer skills.  I was a unix network admin many years ago, and have reasonable skills with programming, scripting and general systems / network admin.  Once I'm a bit more familiar with IIAB I will look to try to find some way to contribute to the project.   If there's anything that you think that a newbie might be able to assist with, particularly with the medical IIAB then please feel free to send me some suggestions.  Note: I have no medical background.

Cheers,
Chris.

holta, 2023-05-18:
Thank you Chris 💯

Please email us and/or post here in GitHub anytime you need a hand.

tim-moody, 2023-05-18:
Sounds like a great project, and one we would certainly want to assist.

In the medical area there is always a need for new and updated content. If there is anything in particular that PNG could use, you could try to incorporate it. Also, you might want to create a 'preset' for PNG. Have a look at /opt/admin/cmdsrv/presets.