iiab/iiab #3585: Nextcloud no longer installs on Ubuntu 23.10 (10 days ago this worked!)

holta, 2023-05-20:
Something changed since 10 days ago:

( FYI Nextcloud 26.0.1. itself has not changed since 2023-04-19, as confirmed by looking at the timestamp of https://download.nextcloud.com/server/releases/latest.tar.bz2 within https://download.nextcloud.com/server/releases/ )

> TASK [nextcloud : Allow Nextcloud access from all hosts and IP addresses -- SEE ALSO THE OLD WAY: https://github.com/iiab/iiab/blob/master/roles/nextcloud/templates/nextcloud.conf.j2.unused] **********************************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["php", "/library/www/nextcloud/occ", "config:system:set", "trusted_domains", "1", "--value=*"], "delta": "0:00:00.078360", "end": "2023-05-20 11:35:47.159154", "msg": "non-zero return code", "rc": 1, "start": "2023-05-20 11:35:47.080794", "stderr": "Nextcloud is not installed - only a limited number of commands are available", "stderr_lines": ["Nextcloud is not installed - only a limited number of commands are available"], "stdout": "No database drivers (sqlite, mysql, or postgresql) installed.\n\n\nPHP module zip not installed.\nPlease ask your server administrator to install the module.\n\nPHP module dom not installed.\nPlease ask your server administrator to install the module.\n\nPHP module XMLWriter not installed.\nPlease ask your server administrator to install the module.\n\nPHP module XMLReader not installed.\nPlease ask your server administrator to install the module.\n\nPHP module libxml not installed.\nPlease ask your server administrator to install the module.\n\nPHP module mbstring not installed.\nPlease ask your server administrator to install the module.\n\nPHP module GD not installed.\nPlease ask your server administrator to install the module.\n\nPHP module SimpleXML not installed.\nPlease ask your server administrator to install the module.\n\nPHP module cURL not installed.\nPlease ask your server administrator to install the module.\n\nAn unhandled exception has been thrown:\nException: Environment not properly prepared. in /library/www/nextcloud/lib/private/Console/Application.php:167\nStack trace:\n#0 /library/www/nextcloud/console.php(99): OC\\Console\\Application->loadCommands()\n#1 /library/www/nextcloud/occ(11): require_once('...')\n#2 {main}", "stdout_lines": ["No database drivers (sqlite, mysql, or postgresql) installed.", "", "", "PHP module zip not installed.", "Please ask your server administrator to install the module.", "", "PHP module dom not installed.", "Please ask your server administrator to install the module.", "", "PHP module XMLWriter not installed.", "Please ask your server administrator to install the module.", "", "PHP module XMLReader not installed.", "Please ask your server administrator to install the module.", "", "PHP module libxml not installed.", "Please ask your server administrator to install the module.", "", "PHP module mbstring not installed.", "Please ask your server administrator to install the module.", "", "PHP module GD not installed.", "Please ask your server administrator to install the module.", "", "PHP module SimpleXML not installed.", "Please ask your server administrator to install the module.", "", "PHP module cURL not installed.", "Please ask your server administrator to install the module.", "", "An unhandled exception has been thrown:", "Exception: Environment not properly prepared. in /library/www/nextcloud/lib/private/Console/Application.php:167", "Stack trace:", "#0 /library/www/nextcloud/console.php(99): OC\\Console\\Application->loadCommands()", "#1 /library/www/nextcloud/occ(11): require_once('...')", "#2 {main}"]}

```
root@u2310:~# php /library/www/nextcloud/occ status
Console has to be executed with the user that owns the file config/config.php
Current user id: 0
Owner id of config.php: 33
Try adding 'sudo -u #33' to the beginning of the command (without the single quotes)
If running with 'docker exec' try adding the option '-u 33' to the docker command (without the single quotes)
```

```
root@u2310:~# ls -l /library/www/nextcloud/config/config.php
-rw-rw-r-- 1 www-data www-data 0 May 20 11:35 /library/www/nextcloud/config/config.php
```

```
root@u2310b:/opt/iiab/iiab# sudo -u www-data php /library/www/nextcloud/occ status
Nextcloud is not installed - only a limited number of commands are available
No database drivers (sqlite, mysql, or postgresql) installed.

PHP module zip not installed.
Please ask your server administrator to install the module.

PHP module dom not installed.
Please ask your server administrator to install the module.

PHP module XMLWriter not installed.
Please ask your server administrator to install the module.

PHP module XMLReader not installed.
Please ask your server administrator to install the module.

PHP module libxml not installed.
Please ask your server administrator to install the module.

PHP module mbstring not installed.
Please ask your server administrator to install the module.

PHP module GD not installed.
Please ask your server administrator to install the module.

PHP module SimpleXML not installed.
Please ask your server administrator to install the module.

PHP module cURL not installed.
Please ask your server administrator to install the module.

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /library/www/nextcloud/lib/private/Console/Application.php:167
Stack trace:
#0 /library/www/nextcloud/console.php(99): OC\Console\Application->loadCommands()
#1 /library/www/nextcloud/occ(11): require_once('...')
#2 {main}
```

Whereas on Ubuntu 22.04 it shows:

```
root@u2204:~# sudo -u www-data php /library/www/nextcloud/occ status
  - installed: true
  - version: 26.0.1.1
  - versionstring: 26.0.1
  - edition:
  - maintenance: false
  - needsDbUpgrade: false
  - productname: Nextcloud
  - extendedSupport: false
```

Related:

- #3528

jvonau, 2023-05-21:
Seems like a temporary OS php packaging issue, most of those dependencies are listed in [install.yml](https://github.com/iiab/iiab/blob/master/roles/nextcloud/tasks/install.yml#L50-L80)  What is the base version of php that is being installed and is there more than one version available to install ie 8.1/8.2?

holta, 2023-05-27:
Resolved with the latest Ubuntu 23.10 on PHP 8.2 ✅

http://sprunge.us/ewCukW?en

closed 2023-05-27 by holta