iiab/iiab #3602: cest-pas-sorcier_fr_top didn't appear on IIAB main page and Python JSONDecodeError

holta, 2023-06-13:
Just a photo for now, but hopefully @nzola can explain more later today:

[image]

Hopefully he can provide *iiab-diagnostics* and additional details within ~24h.

(This is very likely happening on Raspberry Pi OS.  As that's what he generally uses.)

nzola, 2023-06-13:
PUBLISHING TO URL...
http://sprunge.us/MhbXUt

holta, 2023-06-13:
1) I just logged into @nzola's machine and this worked to add the missing `cest-pas-sorcier_fr_top` to his IIAB home page:

   ```
   root@lokole1:/library/www/html/home# diff menu.json.sorcier-missing menu.json
   61a62
   >     "fr-cest-pas-sorcier_fr_top",
   ```

2) @tim-moody do you want to log in, to try to better understand the original Python `JSONDecodeError` issue?

tim-moody, 2023-06-14:
This appears to be another case of #3487. 
- The point of failure was line 1004 in adm_lib.py. 
- But it is almost impossible to reconstruct what events led to this. 
- The log shows that this error happened in the second step of  the download the cest_sorcier top zim, but it need not be due to that zim or its menu def. It simply failed to read the menu.json file prior to adding that zim to the menu. 
- The menu.json file purported to have failed has been preserved by renaming, but adm_lib reads it with no error.
- None of the corruptions in other instances of this problem are found in this file.
- I downloaded that zim through the Admin Console on my VM, and it was added to the home page with no problem.

The only unexpected results are
- No error messages were issued implying that verbose=False
- Seems that no attempt was made to fix the file implying that fix_json=False (it's had to be sure since there are no messages)

holta, 2023-06-14:
> This appears to be another case of https://github.com/iiab/iiab/issues/3487.

Great that diagnostic leads are documented (in case this happens again).

_Thanks @nzola and @tim-moody for having spent the the time to dig into this_ 🕵️