iiab/iiab #3642: Calibre-Web failing to install on "32-bit" Raspberry Pi OS 12 / Bookworm

holta, 2023-09-11:
@deldesir and @ozzieisaacs would you have suggestions or recommendations for the best way to go about solving this?  This appears to be the main error, which is confirmed repeatable:

> error'\n\n:stderr:   error: subprocess-exited-with-error\n  \n  × python setup.py egg_info did not run successfully.\n  │ exit code: 1\n  ╰─> [3 lines of output]\n      Building lxml version 4.9.3.\n      Building without Cython.\n      **Error: Please make sure the libxml2 and libxslt development packages are installed.**

```
root@box:~# apt list "*libxml2*"
Listing... Done
libxml2-dev/stable 2.9.14+dfsg-1.3~deb12u1 armhf
libxml2-doc/stable 2.9.14+dfsg-1.3~deb12u1 all
libxml2-utils/stable 2.9.14+dfsg-1.3~deb12u1 armhf
libxml2/stable,now 2.9.14+dfsg-1.3~deb12u1 armhf [installed]
python3-libxml2/stable 2.9.14+dfsg-1.3~deb12u1 armhf
```

```root@box:~# apt list "*libxslt*"
Listing... Done
libxml-libxslt-perl/stable 2.002001-1 armhf
libxslt1-dev/stable 1.1.35-1 armhf
libxslt1.1/stable,now 1.1.35-1 armhf [installed,automatic]
libxsltc-java/stable 2.7.2-4 all
libxslthl-java/stable 2.1.3-6 all
```

Ironically the "32-bit" (amhf) Raspberry Pi OS is running a 64-bit kernel (aarch64, on a Raspberry Pi 4) as confirmed by the OS's official new policy/strategy enacted 6 months ago in mid-March 2023: raspberrypi/firmware#1795

Here's the full error:

```
TASK [calibre-web : Download Calibre-Web dependencies from 'requirements.txt' into python3 virtual environment /usr/local/calibre-web-py3] **************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/usr/local/calibre-web-py3/bin/pip3", "install", "-r", "/usr/local/calibre-web-py3/requirements.txt"], "msg": "stdout: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple\nCollecting APScheduler<3.11.0,>=3.6.3\n  Downloading https://www.piwheels.org/simple/apscheduler/APScheduler-3.10.4-py3-none-any.whl (59 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 59.3/59.3 kB 378.0 kB/s eta 0:00:00\nRequirement already satisfied: Babel<3.0,>=1.3 in /usr/lib/python3/dist-packages (from -r /usr/local/calibre-web-py3/requirements.txt (line 2)) (2.10.3)\nCollecting Flask-Babel<3.1.0,>=0.11.1\n  Downloading https://www.piwheels.org/simple/flask-babel/flask_babel-3.0.1-py3-none-any.whl (11 kB)\nCollecting Flask-Login<0.6.3,>=0.3.2\n  Downloading Flask_Login-0.6.2-py3-none-any.whl (17 kB)\nCollecting Flask-Principal<0.5.1,>=0.3.2\n  Downloading https://www.piwheels.org/simple/flask-principal/Flask_Principal-0.4.0-py3-none-any.whl (6.7 kB)\nCollecting Flask<2.4.0,>=1.0.2\n  Downloading https://www.piwheels.org/simple/flask/flask-2.3.3-py3-none-any.whl (96 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 96.1/96.1 kB 36.3 kB/s eta 0:00:00\nCollecting iso-639<0.5.0,>=0.4.5\n  Downloading https://www.piwheels.org/simple/iso-639/iso_639-0.4.5-py3-none-any.whl (171 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 171.5/171.5 kB 19.1 kB/s eta 0:00:00\nCollecting PyPDF<3.8.0,>=3.0.0\n  Downloading https://www.piwheels.org/simple/pypdf/pypdf-3.7.1-py3-none-any.whl (248 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 248.1/248.1 kB 816.8 kB/s eta 0:00:00\nRequirement already satisfied: pytz>=2016.10 in /usr/lib/python3/dist-packages (from -r /usr/local/calibre-web-py3/requirements.txt (line 9)) (2022.7.1)\nRequirement already satisfied: requests<2.29.0,>=2.11.1 in /usr/lib/python3/dist-packages (from -r /usr/local/calibre-web-py3/requirements.txt (line 10)) (2.28.1)\nCollecting SQLAlchemy<2.0.0,>=1.3.0\n  Downloading SQLAlchemy-1.4.49.tar.gz (8.5 MB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 8.5/8.5 MB 7.4 MB/s eta 0:00:00\n  Preparing metadata (setup.py): started\n  Preparing metadata (setup.py): finished with status 'done'\nCollecting tornado<6.3,>=4.1\n  Downloading https://www.piwheels.org/simple/tornado/tornado-6.2-cp37-abi3-linux_armv7l.whl (422 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 422.5/422.5 kB 1.2 MB/s eta 0:00:00\nCollecting Wand<0.7.0,>=0.4.4\n  Downloading https://www.piwheels.org/simple/wand/Wand-0.6.11-py3-none-any.whl (143 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 143.6/143.6 kB 3.2 MB/s eta 0:00:00\nCollecting unidecode<1.4.0,>=0.04.19\n  Downloading https://www.piwheels.org/simple/unidecode/Unidecode-1.3.6-py3-none-any.whl (235 kB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 235.5/235.5 kB 3.1 MB/s eta 0:00:00\nCollecting lxml<5.0.0,>=3.8.0\n  Downloading lxml-4.9.3.tar.gz (3.6 MB)\n     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.6/3.6 MB 7.4 MB/s eta 0:00:00\n  Preparing metadata (setup.py): started\n  Preparing metadata (setup.py): finished with status 'error'\n\n:stderr:   error: subprocess-exited-with-error\n  \n  × python setup.py egg_info did not run successfully.\n  │ exit code: 1\n  ╰─> [3 lines of output]\n      Building lxml version 4.9.3.\n      Building without Cython.\n      Error: Please make sure the libxml2 and libxslt development packages are installed.\n      [end of output]\n  \n  note: This error originates from a subprocess, and is likely not a problem with pip.\nerror: metadata-generation-failed\n\n× Encountered error while generating package metadata.\n╰─> See above for output.\n\nnote: This is an issue with the package mentioned above, not pip.\nhint: See above for details.\n"}
```

```
root@box:~# cat /etc/pip.conf
[global]
extra-index-url=https://www.piwheels.org/simple
```

```
root@box:~# /usr/local/calibre-web-py3/bin/pip list
Package            Version
------------------ ----------
anyio              3.6.2
Babel              2.10.3
certifi            2022.9.24
chardet            5.1.0
charset-normalizer 3.0.1
click              8.1.3
colorama           0.4.6
colorzero          2.0
cryptography       38.0.4
dbus-python        1.3.2
distlib            0.3.6
distro             1.8.0
dnspython          2.3.0
filelock           3.9.0
geojson            3.0.0
gpg                1.18.0
gpiozero           1.6.2
h11                0.14.0
h2                 4.1.0
hpack              4.0.0
httpcore           0.16.3
httpx              0.23.3
hyperframe         6.0.0
idna               3.3
Jinja2             3.1.2
kolibri            0.15.12
markdown-it-py     2.1.0
MarkupSafe         2.1.2
mdurl              0.1.2
netifaces          0.11.0
olefile            0.46
passlib            1.7.4
Pillow             9.4.0
pip                23.0.1
platformdirs       2.6.0
psutil             5.9.4
psycopg2           2.9.5
Pygments           2.14.0
PyGObject          3.42.2
PyMySQL            1.0.2
python-apt         2.6.0
python-dateutil    2.8.2
pytz               2022.7.1
PyYAML             6.0
requests           2.28.1
requests-toolbelt  0.10.1
rfc3986            1.5.0
rich               13.3.1
RPi.GPIO           0.7.1a4
setuptools         66.1.1
six                1.16.0
sniffio            1.2.0
ssh-import-id      5.10
toml               0.10.2
urllib3            1.26.12
virtualenv         20.17.1+ds
wget               3.2
```

```
root@box:~# /usr/local/calibre-web-py3/bin/pip freeze
anyio==3.6.2
Babel==2.10.3
certifi==2022.9.24
chardet==5.1.0
charset-normalizer==3.0.1
click==8.1.3
colorama==0.4.6
colorzero==2.0
cryptography==38.0.4
dbus-python==1.3.2
distlib==0.3.6
distro==1.8.0
dnspython==2.3.0
filelock==3.9.0
geojson==3.0.0
gpg==1.18.0
gpiozero==1.6.2
h11==0.14.0
h2==4.1.0
hpack==4.0.0
httpcore==0.16.3
httpx==0.23.3
hyperframe==6.0.0
idna==3.3
Jinja2==3.1.2
kolibri==0.15.12
markdown-it-py==2.1.0
MarkupSafe==2.1.2
mdurl==0.1.2
netifaces==0.11.0
olefile==0.46
passlib==1.7.4
Pillow==9.4.0
platformdirs==2.6.0
psutil==5.9.4
psycopg2==2.9.5
Pygments==2.14.0
PyGObject==3.42.2
PyMySQL==1.0.2
python-apt==2.6.0
python-dateutil==2.8.2
pytz==2022.7.1
PyYAML==6.0
requests==2.28.1
requests-toolbelt==0.10.1
rfc3986==1.5.0
rich==13.3.1
RPi.GPIO==0.7.1a4
six==1.16.0
sniffio==1.2.0
ssh-import-id==5.10
toml==0.10.2
urllib3==1.26.12
virtualenv==20.17.1+ds
wget==3.2
```

iiab-diagnostics showing full machine/OS details etc:
http://sprunge.us/PQKzpJ?en

Tangentially Related:

- #3639
- #3640
- #3641

OzzieIsaacs, 2023-09-12:
Seems somebody else has similar problems reported: https://bugs.launchpad.net/lxml/+bug/1969912

holta, 2023-09-12:
> Seems somebody else has similar problems reported: https://bugs.launchpad.net/lxml/+bug/1969912

Cool!  Looks like lxml [officially works fine with Python 3.11](https://bugs.launchpad.net/lxml/+bug/1969912/comments/7)

(So I'll ask piwheels.org folks to see if they can publish an lxml wheel for lxml, hoping that might solve it ;)

holta, 2023-09-19:
https://www.piwheels.org/project/lxml/ (https://www.piwheels.org/simple/lxml/) appears to have largely solved the problem — thanks to:

```
root@box:/opt/iiab/iiab# /usr/local/calibre-web-py3/bin/pip list | grep lxml
lxml                4.9.2
root@box:/opt/iiab/iiab# /usr/local/calibre-web-py3/bin/pip freeze | grep lxml
lxml==4.9.2
root@box:/opt/iiab/iiab# python3 --version
Python 3.11.2
```

And hopefully they'll also offer a Python 3.11 wheel for the latest version of lxml (4.9.3) soon in coming days.

Fully 32-bit kernel presumably also works, but was not yet tested (hopefully soon in coming days).

In any case Calibre-Web now succeeds in installing onto.... "32-bit" armhf Raspberry Pi OS... that (in this case, on RPi 4, had) auto-selected a 64-bit aarch64 kernel on boot:

```
root@box:~# iiab-summary
install_date = 2023-09-11T06:16:28Z   Current TZ: EDT

iiab: 98349b1e6, 52 PR's / 236 commits since tag 8.1-preview-2
    "Merge pull request #3647 from holta/lshw_for_iiab-diagnostics"
    https://github.com/iiab/iiab   branch: master

iiab-admin-console: f085beb, 19 PR's / 62 commits since tag v0.6.2
    "Merge pull request #545 from tim-moody/0.6.6-job-outp"
    https://github.com/iiab/iiab-admin-console   branch: master

Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 47eee1f0ddcf8811559d51eea1c1bb48335e3e88, stage2
/etc/debian-version: 12.1
display-manager? inactive   Arch1: armhf   Arch2:
box 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64
Cortex-A72   CPU(s): 4   Mem: 1849 Swap: 1023
Raspberry Pi 4 Model B Rev 1.1
Temperature(s): 55991

Filesystem      Size  Used Avail Use% Mounted on
/dev/root        59G   12G   47G  20% /      ZIMs: 1 OER2Go: 1 Apps2B: 2

lo eth0 wlan0 ap0 br0
openvpn_enabled: False
openvpn_handle: 183-raspios-armhf-aarch64-LARGE
192.168.0.183 10.10.10.10
```

holta, 2023-09-26:
@deldesir

Awesome News: https://piwheels.org/ just fixed it (lxml 4.9.3 for Python 3.11) today after I pinged them about this failed build!

- piwheels/packages#391

Do this ticket can probably be closed now! :pray:

closed 2023-10-25 by holta