iiab/iiab #3646: Nested (sub) Menuing

Jamhuri2023, 2023-09-15:
### Expected behavior and actual behavior: I don't know how to create nested (sub) menu's.  I have read the GITHUB item on menuing (https://github.com/iiab/iiab/wiki/IIAB-Menuing#overview) and am still lost.  I have loaded many wiki, khan video, ubongo (in Tanzania), etc. files and they all show up individually.  I would like to create a single menu item for each major type item with submenu drop down or indented or something.  Like to do it with dynamic menu I'm still adding content.  I've read all the GITHUB/FAQs I can find but not clear how I edit the .json/index files lisited.  Not sure how I do 'This allows you to nest menus by using a menu item with type (intended_use) info to point to other menus. " - where, what file, what is the syntax, which type, etc.?  Is there an example?  BTW, I'm create an IIAB for a primary school in Moshi Tanzania for about 600+ students.  Any help would be greatly appreciated!

### Steps to reproduce the problem:  I've done the normal movement of files with the admin console.

### IIAB version, OS version, hardware details:IIAB latest version dowload only a few days ago.  Running on Raspberry PI 4 with Rasbian.  Have attached 2TB drive.
### IIAB version, OS version, hardware details:

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
http://sprunge.us/4vbQBP

holta, 2023-09-15:
@Jamhuri2023

Tough question as work is ongoing here and not quite resolved as to how to solve this cleanly for everyone!

😄

Definitely keep in touch please, and feel free to email if necessary to explain more your Tanzanian context / needs, as many others are also hoping for a breakthrough here in coming months 🙏

Related:

- #2110

Jamhuri2023, 2023-09-15:
Thanks for the quick response.  Sounds like the solution may be being worked but not for today - okay.  For background, my wife and i are teaching at a public primary school here in Moshi, Tanzania - 600 students 1-7 grades - great kids/teachers but limited resources - no internet, just installed solar for power, and 5 computers (various ages).  Have 2 PI's loaded with IIAB and lots of wiki, ubongo, khan (english), maps, etc.  Many, many questions about IIAB; some simple, some tricky.  Current rabbit hole is how to load kalibri and run with IIAB.  Is it a separate OS/system or can it be run from IIAB?  How do you load it, create accounts, what is the admin PW, etc?  Appreciate the quick response to the last question and any help or resources you can provide!  The kids here thank you!  Mark Rose

holta, 2023-09-15:
Awesome news, thanks Mark / @Jamhuri2023 for pushing forward! 💯

Have you read https://github.com/iiab/iiab/blob/master/roles/kolibri/README.rst to get oriented around Kolibri 0.15.12 ?
Or you can install Kolibri 0.16.0 Beta 5 if you're courageous / daring, if you choose to set the following line (prior to installing Kolibri!) in [/etc/iiab/local_vars.yml](https://wiki.iiab.io/go/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it?) ?

```
kolibri_deb_url: https://github.com/learningequality/kolibri/releases/download/v0.16.0-beta5/kolibri_0.16.0b5-0ubuntu1_all.deb
```

(AT YOUR OWN RISK, as Kolibri 0.16 is not-quite-yet released software of course!)  And then of course install Kolibri, e.g. using IIAB's Admin Console, or as the command-line, using:

```
cd /opt/iiab/iiab
sudo ./runrole kolibri
```

PS Feel free to email directly if necessary (holt@unleashkids.org) as you navigate your IIAB customizations.

Jamhuri2023, 2023-09-15:
Many thanks!  Haven't made it all the way through yet but this should answer many of my questions.  Appreciate the help!

Mark

holta, 2023-09-15:
Correction / Clarification:
I just noticed you already have Kolibri (likely version 0.15.12) installed, on Line 915 here:

http://sprunge.us/4vbQBP?en#n-915

Good Luck!

holta, 2023-09-21:
@Jamhuri2023 I hope it's OK that I paste in both your Raspberry Pi 4's disk diagnostics, in the spirit of open source towards helping others, as you continue to make progress here?

Both look great below, FYI!  Let us know if a loose cable on the external disk (or any other theory!?) is verified and confirmed in coming days :smile:

IIAB1 below:

```
admin@box:~ $ ls /library
archiveorg   downloads  kolibri           pgsql-iiab    working
awstats      games      lost+found        public        www
calibre-web  gitea      mediawiki-1.40.0  transmission  zims
dbdata       ka-lite    moodle            wordpress
admin@box:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        29G   11G   18G  39% /
devtmpfs        1.7G     0  1.7G   0% /dev
tmpfs           1.9G   18M  1.9G   1% /dev/shm
tmpfs           759M  1.4M  758M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  255M   31M  225M  13% /boot
/dev/sda1       1.8T  287G  1.5T  17% /library
tmpfs           380M   20K  380M   1% /run/user/1000
admin@box:~ $ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    0  1.8T  0 disk 
└─sda1        8:1    0  1.8T  0 part /media/usb0
mmcblk0     179:0    0 29.7G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 29.5G  0 part /
admin@box:~ $ blkld
bash: blkld: command not found
admin@box:~ $ blkid
/dev/mmcblk0p1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="35DE-9C73" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="5904da31-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="2aa5eb08-26a7-434c-a9f9-6ab89ea0f362" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="5904da31-02"
/dev/sda1: LABEL="iiab-Library1" UUID="139b8556-05ce-4d50-84fd-09dab2b39bd8" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="primary" PARTUUID="b0d4da9c-c718-475e-9eb4-197dd0bf87b9"
```

IIAB2 below:

```
admin@box:~ $ ls /library
archiveorg   downloads  kolibri           moodle        usb0       zims
awstats      games      lokole            pgsql-iiab    wordpress
calibre-web  gitea      lost+found        public        working
dbdata       ka-lite    mediawiki-1.40.0  transmission  www
admin@box:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       117G   15G  101G  13% /
devtmpfs        1.7G     0  1.7G   0% /dev
tmpfs           1.9G   21M  1.9G   2% /dev/shm
tmpfs           759M  1.4M  758M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  255M   31M  225M  13% /boot
/dev/sda1       1.8T  237G  1.5T  14% /library
tmpfs           380M   20K  380M   1% /run/user/1000
admin@box:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0   1.8T  0 disk 
└─sda1        8:1    0   1.8T  0 part /media/usb0
mmcblk0     179:0    0 119.1G  0 disk 
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0 118.8G  0 part /
admin@box:~ $ blkid
/dev/mmcblk0p1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="35DE-9C73" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="173d528a-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="2aa5eb08-26a7-434c-a9f9-6ab89ea0f362" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="173d528a-02"
/dev/sda1: LABEL="iiab-Library" UUID="5fe80dda-ba59-4bd3-ba63-2c9ccacdd073" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="primary" PARTUUID="4ec95de7-a8e5-4141-9cf8-f46f40881196"
```

Somewhat related background:

- #2679
- #3106
- #3247
- #3320

tim-moody, 2023-09-25:
If by submenu you mean a menu item that has a dropdown with links to other items, there is no support for that. It is possible to create a menu item with links to other menus, though it is undocumented. As an example look at https://iiab.me/home/ which links to various other menus.

A menu is a json file in a folder. The easiest and by far safest way to edit it is via the Content Menus section of Admin Console. The most important trick in doing this is that while you cannot create a menu item you can clone one and change all the properties of the cloned item. So you could create a menu, save it to a different folder, go back to the home menu and clone and item, setting the Menu Item Type to App in Web Root then set the Start Url to the name of the folder holding the target menu. This is a pretty sparse description, but maybe gets you started.