iiab/iiab #3650: ansible-core does not install on FULLY 32-bit Raspberry Pi OS Bullseye — attempts cryptography 41.0.4 and 41.0.3 — "ERROR: Failed building wheel for cffi"

holta, 2023-09-30:
Raspberry Pi OS Bullseye is heading into legacy/retirement status starting mid-October, as announced at https://raspberrypi.com/5 — and almost everyone (who can) should now be running 64-BIT, for sure!

Still, FYI for @bennuttall @MichaIng and anybody who truly must run FULLY 32-BIT OS[1] e.g. on Zero W hardware — pip installing ansible-core (i.e. any recent versions of Ansible from recent years) with fail[2] as follows:

[1] If on 64-bit hardware, force a 32-bit kernel by setting `arm_64bit=0` in `/boot/config.txt` then reboot, per raspberrypi/firmware#1795
[2] [/opt/iiab/iiab/scripts/ansible](https://github.com/iiab/iiab/blob/master/scripts/ansible) below is really... just a venv wrapper around `pip install ansible-core`

```
root@raspberrypi:~# uname -a
Linux raspberrypi 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux
```

```
root@raspberrypi:~# cat /etc/pip.conf
[global]
extra-index-url=https://www.piwheels.org/simple
```

```
root@raspberrypi:~# /opt/iiab/iiab/scripts/ansible    # RUNS 'pip install ansible-core' IN A VIRTUALENV

[SNIP]

Create virtual environment for Ansible
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ansible-core
  Downloading https://www.piwheels.org/simple/ansible-core/ansible_core-2.15.4-py3-none-any.whl (2.2 MB)
     |████████████████████████████████| 2.2 MB 395 kB/s
Collecting cryptography
  Downloading cryptography-41.0.4.tar.gz (630 kB)
     |████████████████████████████████| 630 kB 9.2 MB/s
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/local/ansible/bin/python3 /usr/local/ansible/lib/python3.9/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-flt7z7rf/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=61.0.0' wheel 'cffi>=1.12; platform_python_implementation != '"'"'PyPy'"'"'' 'setuptools-rust>=0.11.4'
       cwd: None
  Complete output (109 lines):
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
  Collecting setuptools>=61.0.0
    Downloading https://www.piwheels.org/simple/setuptools/setuptools-68.2.2-py3-none-any.whl (807 kB)
  Collecting wheel
    Downloading https://www.piwheels.org/simple/wheel/wheel-0.41.2-py3-none-any.whl (64 kB)
  Collecting cffi>=1.12
    Downloading cffi-1.16.0.tar.gz (512 kB)
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
    Installing backend dependencies: started
    Installing backend dependencies: finished with status 'done'
      Preparing wheel metadata: started
      Preparing wheel metadata: finished with status 'done'
  Collecting setuptools-rust>=0.11.4
    Downloading https://www.piwheels.org/simple/setuptools-rust/setuptools_rust-1.7.0-py3-none-any.whl (25 kB)
  Collecting pycparser
    Downloading https://www.piwheels.org/simple/pycparser/pycparser-2.21-py2.py3-none-any.whl (119 kB)
  Collecting semantic-version<3,>=2.8.2
    Downloading https://www.piwheels.org/simple/semantic-version/semantic_version-2.10.0-py2.py3-none-any.whl (19 kB)
  Collecting typing-extensions>=3.7.4.3
    Downloading https://www.piwheels.org/simple/typing-extensions/typing_extensions-4.8.0-py3-none-any.whl (31 kB)
  Collecting tomli>=1.2.1
    Downloading https://www.piwheels.org/simple/tomli/tomli-2.0.1-py3-none-any.whl (12 kB)
  Building wheels for collected packages: cffi
    Building wheel for cffi (PEP 517): started
    Building wheel for cffi (PEP 517): finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /usr/local/ansible/bin/python3 /tmp/tmpf48ibbd7_in_process.py build_wheel /tmp/tmpa1v3v2jk
         cwd: /tmp/pip-install-huz264ie/cffi_2a866c15ee9441e1b8c5fde466f58550
    Complete output (73 lines):
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
[... 161 lines cut ...]
  copying src/cffi/cffi_opcode.py -> build/lib.linux-armv7l-cpython-39/cffi
  copying src/cffi/recompiler.py -> build/lib.linux-armv7l-cpython-39/cffi
  copying src/cffi/vengine_cpy.py -> build/lib.linux-armv7l-cpython-39/cffi
  copying src/cffi/_shimmed_dist_utils.py -> build/lib.linux-armv7l-cpython-39/cffi
  copying src/cffi/cparser.py -> build/lib.linux-armv7l-cpython-39/cffi
  copying src/cffi/_cffi_include.h -> build/lib.linux-armv7l-cpython-39/cffi
  copying src/cffi/parse_c_type.h -> build/lib.linux-armv7l-cpython-39/cffi
  copying src/cffi/_embedding.h -> build/lib.linux-armv7l-cpython-39/cffi
  copying src/cffi/_cffi_errors.h -> build/lib.linux-armv7l-cpython-39/cffi
  running build_ext
  building '_cffi_backend' extension
  creating build/temp.linux-armv7l-cpython-39
  creating build/temp.linux-armv7l-cpython-39/src
  creating build/temp.linux-armv7l-cpython-39/src/c
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DFFI_BUILDING=1 -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/usr/include/ffi -I/usr/include/libffi -I/usr/local/ansible/include -I/usr/include/python3.9 -c src/c/_cffi_backend.c -o build/temp.linux-armv7l-cpython-39/src/c/_cffi_backend.o
  src/c/_cffi_backend.c:2:10: fatal error: Python.h: No such file or directory
      2 | #include <Python.h>
        |          ^~~~~~~~~~
  compilation terminated.
  error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for cffi
Failed to build cffi
ERROR: Could not build wheels for cffi which use PEP 517 and cannot be installed directly
```

MichaIng, 2023-09-30:
Not sure why/how I am involved here, but looking at the error message:
```
Package libffi was not found in the pkg-config search path.
```
=>
```sh
apt install libffi-dev
```
should solve it.

As of 32-bit kernel: I guess you ran into the issue that the `cryptography` build failed as `cargo` did not find the matching Rust compiler, looking for an `aarch64` one even when using a 32-bit userland (with 64-bit kernel)? This happens on RPi 4 at least. In this case, one can also enforce a 32-bit host triple when installing rustup:
```sh
curl -sSf 'https://sh.rustup.rs/' | bash -s -- -y --profile minimal --default-host 'armv7-unknown-linux-gnueabihf'
```
This way, `cryptography` and other modules with Rust code should compile successfully, even with 64-bit kernel (on 32-bit userland).

But probably there are other issues why userland and kernel bitness must match.

holta, 2023-09-30:
@MichaIng you saved the day, very profound thanks!  This recipe worked:

```
apt install libffi-dev python3-dev
pip install cryptography==41.0.3
pip install ansible-core
```

@bennuttall does this mean that a `cp39` wheel for cryptography-41.0.4 is likely needed at... ?

https://www.piwheels.org/simple/cryptography/

In any case, let's hope this helps others (DietPi has no doubt already figured this out!!) navigating Bullseye-to-Bookworm over coming weeks :pray:

MichaIng, 2023-09-30:
Ah, I was not looking carefully, there is nothing compiled which requires a Rust compiler, `setuptools-rust` and `cryptography` are downloaded as wheels. Then the reason for enforcing the 32-bit kernel seems to be a different one.

On piwheels, recent `cffi` wheels are still missing, which is the build which failed in OP: https://www.piwheels.org/project/cffi/
The latest v1.16.0 as released just two days ago, and piwheels runners are still busy with Bookworm/Python 3.11 builds, before new upstream releases can be compiled quickly: https://github.com/piwheels/packages/issues/388

holta, 2023-10-01:
Above [workaround](https://github.com/iiab/iiab/issues/3650#issuecomment-1741882403) (thanks to @MichaIng) is now merged/mainlined:

- PR #3651

closed 2023-10-01 by holta

bennuttall, 2023-10-01:
We've been dedicating piwheels builders to the cp311 backlog and not building new releases on 37 and 39 for a couple of weeks. Normal service will resume in the next few days and we'll begin to catch up on those.

If you depend on wheels from piwheels, I'd suggest adding `--prefer-binary` or `--only-binary` to your pip commands. That will ensure that you get a wheel, even if it's not the very latest release.

holta, 2023-10-01:
> `--prefer-binary` or `--only-binary`

> will ensure that you get a wheel, even if it's not the very latest release

Huge thanks @bennuttall for the tip. 💯 

Just as confirmation for others/all — here's the output on fully 32-bit Raspberry Pi OS Bullseye, showing that `--prefer-binary` is a simple/fast/reliable a.k.a. beautiful solution:

```
root@raspberrypi:~# cat /etc/pip.conf
[global]
extra-index-url=https://www.piwheels.org/simple
```
```
root@raspberrypi:~# uname -a
Linux raspberrypi 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux
```
```
root@raspberrypi:~# cat /etc/debian_version
11.7
```
```
root@raspberrypi:~# python3 -m venv /usr/local/ansible    # CREATE VIRTUAL ENV
```
```
root@raspberrypi:~# /usr/local/ansible/bin/python3 -m pip install --prefer-binary --upgrade ansible-core                                                                                                                                     Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ansible-core
  Downloading https://www.piwheels.org/simple/ansible-core/ansible_core-2.15.4-py3-none-any.whl (2.2 MB)
     |████████████████████████████████| 2.2 MB 430 kB/s
Collecting cryptography
  Downloading https://www.piwheels.org/simple/cryptography/cryptography-41.0.3-cp39-cp39-linux_armv7l.whl (2.1 MB)
     |████████████████████████████████| 2.1 MB 404 kB/s
Collecting jinja2>=3.0.0
  Downloading https://www.piwheels.org/simple/jinja2/Jinja2-3.1.2-py3-none-any.whl (133 kB)
     |████████████████████████████████| 133 kB 14.6 MB/s
Collecting packaging
  Downloading packaging-23.2-py3-none-any.whl (53 kB)
     |████████████████████████████████| 53 kB 899 kB/s
Collecting importlib-resources<5.1,>=5.0
  Downloading https://www.piwheels.org/simple/importlib-resources/importlib_resources-5.0.7-py3-none-any.whl (24 kB)
Collecting resolvelib<1.1.0,>=0.5.3
  Downloading https://www.piwheels.org/simple/resolvelib/resolvelib-1.0.1-py2.py3-none-any.whl (17 kB)
Collecting PyYAML>=5.1
  Downloading https://www.piwheels.org/simple/pyyaml/PyYAML-6.0.1-cp39-cp39-linux_armv7l.whl (45 kB)
     |████████████████████████████████| 45 kB 1.6 MB/s
Collecting MarkupSafe>=2.0
  Downloading https://www.piwheels.org/simple/markupsafe/MarkupSafe-2.1.3-cp39-cp39-linux_armv7l.whl (23 kB)
Collecting cffi>=1.12
  Downloading https://www.piwheels.org/simple/cffi/cffi-1.15.1-cp39-cp39-linux_armv7l.whl (364 kB)
     |████████████████████████████████| 364 kB 407 kB/s
Collecting pycparser
  Downloading https://www.piwheels.org/simple/pycparser/pycparser-2.21-py2.py3-none-any.whl (119 kB)
     |████████████████████████████████| 119 kB 16.9 MB/s
Installing collected packages: pycparser, MarkupSafe, cffi, resolvelib, PyYAML, packaging, jinja2, importlib-resources, cryptography, ansible-core
Successfully installed MarkupSafe-2.1.3 PyYAML-6.0.1 ansible-core-2.15.4 cffi-1.15.1 cryptography-41.0.3 importlib-resources-5.0.7 jinja2-3.1.2 packaging-23.2 pycparser-2.21 resolvelib-1.0.1
```
```
root@raspberrypi:~# /usr/local/ansible/bin/ansible --version
ansible [core 2.15.4]
  config file = None
  configured module search path = ['/root/.ansible/plugins/modules', '/usr/share/ansible/plugins/modules']
  ansible python module location = /usr/local/ansible/lib/python3.9/site-packages/ansible
  ansible collection location = /root/.ansible/collections:/usr/share/ansible/collections
  executable location = /usr/local/ansible/bin/ansible
  python version = 3.9.2 (default, Mar 12 2021, 04:06:34) [GCC 10.2.1 20210110] (/usr/local/ansible/bin/python3)
  jinja version = 3.1.2
  libyaml = False
```