iiab/iiab #3673: Asterisk fails to load on Debian 11 as before:

EMG70, 2023-11-19:
### Expected behavior and actual behavior:
Full installation with Freepbx installed add enabled
[image]
 
### Steps to reproduce the problem:
Retried several times.
### IIAB version, OS version, hardware details:
[image]

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
http://sprunge.us/4K56PQ?en

holta, 2023-11-19:
Possibly IIAB's Asterisk patch for Raspberry Pi is no longer necessary?  (Or needs a quick fix?)

Certainly FreePBX 17 pre-releases will run on Debian 12 very shortly — according to these recent announcements:

- #3556

holta, 2023-11-19:
> Possibly IIAB's Asterisk patch for Raspberry Pi is no longer necessary? (Or needs a quick fix?)

Here's `install_prereq.diff` used with `when: asterisk_rpi_patch` that needs refining (or replacing?)

https://github.com/iiab/iiab/blob/acf125f46d3eff3ac7550faeb3d00711d73131a6/roles/pbx/files/install_prereq.diff#L1-L16

holta, 2023-11-19:
Workaround for Debian 11 on non-RPi for now:

Set `asterisk_rpi_patch: False` in /etc/iiab/local_vars.yml

EMG70, 2023-11-19:
Installed & Tested OK with asterisk_rpi_patch :False

holta, 2023-11-20:
It appears `roles/pbx/files/intall_prereq.diff` is no longer relevant, as a result of...

> install_prereq: Fix dependency install on aarch64.
Fixes dependency solutions in install_prereq for Debian aarch64
platforms. install_prereq was attempting to forcibly install 32-bit
armhf packages due to the aptitude search for dependencies.

...merged on 2023-08-14 here: https://github.com/asterisk/asterisk/commit/6a55551c5cceb01e464a4defcb836684e7709920
- https://github.com/asterisk/asterisk/pull/38
- https://github.com/asterisk/asterisk/commits/master/contrib/scripts/install_prereq

...and now part of Asterisk 20 and 21, resolving the original issue:

- https://github.com/asterisk/asterisk/issues/37

holta, 2023-11-20:
Should now be resolved by:

- PR #3674

Thanks @EMG70 for the bug report!

closed 2023-11-20 by holta