iiab/iiab #3682: rpi w2 various glitches

tim-moody, 2023-12-10:
Did more or less medium install on rpi 4 with 2023-10-05 raspian and added 256G of content. I then put this sd card in a rpi w2 and booted. It seemed to boot fine and got to a linux prompt shown on the monitor. There were a number of glitches.

tim-moody, 2023-12-10:
1) hostapd failed to start:

Nov 28 16:50:08 box systemd[1]: Starting hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator...
Nov 28 16:50:08 box hostapd[615]: rfkill: WLAN soft blocked
Nov 28 16:50:08 box hostapd[615]: ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Nov 28 16:50:08 box ip[627]: RTNETLINK answers: Operation not possible due to RF-kill
Nov 28 16:50:08 box systemd[1]: Started hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
Nov 28 16:50:09 box systemd[1]: hostapd.service: Deactivated successfully.

I was surprised at this as hostapd had been running on the rpi 4

After a restart hostapd ran successfully and I was able to connect to it with my laptop. Not sure how to automate this so that a card can be prepared on a bigger rpi and deployed on w2.

tim-moody, 2023-12-10:
2) Kiwix failed to start:

Dec 10 15:13:50 box systemd[1]: Starting kiwix-serve.service - Provides the Kiwix ZIM File Server...
Dec 10 15:13:50 box (ix-serve)[680]: kiwix-serve.service: Failed to execute /opt/iiab/kiwix/bin/kiwix-serve: Exec format error
Dec 10 15:13:50 box (ix-serve)[680]: kiwix-serve.service: Failed at step EXEC spawning /opt/iiab/kiwix/bin/kiwix-serve: Exec format e>
Dec 10 15:13:51 box systemd[1]: kiwix-serve.service: Control process exited, code=exited, status=203/EXEC
Dec 10 15:13:51 box systemd[1]: kiwix-serve.service: Failed with result 'exit-code'.
Dec 10 15:13:51 box systemd[1]: Failed to start kiwix-serve.service - Provides the Kiwix ZIM File Server.

Again, Kiwix was running on the rpi 4. OS is 64 bit and w2 is also 64 bit, so I expected it to work, but perhaps it is the arm level.

root@box:~# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
root@box:~# uname -a
Linux box 6.1.0-rpi4-rpi-v7 #1 SMP Raspbian 1:6.1.54-1+rpt2 (2023-10-05) armv7l GNU/Linux

tim-moody, 2023-12-10:
root@box:~# getconf LONG_BIT
32

I assumed that running the 64 bit raspios would come up in 64, not 32 bit.

tim-moody, 2023-12-10:
put the card back into the rpi 4 and

root@box:~# uname -a
Linux box 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux

kiwix runs.

root@box:~# getconf LONG_BIT
32

another surprise

checked kiwix download version: kiwix-tools_linux-aarch64-musl-3.5.0-2.tar.gz

tim-moody, 2023-12-10:
starting again with 12-5-2023 raspios and small install
on rpi 4 both kiwix and hostapd are working

tim-moody, 2023-12-10:
now in rpi w2:
root@box:~# uname -a
Linux box 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

```
root@box:~# systemctl status hostapd
○ hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: enabled)
     Active: inactive (dead) since Sun 2023-12-10 18:23:30 GMT; 1min 55s ago
   Duration: 68ms
    Process: 638 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid /etc/hostapd/hostapd.conf (code=exited, status=0/SUCCESS)
    Process: 644 ExecStartPost=/sbin/ip link set ap0 up (code=exited, status=2)
   Main PID: 643 (code=exited, status=0/SUCCESS)
        CPU: 71ms

Dec 10 18:23:29 box systemd[1]: Starting hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator...
Dec 10 18:23:29 box hostapd[638]: rfkill: WLAN soft blocked
Dec 10 18:23:29 box hostapd[638]: ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Dec 10 18:23:30 box ip[644]: RTNETLINK answers: Operation not possible due to RF-kill
Dec 10 18:23:30 box systemd[1]: Started hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
Dec 10 18:23:30 box systemd[1]: hostapd.service: Deactivated successfully.
```

```
root@box:~# rfkill list
0: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
root@box:~# cat /etc/hostapd/hostapd.conf
# Basic configuration

interface=ap0
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

ssid=Internet in a Box
channel=6
bridge=br0

# Hardware configuration
driver=nl80211
country_code=US
# limit emissions to what is legal in country
ieee80211d=1
hw_mode=g

```
```
root@box:~# systemctl restart hostapd
root@box:~# systemctl status hostapd
● hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: enabled)
     Active: active (running) since Sun 2023-12-10 18:32:10 GMT; 6s ago
    Process: 1094 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid /etc/hostapd/hostapd.conf (code=exited, status=0/SUCCESS)
    Process: 1096 ExecStartPost=/sbin/ip link set ap0 up (code=exited, status=0/SUCCESS)
   Main PID: 1095 (hostapd)
      Tasks: 1 (limit: 176)
        CPU: 64ms
     CGroup: /system.slice/hostapd.service
             └─1095 /usr/sbin/hostapd -B -P /run/hostapd.pid /etc/hostapd/hostapd.conf

Dec 10 18:32:06 box systemd[1]: Starting hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator...
Dec 10 18:32:06 box hostapd[1094]: ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Dec 10 18:32:07 box hostapd[1094]: ap0: interface state COUNTRY_UPDATE->ENABLED
Dec 10 18:32:07 box hostapd[1094]: ap0: AP-ENABLED
Dec 10 18:32:10 box systemd[1]: Started hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
```

tim-moody, 2023-12-10:
kiwix worked, presumably because the OS is now 64 bit

jvonau, 2023-12-10:
This seems like a boot time race between hostapd and whatever is invoking rfkill to enable the wifi. Is the problem gone on subsequent boots? Check the logs of sequence of events.

tim-moody, 2023-12-10:
it does not happen on the rpi 4. It only happens on the first book of the w2 with the sd card that was previously in the rpi 4

tim-moody, 2023-12-10:
I'm not really seeing anything in the logs, but I also don't know what to look for.

holta, 2023-12-11:
> Did more or less medium install on rpi 4 with 2023-10-05 raspian and added 256G of content. I then put this sd card in a rpi w2 and booted. It seemed to boot fine and got to a linux prompt shown on the monitor. There were a number of glitches.

Are you installing the "Bookworm" version 64-bit Raspberry Pi OS Lite ?

There are contradictory claims circulating — that "Bookworm" is or was unfortunately not fully supported on Raspberry Pi Zero 2 W hardware. 🤔

_In Short: Answering all these questions will be extremely valuable, if in the end we can understand a bit better what aspects are not workable._

tim-moody, 2023-12-11:
>Are you installing the "Bookworm" version of 64-bit Raspberry Pi OS Lite ?

yes

In my experience 64 bit 12/5/23 properly boots as 64 bit on w2 whereas 64 bit previous version booted as 32 bit on w2 as it does on w 0

The rfkill problem has a work around in that you can simply restart hostapd, but that makes for complications.

holta, 2023-12-11:
Huge thanks @tim-moody for researching this critical question 💯 

(I wish I could help but this coming week is extremely difficult for me.)

tim-moody, 2023-12-17:
I repeated the install on 12-5-2023 with the same result that an install on rpi 4 with working hostapd when the card is moved to w2 has hostapd killed by rfkill. But after systemctl restart hostapd works fine.

tim-moody, 2023-12-17:
I then repeated with 12-11-2023 with the same result.

For any who are interested here are some logs:

http://sprunge.us/6Cv8RM entire log from install on rpi4

http://sprunge.us/GVveEQ log from boot on w2

http://sprunge.us/Q7bd4H hostapd references on w2

http://sprunge.us/9CJ3kQ kill references on w2

jvonau, 2023-12-19:
I'll note that NetworkManager is active and looks to be the new out of the box default on this image. Think the easiest with 'systemd-rfkill.service' in the logs is to add to the both versions of hostapd.service file `Requires=systemd-rfkill.service`

jvonau, 2023-12-19:
Perhaps Before=NetworkManager.service might become After= as NM is the process unlocking the WiFi