iiab/iiab #3686: rpi w2 and alternative wifi firmware

tim-moody, 2023-12-18:
Environment:
- rpi w2
- 12/11/2023 64 bit raspios lite
- wifi_up_down: True
- rpi3bplus_rpi4_wifi_firmware: 19
- rpizerow_rpi3_wifi_firmware: os

dmesg | grep Firmware:
`[   14.091548] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7`

Note that this firmware not not correspond with any strings output in /lib/firmware/cypress/:

```
-rw-r--r-- 1 root root 397444 Dec 16 19:00 ../cypress/brcmfmac43430-sdio.bin_2018-09-11_7.45.98.65
-rw-r--r-- 1 root root   7222 Dec 16 19:00 ../cypress/brcmfmac43430-sdio.clm_blob_2018-09-11_7.45.98.65
-rw-r--r-- 1 root root 399344 Nov 15 14:19 ../cypress/cyfmac43430-sdio.bin
lrwxrwxrwx 1 root root     25 Dec 16 19:00 ../cypress/cyfmac43430-sdio.bin.iiab -> cyfmac43430-sdio.bin.orig
-rw-r--r-- 1 root root 399344 Nov 15 14:19 ../cypress/cyfmac43430-sdio.bin.orig
-rw-r--r-- 1 root root   4733 Nov 15 14:19 ../cypress/cyfmac43430-sdio.clm_blob
lrwxrwxrwx 1 root root     30 Dec 16 19:00 ../cypress/cyfmac43430-sdio.clm_blob.iiab -> cyfmac43430-sdio.clm_blob.orig
-rw-r--r-- 1 root root   4733 Nov 15 14:19 ../cypress/cyfmac43430-sdio.clm_blob.orig
```

But it does correspond with /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436s-sdio.bin
(and not with brcmfmac43430b0-sdio.raspberrypi,model-zero-2-w.bin -> brcmfmac43436-sdio.bin)

That suggests and new pattern for firmware deployment. I'm not sure if it is statically compiled or what tells the kernel to load that one.

tim-moody, 2023-12-18:
I ran my standard wifi connection test with 11 possible connections.

I got 9, presumably because wifi_up_down was True. The other day I got 10 connections with that False.

holta, 2023-12-18:
> firmware [doesn't] correspond with any strings output in /lib/firmware/cypress/

Very strange indeed!

1. Over the past 2 years `dmesg | grep Firmware:` has always matched the version shown by...

   ```
   strings /lib/cypress/cyfmac43430-sdio.bin | grep Version
   ```

   ...with Zero W, RPi 3 and presumably also Zero 2 W.

2. Whereas with RPi 3 B+ and RPi 4, `dmesg | grep Firmware:` has always matched the version shown by...

   ```
   strings /lib/cypress/cyfmac43455-sdio.bin | grep Version
   ```

tim-moody, 2023-12-19:
FWIW
- set rpizerow_rpi3_wifi_firmware: 30 in local vars per https://github.com/iiab/iiab/pull/3103
- did ./runrole firmware && sudo reboot

no change in output of dmesg | grep Firmware:

holta, 2023-12-19:
> did ./runrole firmware && sudo reboot

Do you mean `./runrole --reinstall firmware` then reboot?

tim-moody, 2023-12-19:
no. followed https://github.com/iiab/iiab/pull/3103

tim-moody, 2023-12-19:
reran with ./runrole --reinstall firmware
no change

```
root@box:~# dmesg | grep Firmware:
[   13.138203] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7

```

holta, 2023-12-19:
Thanks for having confirmed that!