iiab/iiab #3707: IIAB Install fails  "https://download.iiab.io/packages/iwlwifi-8000C-13.ucode "response": "HTTP Error 404

shanti-bhardwa, 2024-02-09:
IIAB 8.3 install fails with following error:

TASK [1-prep : include_tasks] ***************************************************************************************************************************************************************************************************************
skipping: [127.0.0.1]

TASK [1-prep : Check if the identifier for Intel's NUC6 built-in WiFi is present] ***********************************************************************************************************************************************************
changed: [127.0.0.1]

TASK [1-prep : Download https://download.iiab.io/packages/iwlwifi-8000C-13.ucode to /lib/firmware for built-in WiFi on NUC6] ****************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "dest": "/lib/firmware", "elapsed": 0, "gid": 0, "group": "root", "mode": "0755", "msg": "Request failed", "owner": "root", "response": "HTTP Error 404: Not Found", "size": 69632, "state": "directory", "status_code": 404, "uid": 0, "url": "https://download.iiab.io/packages/iwlwifi-8000C-13.ucode"}

PLAY RECAP **********************************************************************************************************************************************************************************************************************************
127.0.0.1                  : ok=74   changed=12   unreachable=0    failed=1    skipped=25   rescued=0    ignored=1   ault

Hardware is Lenovo T460 Laptop running Linux Mint 21.2

Installing with local_vars.yml with Small build and all values set up as default.

Checking on WIFI interface, It is Intel 8260 pci card.

[    5.354016] ee1004 0-0051: 512 byte EE1004-compliant SPD EEPROM, read-only
[    5.372892] Intel(R) Wireless WiFi driver for Linux
[    5.373261] iwlwifi 0000:04:00.0: enabling device (0000 -> 0002)
[    5.380205] iwlwifi 0000:04:00.0: loaded firmware version 36.ca7b901d.0 8000C-36.ucode op_mode iwlmvm
[    5.415297] iwlwifi 0000:04:00.0: Detected Intel(R) Dual Band Wireless AC 8260, REV=0x204

lshw -C network gives the following

lshw -C network
WARNING: you should run this program as super-user.
  *-network                 
       description: Wireless interface
       product: Wireless 8260
       vendor: Intel Corporation
       physical id: 0
       bus info: pci@0000:04:00.0
       logical name: wlp4s0
       version: 3a
       serial: f0:d5:bf:72:4d:d1
       width: 64 bits
       clock: 33MHz
       capabilities: bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=iwlwifi driverversion=5.15.0-92-generic firmware=36.ca7b901d.0 8000C-36.ucode latency=0 link=no multicast=yes wireless=IEEE 802.11
       resources: irq:132 memory:f1100000-f1101fff

looks like Iike IIAB Install is trying download the following, which does not exist.

https://download.iiab.io/packages/iwlwifi-8000C-13.ucode

when the driver is already

 loaded firmware version 36.ca7b901d.0 8000C-36.ucode

Can't do pastebin as the install has not completed

holta, 2024-02-09:
Thanks @shanti-bhardwa

I suspect it's time to comment out this WiFi special case about 6 years later:

https://github.com/iiab/iiab/blob/7d73fca112caca33715140db746db5a341a118e4/roles/1-prep/tasks/hardware.yml#L10-L20

(Unless others confirm it's definitely still needed?!)

holta, 2024-02-09:
FYI this old driver is here if anybody really needs it:

https://download.iiab.io/packages/Old/iwlwifi-8000C-13.ucode

holta, 2024-02-09:
@shanti-bhardwa can we close this now?

(If you've tested that PR #3708 is merged?)

andrewclark5, 2024-02-09:
Adam has suggested commenting something out.

Is that to be done in the install script?  I am happy to test it on a clean build of Mint 21.2 if someone tells me how to do that.

Regards, Andrew

holta, 2024-02-09:
@andrewclark5 best to test on the identical hardware that @shanti-bhardwa used if possible?

CLARIF: A fresh install itf IIAB's latest is all that needs testing, now that PR #3708 is merged!

andrewclark5, 2024-02-09:
Just tested iiab clean install on an almost identical Lenovo laptop to the
one it originally failed on.

The script ran through perfectly, thanks guys!

Regards, Andrew

closed 2024-02-09 by holta