iiab/iiab #3709: Recent Kolibri installs show "502 Bad Gateway"

holta, 2024-02-10:
@nzola wrote:

> Kolibri gives error "502 Bad Gateway" when trying to access it on the browser?
 I did 3 iiab install on rpi4. All of them give the same error

@nzola trying a recent Beta release of Kolibri 0.16 will probably work around this?

https://github.com/learningequality/kolibri/releases

In other words, try setting the following in your [/etc/iiab/local_vars.yml](https://wiki.iiab.io/go/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it%3F) prior to installing IIAB:

```
kolibri_deb_url: https://github.com/learningequality/kolibri/releases/download/v0.16.0-beta13/kolibri_0.16.0b13-0ubuntu1_all.deb
```

jvonau, 2024-02-11:
> 2024-02-10 17:46:35,213 p=44617 u=root n=ansible | TASK [kolibri : apt install https://github.com/learningequality/kolibri/releases/download/v0.16.0-beta13/kolibri_0.16.0b13-0ubuntu1_all.deb (if kolibri_deb_url IS defined)] ***
2024-02-10 17:46:48,968 p=44617 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Conflicts with the installed package 'python3-markdown'"}
2024-02-10 17:46:49,014 p=44617 u=root n=ansible | TASK [kolibri : SEE ERROR ABOVE (skip_role_on_error: False)]

jvonau, 2024-02-11:
control file

> Package: kolibri
Source: kolibri-source
Version: 0.16.0b13-0ubuntu1
Architecture: all
Maintainer: José L. Redrejo Rodríguez <jredrejo@debian.org>
Installed-Size: 219964
Depends: python3 (>= 3.6), python3-pkg-resources, adduser
Recommends: python3-cryptography (>= 1.2.3)
Conflicts: python3-markdown (>= 3.4.1)
Section: education
Priority: optional
Homepage: https://learningequality.org/kolibri
Description: The offline app for universal education
 An educational platform for learners of all ages.
 .
 Retrieve and Distribute Open Educational Resources (OER) from online
 resources or from device to device.
 .
 Resources are shared as 'channels', amongst which can be found
 Khan Academy, PhET and CK-12.
 .
 To find more channels and create your own, please visit
 our website: http://studio.learningequality.org/

jvonau, 2024-02-11:
https://github.com/iiab/iiab/issues/3343#issuecomment-1232043584

jvonau, 2024-02-11:
python3-markdown was probably pulled in by samba need to run `apt info python3-markdown` on the problem OS 
> apt info python3-markdown
Package: python3-markdown
Version: 3.3.6-1
Priority: optional
Section: python
Source: python-markdown
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian Python Team <team+python@tracker.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 338 kB
Depends: python3-importlib-metadata (>= 4.4), python3:any
Recommends: python3-pygments, python3-yaml
Suggests: python-markdown-doc
Breaks: python-markdown (<< 3.1.1-4), python3-mdx-math (<< 0.7-2~)
Replaces: python-markdown (<< 3.1.1-4)
Homepage: https://github.com/Python-Markdown/markdown
Task: samba-server, ubuntustudio-publishing, ubuntukylin-desktop
Download-Size: 68.5 kB
APT-Sources: http://ca.archive.ubuntu.com/ubuntu jammy/main amd64 Packages
Description: text-to-HTML conversion library/tool (Python 3 version)
 Markdown is a text-to-HTML conversion tool for web writers. Markdown
 allows you to write using an easy-to-read, easy-to-write plain text
 format, then convert it to structurally valid XHTML (or HTML).
 .
 This is a Python implementation of John Gruber's Markdown. The current
 version of python-markdown implements all Markdown syntax features and
 fully passes Markdown Test Suite 1.0. It also supports footnotes and
 attributes.
 .
 This package contains all the files needed to use Markdown with Python
 3.x.

holta, 2024-02-11:
Thanks @nzola for your logs posted here:

https://sprunge.us/MObVH6?en

Thanks @jvonau for explaining the apparent source of the conflict.

tim-moody, 2024-02-11:
@jvonau, a masterful dissection of the problem

nzola, 2024-02-12:
@jvonau @holta Just finished the new install with Samba_Install: False including kolibri 0.16 Beta 13 line, but Kolibri still give 502 error as before.  http://sprunge.us/iZ0t7R

tim-moody, 2024-02-12:
root@box:~# systemctl status kolibri
× kolibri.service - Kolibri
     Loaded: loaded (/etc/systemd/system/kolibri.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Mon 2024-02-12 12:08:35 EST; 48s ago
    Process: 553 ExecStart=/usr/bin/kolibri start (code=exited, status=203/EXEC)
        CPU: 1ms

Feb 12 12:08:35 box systemd[1]: Starting kolibri.service - Kolibri...
Feb 12 12:08:35 box (kolibri)[553]: kolibri.service: Failed to locate executable /usr/bin/kolibri: No such file or directory
Feb 12 12:08:35 box (kolibri)[553]: kolibri.service: Failed at step EXEC spawning /usr/bin/kolibri: No such file or directory
Feb 12 12:08:35 box systemd[1]: kolibri.service: Control process exited, code=exited, status=203/EXEC
Feb 12 12:08:35 box systemd[1]: kolibri.service: Failed with result 'exit-code'.
Feb 12 12:08:35 box systemd[1]: Failed to start kolibri.service - Kolibri.

tim-moody, 2024-02-12:
root@box:/opt/iiab/iiab# grep kolibri iiab-install.log

```
2024-02-12 11:39:19,960 p=914 u=root n=ansible | ok: [127.0.0.1] => (item=kolibri)
2024-02-12 11:39:20,285 p=914 u=root n=ansible | ok: [127.0.0.1] => (item=kolibri)
2024-02-12 11:39:20,585 p=914 u=root n=ansible | ok: [127.0.0.1] => (item=kolibri)
2024-02-12 11:39:20,886 p=914 u=root n=ansible | ok: [127.0.0.1] => (item=kolibri)
2024-02-12 11:39:21,168 p=914 u=root n=ansible | ok: [127.0.0.1] => (item=kolibri)
2024-02-12 11:39:21,480 p=914 u=root n=ansible | ok: [127.0.0.1] => (item=kolibri)
2024-02-12 11:41:48,021 p=914 u=root n=ansible | changed: [127.0.0.1] => (item={'option': 'usb_lib_umask0000_for_kolibri', 'value': True})
2024-02-12 11:43:26,279 p=914 u=root n=ansible | TASK [kolibri : Assert that "kolibri_install is sameas true" (boolean not string etc)] ***********************************************
2024-02-12 11:43:26,310 p=914 u=root n=ansible | TASK [kolibri : Assert that "kolibri_enabled | type_debug == 'bool'" (boolean not string etc)] ***************************************
2024-02-12 11:43:26,339 p=914 u=root n=ansible | TASK [kolibri : Install Kolibri, if 'kolibri_installed' not defined, e.g. in /etc/iiab/iiab_state.yml] *******************************
2024-02-12 11:43:26,383 p=914 u=root n=ansible | included: /opt/iiab/iiab/roles/kolibri/tasks/install.yml for 127.0.0.1
2024-02-12 11:43:26,397 p=914 u=root n=ansible | TASK [kolibri : Record (initial) disk space used] ************************************************************************************
2024-02-12 11:43:26,559 p=914 u=root n=ansible | TASK [kolibri : Create Linux user kolibri and add it to groups www-data, disk] *******************************************************
2024-02-12 11:43:26,926 p=914 u=root n=ansible | TASK [kolibri : Create directory /library/kolibri for Kolibri content, configuration, sqlite3 databases (kolibri:www-data, by default 0755)] ***
2024-02-12 11:43:27,092 p=914 u=root n=ansible | TASK [kolibri : Create directory /etc/kolibri (by default 0755)] *********************************************************************
2024-02-12 11:43:27,253 p=914 u=root n=ansible | TASK [kolibri : Save kolibri_user (kolibri) to /etc/kolibri/username] ****************************************************************
2024-02-12 11:43:27,535 p=914 u=root n=ansible | TASK [kolibri : Save kolibri_home (KOLIBRI_HOME="/library/kolibri") to /etc/kolibri/daemon.conf] *************************************
2024-02-12 11:43:27,816 p=914 u=root n=ansible | TASK [kolibri : Download Kolibri's apt key to /usr/share/keyrings/learningequality-kolibri.gpg] **************************************
2024-02-12 11:43:28,366 p=914 u=root n=ansible | TASK [kolibri : Add signed Kolibri PPA 'jammy' (if Ubuntu 22.04+ or Mint 21 or Debian 12)] *******************************************
2024-02-12 11:43:30,235 p=914 u=root n=ansible | TASK [kolibri : Add signed Kolibri PPA 'focal' (if other/older OS's)] ****************************************************************
2024-02-12 11:43:30,267 p=914 u=root n=ansible | TASK [kolibri : apt install kolibri (using apt source specified above, if kolibri_deb_url ISN'T defined)] ****************************
2024-02-12 11:43:30,298 p=914 u=root n=ansible | TASK [kolibri : apt install https://github.com/learningequality/kolibri/releases/download/v0.16.0-beta13/kolibri_0.16.0b13-0ubuntu1_all.deb (if kolibri_deb_url IS defined)] ***
2024-02-12 11:44:09,262 p=914 u=root n=ansible | TASK [kolibri : Run 'rm -rf /root/.kolibri' to remove "unavoidable" pollution created above] *****************************************
2024-02-12 11:44:09,426 p=914 u=root n=ansible | TASK [kolibri : Install from template: /etc/systemd/system/kolibri.service] **********************************************************
2024-02-12 11:44:09,720 p=914 u=root n=ansible | TASK [kolibri : Stop 'kolibri' systemd service, for Kolibri provisioning (after daemon_reload)] **************************************
2024-02-12 11:44:10,342 p=914 u=root n=ansible | TASK [kolibri : Provision Kolibri, while setting: facility name, admin acnt / password, preset type, and language] *******************
2024-02-12 11:44:18,750 p=914 u=root n=ansible | TASK [kolibri : Record (final) disk space used] **************************************************************************************
2024-02-12 11:44:18,914 p=914 u=root n=ansible | TASK [kolibri : Add 'kolibri_disk_usage = 257888256' to /etc/iiab/iiab.ini] **********************************************************
2024-02-12 11:44:19,077 p=914 u=root n=ansible | TASK [kolibri : Set 'kolibri_installed: True'] ***************************************************************************************
2024-02-12 11:44:19,102 p=914 u=root n=ansible | TASK [kolibri : Add 'kolibri_installed: True' to /etc/iiab/iiab_state.yml] ***********************************************************
2024-02-12 11:44:19,262 p=914 u=root n=ansible | TASK [kolibri : include_tasks] *******************************************************************************************************
2024-02-12 11:44:19,320 p=914 u=root n=ansible | included: /opt/iiab/iiab/roles/kolibri/tasks/enable-or-disable.yml for 127.0.0.1
2024-02-12 11:44:19,342 p=914 u=root n=ansible | TASK [kolibri : Enable & Start 'kolibri' systemd service, if kolibri_enabled] ********************************************************
[... cut ...]
2024-02-12 11:44:24,014 p=914 u=root n=ansible | changed: [127.0.0.1] => (item={'option': 'kolibri_enabled', 'value': True})
2024-02-12 11:44:24,158 p=914 u=root n=ansible | changed: [127.0.0.1] => (item={'option': 'kolibri_url', 'value': '/kolibri/'})
2024-02-12 11:44:24,295 p=914 u=root n=ansible | changed: [127.0.0.1] => (item={'option': 'kolibri_path', 'value': '/usr/bin/kolibri'})
2024-02-12 11:44:24,432 p=914 u=root n=ansible | changed: [127.0.0.1] => (item={'option': 'kolibri_port', 'value': 8009})

```

holta, 2024-02-12:
Wow.

Looks like Kolibri 0.16 Beta 13 has likely regressed (or changed dramatically since December Betas!)

(I'm assuming the earlier versions of Kolibri continue to work, as they did back in December...)

tim-moody, 2024-02-12:
is it possibly that things are now to be in a venv?

nzola, 2024-02-12:
> Wow.
> 
> Looks like Kolibri 0.16 Beta 13 has likely regressed (or changed dramatically since December Betas!?)
> 
> (I'm assuming the earlier versions of Kolibri continue to work, as they did back in December...)

Should we revert to old version until the newer is fixed?

tim-moody, 2024-02-12:
root@box:/library/kolibri# apt install https://github.com/learningequality/kolibri/releases/download/v0.16.0-beta13/kolibri_0.16.0b13-0ubuntu1_all.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package https://github.com/learningequality/kolibri/releases/download/v0.16.0-beta13

holta, 2024-02-12:
> root@box:/library/kolibri# apt install https://github.com/learningequality/kolibri/releases/download/v0.16.0-beta13/kolibri_0.16.0b13-0ubuntu1_all.deb

CLARIF: Above command is not supposed to work, and never did.  (It's merely shorthand in a Ansible comment hinting at what the code below does...)

https://github.com/iiab/iiab/blob/7e8873544588628e2deac5c9f806ef68b5acfad8/roles/kolibri/tasks/install.yml#L132-L135

holta, 2024-02-12:
Mysteriously, one can definitely upgrade from 0.16.0 Beta 10 to Beta 11 to Beta 12 to Beta 13... by running couplets like the following:

```
wget https://github.com/learningequality/kolibri/releases/download/v0.16.0-beta13/kolibri_0.16.0b13-0ubuntu1_all.deb
apt install kolibri_0.16.0b13-0ubuntu1_all.deb
```

(Above test was done on Debian 12.5 VM.)

holta, 2024-02-12:
> gave this a try on deb12 vm with local vars none and
> 
> kolibri_deb_url: https://github.com/learningequality/kolibri/releases/download/v0.16.0-beta13/kolibri_0.16.0b13-0ubuntu1_all.deb kolibri_install: True kolibri_enabled: True
> 
> samba is false in default vars
> 
> also got 502 error

I cannot reproduce @tim-moody's experience — a fresh install of Kolibri 0.16.0 Beta 13 onto Debian 12.5 (VM) worked for me as can be seen here:

http://sprunge.us/wLCkq3?en

(Tim can you share your iiab-diagnostics, to see what might be different?)

holta, 2024-02-12:
Another successful IIAB install of Kolibri 0.16.0 Beta 13 onto a Debian 12.5 VM:

https://sprunge.us/v4JMRC?en

Hopefully @tim-moody or @nzola can shed more light on the situation, as I cannot reproduce it so far.

(FWIW above install used the following /etc/iiab/local_vars.yml)

```
# turn off defaults
remoteit_install: False
openvpn_install: False
kalite_install: False
kalite_enabled: False
kiwix_install: False
kiwix_enabled: False
osm_vector_maps_install: False
awstats_install: False
awstats_enabled: False
matomo_install: False
matomo_enabled: False
captiveportal_install: False
calibreweb_install: False
calibreweb_enabled: False
kolibri_install: True
kolibri_enabled: True
kolibri_deb_url: https://github.com/learningequality/kolibri/releases/download/v0.16.0-beta13/kolibri_0.16.0b13-0ubuntu1_all.deb
```

holta, 2024-02-12:
@Nzola can you send the link requested earlier?
Please, after running this 1-LINE command...

```
sudo journalctl -u kolibri | pastebinit -b sprunge.us
```

tim-moody, 2024-02-13:
There is one difference. My local vars did not have

calibreweb_install: False
calibreweb_enabled: False

holta, 2024-02-13:
Indeed:

`apt install ffmpeg pipx` causes apt to install `python3-markdown`, no doubt due to FFmpeg not pipx, which very unfortunately causes apt to remove kolibri.

Kolibri's .deb control file very likely needs to be updated.

That is my hope anyway, that Kolibri can find a way tolerate the `python3-markdown` package on modern OS's.

holta, 2024-02-13:
CORRECTION & CLARIF:

1) apt package `pipx` (commonly requested by major installs like in Ansible's official install instructions!) is what forces apt to install `python3-markdown` on Debian 12 and derivatives.

2) Interestingly this does not happen on Ubuntu e.g. the most recent Ubuntu 24.04 pre-release.

3) In any case, apt package `ffmpeg` is definitely not involved here, on both OS's.

jvonau, 2024-02-13:
So what version of markdown is being installed?

jvonau, 2024-02-13:
I'm wondering why pipx is being installed when Kolibri is running in a virtual environment and 3b70f38544f0703606852569f0ab95ea789ac222 removed the access to python3-netifaces and any other system level python3 package...

holta, 2024-02-13:
> So what version of markdown is being installed? anything under 3.4.1 should be ok with Kolibri
> 
> 22.04
> 
> > apt list python3-markdown
> > Listing... Done
> > python3-markdown/jammy,jammy 3.3.6-1 all

On Debian 12 & close derivatives the problem kicks in...

```
# apt list python3-markdown
Listing... Done
python3-markdown/stable 3.4.1-2 all
```

ASIDE: `pipx` is installed and used by roles/calibre-web to simplify its several venvs.  Hopefully Kolibri will be understanding of this.  Given pipx' s growing popularity, including in Ansible's official install instructions (just as one example).

jvonau, 2024-02-13:
Think pipx can be avoided by just using pip to install xklb in the same way as ansible is installed ie into /usr/local with the required links to /usr/bin/ originating from within the new virtual environment in place of using '/root' as with pipx.

holta, 2024-02-13:
Yes, pipx's mechanized approach can be avoided if absolutely necessary.

At the same time Kolibri made a sincere effort in 2022 not to blockade other packages (Samba, even if didn't quite.

So I remain hopeful that they can make real progress tightening up their .deb installer's control file so that common packages on Debian 12 aren't accidentally sabotaged.

holta, 2024-02-24:
Just fyi there appears to be a separate issue with Kolibri 0.16.0 not quite yet installing on emerging OS's with Python 3.12

I'm sure they're on it, and will provide resolution shortly in coming weeks if not days, but I posted here to be sure:

- https://github.com/learningequality/kolibri/issues/11895#issuecomment-1962448679

tim-moody, 2024-02-29:
just did fresh install on dev12 VM using 
`kolibri_deb_url: https://github.com/learningequality/kolibri/releases/download/v0.16.0/kolibri_0.16.0-0ubuntu1_all.deb`
with no problems
but it was the only app installed

holta, 2024-02-29:
> just did fresh install on dev12 VM using `kolibri_deb_url: https://github.com/learningequality/kolibri/releases/download/v0.16.0/kolibri_0.16.0-0ubuntu1_all.deb` with no problems but it was the only app installed

@EMG70 might have confirmed the same (?) on Raspberry Pi OS Bookworm too?

(But he can reconfirm!)

tim-moody, 2024-02-29:
I still need an easy way to invoke kolibri manage. The kalite model is different. The best I can do is as follows. I'm sure others could do better.

kolibri_env.sh
```
#!/bin/bash
export KOLIBRI_HOME=/library/kolibri

```
kolibri_manage_run.sh
```
#!/bin/bash
. /usr/local/bin/kolibri_env.sh
/usr/bin/kolibri manage $@

```
kolmanage.sh
```
#!/bin/bash
su - kolibri -s /bin/bash -c "/usr/local/bin/kolibri_manage_run.sh $*"
```

holta, 2024-02-29:
> I still need an easy way to invoke kolibri manage. The kalite model is different. The best I can do is as follows. I'm sure others could do better.

Any obvious glitches?

(Or does it basically all work so far!)

tim-moody, 2024-02-29:
this works, but there could be a more elegant or bashly way to do it

holta, 2024-02-29:
> FR Khan imported in 1 hour 45 min (16308 resources, 20G)

I've noticed a few resources are often missing after the 1st pass download (when using Kolibri's web interface).

Not the end of the world — but then again super-annoying if a few videos are missing — and this is only noticed after field deployment.

A 2nd pass or 3rd pass (e.g. using Kolibri's sometimes clunky web interface, to re-scan which resources / downloads are missing) usually solves this.

(So presumably very large Khan Academy downloads misbehave in the same when automated using `kolibri manage` ?)

tim-moody, 2024-02-29:
still not ready for prime time. Phet EN downloaded database, but not content

holta, 2024-04-08:
FYI apt-based install of recent versions of Kolibri (e.g. 0.16.1 for now) should be restored in coming hours — thanks to those upstream for getting this back on track:

- PR #3726

nzola, 2024-04-09:
👍

jvonau, 2024-04-30:
This is no longer an issue right? Why is this still open?

holta, 2024-04-30:
Yeah this is almost all working now.

Except for new OS's like Ubuntu 24.04 with Python 3.12, awaiting Kolibri 0.17 Alpha pre-releases in coming weeks.

holta, 2024-06-17:
Just FYI Kolibri 0.16.2 doesn't work with Ubuntu 24.04 LTS as Learning Quality has hoped — so they're looking into this &/or 0.17 alpha pre-releases — allowing for practical options on 24.04:

- https://github.com/learningequality/kolibri/issues/11316#issuecomment-2166391020

closed 2024-07-04 by holta