iiab/iiab #3711: roles/kalite fails to pin setuptools to 44 on emerging OS's (e.g. Ubuntu 24.04) with Python 3.12 — "Wheel for pip for Python 2.7 is unavailable. apt install python2-pip-whl"

holta, 2024-02-24:
I don't know yet if this will be easily fixable or not?

@jvonau might know:

iiab-diagnostics: http://sprunge.us/L2kogP?en

```
root@u24:~# python3 --version
Python 3.12.2
```

```
root@u24:~# cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=24.04
DISTRIB_CODENAME=noble
DISTRIB_DESCRIPTION="Ubuntu Noble Numbat (development branch)"
```

```
root@u24:~# hostnamectl
 Static hostname: box
       Icon name: computer-vm
         Chassis: vm 🖴
      Machine ID: 7a3d1f28a7fd4c97a0f1e93bf8ae69d8
         Boot ID: e711e5e242344d81a0cce0ae2680f4f5
  Virtualization: kvm
Operating System: Ubuntu Noble Numbat (development branch)
          Kernel: Linux 6.6.0-14-generic
    Architecture: x86-64
 Hardware Vendor: QEMU
  Hardware Model: Standard PC _Q35 + ICH9, 2009_
Firmware Version: unknown
   Firmware Date: Wed 2022-02-02
    Firmware Age: 2y 3w 1d
```

Error from /opt/iiab/iiab/iiab-install.log:

> TASK [kalite : Use pip to pin setuptools to 44 in /usr/local/kalite/venv] *******************************************************************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/usr/bin/virtualenv", "-ppython2.7", "/usr/local/kalite/venv"], "msg": "stdout: fail\nTraceback (most recent call last):\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/embed/via_app_data/via_app_data.py\", line 84, in _get\n    result = get_wheel(\n             ^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py\", line 26, in get_wheel\n    wheel = from_bundle(distribution, version, for_py_version, search_dirs, app_data, do_periodic_update, env)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py\", line 13, in from_bundle\n    wheel = load_embed_wheel(app_data, distribution, for_py_version, of_version)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py\", line 33, in load_embed_wheel\n    wheel = get_embed_wheel(distribution, for_py_version)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/embed/__init__.py\", line 77, in get_embed_wheel\n    raise Exception((\nException: Wheel for pip for Python 2.7 is unavailable. apt install python2-pip-whl\nfail\nTraceback (most recent call last):\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/embed/via_app_data/via_app_data.py\", line 84, in _get\n    result = get_wheel(\n             ^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py\", line 30, in get_wheel\n    wheel = download_wheel(\n            ^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py\", line 65, in download_wheel\n    env = pip_wheel_env_run(search_dirs, app_data, env)\n          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py\", line 115, in pip_wheel_env_run\n    wheel = get_wheel(\n            ^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py\", line 26, in get_wheel\n    wheel = from_bundle(distribution, version, for_py_version, search_dirs, app_data, do_periodic_update, env)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py\", line 13, in from_bundle\n    wheel = load_embed_wheel(app_data, distribution, for_py_version, of_version)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py\", line 33, in load_embed_wheel\n    wheel = get_embed_wheel(distribution, for_py_version)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/embed/__init__.py\", line 77, in get_embed_wheel\n    raise Exception((\nException: Wheel for pip for Python 3.12 is unavailable. apt install python3-pip-whl\nException('Wheel for pip for Python 3.12 is unavailable. apt install python3-pip-whl')\nfail\nTraceback (most recent call last):\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/embed/via_app_data/via_app_data.py\", line 84, in _get\n    result = get_wheel(\n             ^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py\", line 26, in get_wheel\n    wheel = from_bundle(distribution, version, for_py_version, search_dirs, app_data, do_periodic_update, env)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py\", line 13, in from_bundle\n    wheel = load_embed_wheel(app_data, distribution, for_py_version, of_version)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py\", line 33, in load_embed_wheel\n    wheel = get_embed_wheel(distribution, for_py_version)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/embed/__init__.py\", line 77, in get_embed_wheel\n    raise Exception((\nException: Wheel for setuptools for Python 2.7 is unavailable. apt install python2-setuptools-whl\nfail\nTraceback (most recent call last):\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/embed/via_app_data/via_app_data.py\", line 84, in _get\n    result = get_wheel(\n             ^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py\", line 30, in get_wheel\n    wheel = download_wheel(\n            ^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py\", line 65, in download_wheel\n    env = pip_wheel_env_run(search_dirs, app_data, env)\n          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py\", line 115, in pip_wheel_env_run\n    wheel = get_wheel(\n            ^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py\", line 26, in get_wheel\n    wheel = from_bundle(distribution, version, for_py_version, search_dirs, app_data, do_periodic_update, env)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py\", line 13, in from_bundle\n    wheel = load_embed_wheel(app_data, distribution, for_py_version, of_version)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py\", line 33, in load_embed_wheel\n    wheel = get_embed_wheel(distribution, for_py_version)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/embed/__init__.py\", line 77, in get_embed_wheel\n    raise Exception((\nException: Wheel for pip for Python 3.12 is unavailable. apt install python3-pip-whl\nException('Wheel for pip for Python 3.12 is unavailable. apt install python3-pip-whl')\nfail\nTraceback (most recent call last):\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/embed/via_app_data/via_app_data.py\", line 84, in _get\n    result = get_wheel(\n             ^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py\", line 26, in get_wheel\n    wheel = from_bundle(distribution, version, for_py_version, search_dirs, app_data, do_periodic_update, env)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py\", line 13, in from_bundle\n    wheel = load_embed_wheel(app_data, distribution, for_py_version, of_version)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py\", line 33, in load_embed_wheel\n    wheel = get_embed_wheel(distribution, for_py_version)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ [... cut ...] t_embed_wheel(distribution, for_py_version)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/usr/lib/python3/dist-packages/virtualenv/seed/wheels/embed/__init__.py\", line 77, in get_embed_wheel\n    raise Exception((\nException: Wheel for pip for Python 3.12 is unavailable. apt install python3-pip-whl\nException('Wheel for pip for Python 3.12 is unavailable. apt install python3-pip-whl')\nRuntimeError: seed failed due to failing to download wheels pip, setuptools, wheel\n"}

holta, 2024-02-24:
Some [Python 2](https://github.com/iiab/iiab/blob/master/scripts/install_python2.sh) clarifications, just for background / context:

```
root@box:~# apt list python3-platformdirs -a
Listing... Done
python3-platformdirs/noble 4.2.0-1 all [upgradable from: 2.5.1-1]
python3-platformdirs/now 2.5.1-1 all [installed,upgradable to: 4.2.0-1]

root@box:~# apt list python3-virtualenv -a
Listing... Done
python3-virtualenv/noble 20.25.0+ds-1 all [upgradable from: 20.13.0+ds-2]
python3-virtualenv/now 20.13.0+ds-2 all [installed,upgradable to: 20.25.0+ds-1]

root@box:~# apt-mark showhold
python3-platformdirs
python3-virtualenv

root@box:~# apt list virtualenv -a
Listing... Done
virtualenv/noble 20.25.0+ds-1 all

root@box:~# apt list python2 -a
Listing... Done
python2/now 2.7.18-3 amd64 [installed,local]
```

holta, 2024-02-24:
Not sure if this is helpful, but FYI just in case:

```
root@box:/opt/iiab/iiab# apt list python2-pip-whl -a
Listing... Done

root@box:/opt/iiab/iiab# apt list python3-pip-whl -a
Listing... Done
python3-pip-whl/noble,now 23.3+dfsg-1 all [installed,automatic]
```

holta, 2024-02-24:
The availability of apt package `python2-pip-whl` might not actually be essential here?

Still, just for the record, here is its official availability on 3 different/recent OS's:

```
root@u22:~# apt list python2-pip-whl -a
Listing... Done
python2-pip-whl/jammy 20.3.4+dfsg-4 all

root@u22:~# cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.4 LTS"
```

```
root@u23:~# apt list python2-pip-whl -a
Listing... Done

root@u23:~# cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=23.10
DISTRIB_CODENAME=mantic
DISTRIB_DESCRIPTION="Ubuntu 23.10"
```

```
root@deb12:~# apt list python2-pip-whl -a
Listing... Done

root@deb12:~# hostnamectl
 Static hostname: deb12
       Icon name: computer-vm
         Chassis: vm 🖴
      Machine ID: 3e2d75df41a5436bac0933661916c39f
         Boot ID: 134dce066999440aa8ed156978eaa346
  Virtualization: kvm
Operating System: Debian GNU/Linux 12 (bookworm)
          Kernel: Linux 6.1.0-18-amd64
    Architecture: x86-64
 Hardware Vendor: QEMU
  Hardware Model: Standard PC _Q35 + ICH9, 2009_
Firmware Version: unknown
```

holta, 2024-02-25:
More readably with indentation preserved, here's the (equivalent) error message on Ubuntu 24.04, obtained by running [the equivalent command](https://github.com/iiab/iiab/blob/3455b6b848a951c82aee4b072c197aa437424db1/roles/kalite/tasks/install.yml#L43-L50) without Ansible:

```
root@box:~# virtualenv -ppython2.7 /usr/local/kalite/venv
fail
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv/seed/embed/via_app_data/via_app_data.py", line 84, in _get
    result = get_wheel(
             ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 26, in get_wheel
    wheel = from_bundle(distribution, version, for_py_version, search_dirs, app_data, do_periodic_update, env)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py", line 13, in from_bundle
    wheel = load_embed_wheel(app_data, distribution, for_py_version, of_version)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py", line 33, in load_embed_wheel
    wheel = get_embed_wheel(distribution, for_py_version)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/embed/__init__.py", line 77, in get_embed_wheel
    raise Exception((
Exception: Wheel for pip for Python 2.7 is unavailable. apt install python2-pip-whl
fail
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv/seed/embed/via_app_data/via_app_data.py", line 84, in _get
    result = get_wheel(
             ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 26, in get_wheel
    wheel = from_bundle(distribution, version, for_py_version, search_dirs, app_data, do_periodic_update, env)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py", line 13, in from_bundle
    wheel = load_embed_wheel(app_data, distribution, for_py_version, of_version)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py", line 33, in load_embed_wheel
    wheel = get_embed_wheel(distribution, for_py_version)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/embed/__init__.py", line 77, in get_embed_wheel
    raise Exception((
Exception: Wheel for wheel for Python 2.7 is unavailable. apt install python2-wheel-whl
fail
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv/seed/embed/via_app_data/via_app_data.py", line 84, in _get
    result = get_wheel(
             ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 30, in get_wheel
    wheel = download_wheel(
            ^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 65, in download_wheel
    env = pip_wheel_env_run(search_dirs, app_data, env)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 115, in pip_wheel_env_run
    wheel = get_wheel(
            ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 26, in get_wheel
    wheel = from_bundle(distribution, version, for_py_version, search_dirs, app_data, do_periodic_update, env)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py", line 13, in from_bundle
    wheel = load_embed_wheel(app_data, distribution, for_py_version, of_version)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py", line 33, in load_embed_wheel
    wheel = get_embed_wheel(distribution, for_py_version)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/embed/__init__.py", line 77, in get_embed_wheel
    raise Exception((
Exception: Wheel for pip for Python 3.12 is unavailable. apt install python3-pip-whl
Exception('Wheel for pip for Python 3.12 is unavailable. apt install python3-pip-whl')
fail
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv/seed/embed/via_app_data/via_app_data.py", line 84, in _get
    result = get_wheel(
             ^^^^^^^^^^
[... 43 lines cut ...]
             ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 30, in get_wheel
    wheel = download_wheel(
            ^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 65, in download_wheel
    env = pip_wheel_env_run(search_dirs, app_data, env)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 115, in pip_wheel_env_run
    wheel = get_wheel(
            ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 26, in get_wheel
[... cut ...]
    wheel = get_embed_wheel(distribution, for_py_version)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/embed/__init__.py", line 77, in get_embed_wheel
    raise Exception((
Exception: Wheel for pip for Python 3.12 is unavailable. apt install python3-pip-whl
Exception('Wheel for pip for Python 3.12 is unavailable. apt install python3-pip-whl')
RuntimeError: seed failed due to failing to download wheels wheel, pip, setuptools
```

jvonau, 2024-02-26:
Yes I understand the issue but ask yourself where python2 originates from and where python2-pip-whl is available.

holta, 2024-02-26:
Line 96 could be inserted in scripts/install_python2.sh

```
apt -y install python2-pip-whl python2-setuptools-whl python3-wheel-whl
```

Leading to some-but-not-all of the needed packages:

```
root@box:~# apt list python2-pip-whl python2-setuptools-whl python3-pip-whl python3-wheel-whl python2-wheel-whl -a                                                                                                      
Listing... Done
python2-pip-whl/now 20.3.4+dfsg-4 all [installed,local]

python2-setuptools-whl/now 44.1.1-1.2ubuntu0.22.04.1 all [installed,local]

python3-pip-whl/noble,now 23.3+dfsg-1 all [installed,automatic]

python3-wheel-whl/noble,now 0.42.0-1 all [installed]
```

The final result remains incomplete:

```
root@box:~# virtualenv -ppython2.7 /usr/local/kalite/venv                                                   
fail
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv/seed/embed/via_app_data/via_app_data.py", line 84, in _get
    result = get_wheel(
             ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 26, in get_wheel
    wheel = from_bundle(distribution, version, for_py_version, search_dirs, app_data, do_periodic_update, en
v)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py", line 13, in from_bundle
    wheel = load_embed_wheel(app_data, distribution, for_py_version, of_version)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py", line 33, in load_embed_wheel
    wheel = get_embed_wheel(distribution, for_py_version)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/embed/__init__.py", line 77, in get_embed_whee
l
    raise Exception((
Exception: Wheel for wheel for Python 2.7 is unavailable. apt install python2-wheel-whl
fail
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv/seed/embed/via_app_data/via_app_data.py", line 84, in _get
    result = get_wheel(
             ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 30, in get_wheel
    wheel = download_wheel(
            ^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 65, in download_wheel
    env = pip_wheel_env_run(search_dirs, app_data, env)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 115, in pip_wheel_env_run
    wheel = get_wheel(
            ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/acquire.py", line 26, in get_wheel
    wheel = from_bundle(distribution, version, for_py_version, search_dirs, app_data, do_periodic_update, en
v)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py", line 13, in from_bundle
    wheel = load_embed_wheel(app_data, distribution, for_py_version, of_version)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/bundle.py", line 33, in load_embed_wheel
    wheel = get_embed_wheel(distribution, for_py_version)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/virtualenv/seed/wheels/embed/__init__.py", line 77, in get_embed_whee
l
    raise Exception((
Exception: Wheel for pip for Python 3.12 is unavailable. apt install python3-pip-whl
Exception('Wheel for pip for Python 3.12 is unavailable. apt install python3-pip-whl')
RuntimeError: seed failed due to failing to download wheels wheel
```

(Possibly it's no longer worth it to try to keep KA Lite running on modern OS's.)

jvonau, 2024-02-26:
> Line 96 could be inserted in scripts/install_python2.sh
> 
> ```
> apt -y install python2-pip-whl python2-setuptools-whl python3-wheel-whl
> ```
> 
> Leading to some-but-not-all of the needed packages:

You tend to do more than what was suggested, I'd drop the last 2 in the list... setuptools is being installed directly into the virtual environment via pip in the next stanza within the play and python3-wheel-whl is an automatic dependence already present but gets downgraded to the version in U22.04? Doesn't look like it. Come to think about it you should be able to install [wheel](https://wheel.readthedocs.io/en/stable/news.html) directly into the virtual environment in the same way as setuptools is pinned to 44.X I would suggest version 0.37.1 installed before setuptools

Think upstream has taken direct aim at this workaround, as this is more of a new issue with a change in upstream packaging as this routine used to install on 24.04 as you reported earlier or any I misremembering something?

> root@box:~# virtualenv -ppython2.7 /usr/local/kalite/venv

I suggest using  `source /usr/local/kalite/venv/bin/activate` to access the venv but I don't think its created at all.

> (Possibly it's no longer worth it to try to keep KA Lite running on modern OS's.)

Should of died years ago. One could tar up older OSs' /usr/local/kalite/venv (one for each supported arch) and just extract it into place.

jvonau, 2024-02-26:
I'm thinking a change to **virtualenv** might need to use the version from U22.04. I really don't see the need to use `apt-mark hold` for those packages as we don't care if they are later upgraded once the virtual environment is created. A problem might will crop up if somebody is doing something stupid like forcing the install a second time with `./runrole --reinstall kalite` or some other nonsense that would ignore the iiab_state.yml file's purpose.

Side note  [references wrong script](https://github.com/iiab/iiab/blob/master/runrole#L42)

holta, 2024-02-26:
> Think upstream has taken direct aim at this workaround, as this is more of a new issue with a change in upstream packaging as this routine used to install on 24.04 as you reported earlier or any I misremembering something?

I tend to agree:

24.04 upgrading from Python 3.11 to 3.12 late last week changed many things (:

tim-moody, 2024-02-29:
just installed kalite on debian 12 vm. works fine. I have the impression that python 3.11 is what deb12 gets unless you explicitly install 3.12 which I would not do. If that remains true for the foreseeable future works for me.

holta, 2024-02-29:
It's about how Python 3.12 (and many associated things) are packaged.

Ubuntu 24.04 pre-releases are affected since about a week ago.

Debian 12 and Bookworm-like OS's are completely unaffected.

jvonau, 2024-03-04:
I got this to go on U24.04 at the cmdline in a multipass VM with a bit of screwing around. I'll see if I can condense the notes down into something usable. Raw notes pulled from bash_ history: [py2-raw.txt](https://github.com/iiab/iiab/files/14476437/py2-raw.txt)

holta, 2024-03-06:
Thanks to @jvonau for the offering an Ubuntu 24.04 solution (for Python 2 and KA Lite) which is now merged!

- PR #3714 
- PR #3715

closed 2024-03-06 by holta