iiab/iiab #3716: python3-venv / python3.12-venv missing: Is Python 2 / KA Lite workaround blocking JupyterHub venv on Ubuntu 24.04?

holta, 2024-03-06:
@jvonau is #3714 / #3715 triggering this new venv problem(s) on Ubuntu 24.04?

Either way, would you have any recommendation to fix this JupyterHub venv problem?

> TASK [jupyterhub : pip install 3 packages into virtual environment: /opt/iiab/jupyterhub (~326 MB total, after 2 Ansible calls)] ************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/usr/bin/python3", "-m", "venv", "--system-site-packages", "/opt/iiab/jupyterhub", "/opt/iiab/jupyterhub"], "msg": "stdout: The virtual environment was not created successfully because ensurepip is not\navailable.  On Debian/Ubuntu systems, you need to install the python3-venv\npackage using the following command.\n\n    apt install python3.12-venv\n\nYou may need to use sudo with that command.  After installing the python3-venv\npackage, recreate your virtual environment.\n\nFailing command: /opt/iiab/jupyterhub/bin/python3\n\n"}

FWIW after I (1) upgraded from KA Lite (as IIAB's only app/service) to LARGE-sized using ./iiab-configure (leading to the error above) I then (2) ran `./runrole jupyterhub` which completed, leading to iiab-diagnostics: http://sprunge.us/Ln1IXC?en

And then:

```
root@box:~# apt list python3-venv python3.12-venv
Listing... Done
python3-venv/noble,now 3.12.1-0ubuntu2 amd64 [installed,automatic]
python3.12-venv/noble,now 3.12.2-1 amd64 [installed,automatic]
```
Possibly related:

- #3711
- PR #3714
- PR #3715

jvonau, 2024-03-07:
I'll go with strange as calibe-web installed cleanly after kalite, so I'm a bit perplexed as to why. When was the last time JupyterHub was tested on U24.04?

jvonau, 2024-03-07:
--system-site-packages would override the ansible clause of system-stie-packages no in install,yml might have to go with yes and remove the --system-site-packages from the command line. no time to look further.

holta, 2024-03-07:
FYI I've traced the culprit to...

```
apt -y --allow-downgrades install python3-pip-whl=22.0.2+dfsg-1
```

Immediately beforehand it showed...

```
# apt list python3-venv python3.11-venv python3.12-venv
Listing... Done
python3-venv/noble,now 3.12.1-0ubuntu2 amd64 [installed]
python3.11-venv/noble 3.11.8-1 amd64
python3.12-venv/noble,now 3.12.2-1 amd64 [installed,automatic]
```

Whereas immediately after downgrading `python3-pip-whl` 24.0 to 22.0.2, it shows...

```
# apt list python3-venv python3.11-venv python3.12-venv
Listing... Done
python3-venv/noble 3.12.1-0ubuntu2 amd64
python3.11-venv/noble 3.11.8-1 amd64
python3.12-venv/noble,now 3.12.2-1 amd64 [residual-config]
```

closed 2024-03-07 by holta