iiab/iiab #3720: JupyterHub failed to install on Raspberry Pi OS Bookworm on RPi 4

holta, 2024-03-11:
@kmw is this possibly intermittent?  Thanks for the report:

- Did you try `sudo iiab` repeatedly?
- Did you try a fresh install?

Not yet clear if this happens very regularly:

> TASK [jupyterhub : Break up jupyterhub/jupyterlab pip installs into 2 parts (3 packages above + 4 packages here) due to mutual dependency deadlock on some OS's] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/opt/iiab/jupyterhub/bin/pip3", "install", "--no-cache-dir", "--prefer-binary", "jupyterlab", "jupyterhub_firstuseauthenticator", "jupyterhub-systemdspawner", "ipywidgets"], "msg": "stdout: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple\nCollecting jupyterlab\n  Downloading https://www.piwheels.org/simple/jupyterlab/jupyterlab-4.1.4-py3-none-any.whl (11.4 MB)\n     â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”                     5.8/11.4 MB 138.5 kB/s eta 0:00:41\n\n:stderr: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f91806f10>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/jupyterlab/\nERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.\n    jupyterlab from https://www.piwheels.org/simple/jupyterlab/jupyterlab-4.1.4-py3-none-any.whl#sha256=f92c3f2b12b88efcf767205f49be9b2f86b85544f9c4f342bb5e9904a16cf931:\n        Expected sha256 f92c3f2b12b88efcf767205f49be9b2f86b85544f9c4f342bb5e9904a16cf931\n             Got        ab9120dc29988403da61c520d41a363de6bf5bb65c89bc6c104721f253d24820\n\n"}

kmw, 2024-03-11:
[image]

I have not done multiple sudo iiab on this error I did on the previous kilibre error. I will try it again. It's still running.
Yes I've done a fresh install each time using the imager as per the attachment. 
Thanks

kmw, 2024-03-11:
Great! I ran sudo iiab one more time and the installation completed. I'm now install over 100gig of content.
Thank you very much

jvonau, 2024-04-30:
This is no longer an issue right? Why is this still open?

holta, 2024-04-30:
Agreed.  We can always re-open if absolutely nec.

closed 2024-04-30 by holta