iiab/iiab #3727: Test ansible-core 2.17 ETA 2024-05-20

holta, 2024-04-08:
ansible-core 2.17 RC1 (Release Candidate 1) is expected 3 weeks prior, on Monday 2024-04-29:
https://docs.ansible.com/ansible/latest/roadmap/ROADMAP_2_17.html

Instructions for testing any ansible-core 2.17 pre-release:

- Look for the most recent 2.17 pre-release at: https://pypi.org/project/ansible-core/#history
- As of today, that's 2.17.0b1, whose page shows: `pip install ansible-core==2.17.0b1`
- So using today's example (above) we can modify the right side of [Line 278 in /opt/iiab/iiab/scripts/ansible](https://github.com/iiab/iiab/blob/20a48b3de0ddb662168358cb09c2e5447bc88514/scripts/ansible#L278C1-L278C89) to be...
  ```
      /usr/local/ansible/bin/python3 -m pip install --prefer-binary --upgrade ansible-core==2.17.0b1
  ```
- Run `cd /opt/iiab/iiab` then `sudo scripts/ansible` to install it.
- Continue installing IIAB software (or install an individual IIAB role, by running `sudo ./runrole <role name>`) to confirm Ansible's working well (or well enough!)

holta, 2024-04-08:
@EMG70 no rush — but awesome if you have time to test this in April and/or May — on any fresh/clean OS:

1. mkdir /opt/iiab
2. cd /opt/iiab
3. git clone https://github.com/iiab/iiab
4. cd /opt/iiab/iiab
5. nano scripts/ansible    # Edit line 278 as outlined above!
6. curl iiab.io/install.txt | bash

holta, 2024-04-26:
Something to watch for as ansible-core 2.17 lands in coming weeks...

...just a warning for now (on Ubuntu 24.04 with Python 3.12) but looking more serious soon:

> TASK [cups : CUPS web administration: Create Linux username 'Admin' with password 'changeme' in Linux group
'lpadmin' (shell: /usr/sbin/nologin, create_home: no)] ***
[DEPRECATION WARNING]: Encryption using the Python crypt module is deprecated. The Python crypt module is
deprecated and will be removed from Python 3.13. Install the passlib library for continued encryption
functionality. This feature will be removed in version 2.17. Deprecation warnings can be disabled by
setting deprecation_warnings=False in ansible.cfg.

holta, 2024-04-29:
2.17 RC1 is now available:

- https://github.com/ansible/ansible/releases/tag/v2.17.0rc1
- https://pypi.org/project/ansible-core/#history

jvonau, 2024-04-30:
#3571 is an issue with cups' dependencies

holta, 2024-05-01:
Great news:

A LARGE-sized IIAB install (minus Kolibri and KA Lite of course!) worked with ansible-core 2.17 Release Candidate 1 on Ubuntu Server 24.04 (iiab-diagnostics http://sprunge.us/xAZ8UL) after this CUPS fix was applied...

- PR #3735

Aside, just fyi if anyone's curious about CUPS on Ubuntu 24.04...

```
root@u24-ansible-217:~# apt list "*cups*" | grep installed

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

cups-browsed/noble,now 2.0.0-0ubuntu10 amd64 [installed,automatic]
cups-client/noble,now 2.4.7-1.2ubuntu7 amd64 [installed,automatic]
cups-common/noble,now 2.4.7-1.2ubuntu7 all [installed,automatic]
cups-core-drivers/noble,now 2.4.7-1.2ubuntu7 amd64 [installed,automatic]
cups-daemon/noble,now 2.4.7-1.2ubuntu7 amd64 [installed,automatic]
cups-filters-core-drivers/noble,now 2.0.0-0ubuntu4 amd64 [installed,automatic]
cups-filters/noble,now 2.0.0-0ubuntu4 amd64 [installed,automatic]
cups-ipp-utils/noble,now 2.4.7-1.2ubuntu7 amd64 [installed,automatic]
cups-ppdc/noble,now 2.4.7-1.2ubuntu7 amd64 [installed,automatic]
cups-server-common/noble,now 2.4.7-1.2ubuntu7 all [installed,automatic]
cups/noble,now 2.4.7-1.2ubuntu7 amd64 [installed]
libcups2t64/noble,now 2.4.7-1.2ubuntu7 amd64 [installed,automatic]
libcupsfilters2-common/noble,now 2.0.0-0ubuntu7 all [installed,automatic]
libcupsfilters2t64/noble,now 2.0.0-0ubuntu7 amd64 [installed,automatic]
```

CONCLUSION: We're likely all set for ansible-core's final release on 2024-05-20. ✅

holta, 2024-05-13:
Near-final release (RC2) of ansible-core 2.17:

https://github.com/ansible/ansible/releases/tag/v2.17.0rc2

closed 2024-05-31 by holta