iiab/iiab #3731: KA Lite fails to install on newly released Ubuntu 24.04

holta, 2024-04-26:
Error is:

> TASK [kalite : Fix KA Lite bug in regex parsing ifconfig output (ifcfg/parser.py) for @m-anish's network nam
es that contain dashes] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Path /usr/local/kalite/venv/lib/python2.7/site-pac
kages/kalite/packages/dist/ifcfg/parser.py does not exist !", "rc": 257}

cc: @jvonau

jvonau, 2024-04-29:
Dependencies changed within `sudo ./scripts/install_python2_kalite-venv_u2404.sh`
```
The following package was automatically installed and is no longer required:
  python3-setuptools-whl
Use 'sudo apt autoremove' to remove it.
The following packages will be REMOVED:
  python3-venv python3.12-venv
The following packages will be DOWNGRADED:
  python3-pip-whl
0 upgraded, 0 newly installed, 1 downgraded, 2 to remove and 1 not upgraded.
Need to get 1679 kB of archives.
After this operation, 59.4 kB disk space will be freed.
Get:1 http://archive.ubuntu.com/ubuntu jammy/universe amd64 python3-pip-whl all 22.0.2+dfsg-1 [1679 kB]
Fetched 1679 kB in 1s (1592 kB/s)         
(Reading database ... 70856 files and directories currently installed.)
Removing python3-venv (3.12.3-0ubuntu1) ...
Removing python3.12-venv (3.12.3-1) ...
dpkg: warning: downgrading python3-pip-whl from 24.0+dfsg-1ubuntu1 to 22.0.2+dfs
g-1
(Reading database ... 70849 files and directories currently installed.)
Preparing to unpack .../python3-pip-whl_22.0.2+dfsg-1_all.deb ...
Unpacking python3-pip-whl (22.0.2+dfsg-1) over (24.0+dfsg-1ubuntu1) ...
Setting up python3-pip-whl (22.0.2+dfsg-1) ...
Scanning processes...                                                           
Scanning linux images...                                                        

Running kernel seems to be up-to-date.

No services need to be restarted.

No containers need to be restarted.

No user sessions are running outdated binaries.

No VM guests are running outdated hypervisor (qemu) binaries on this host.
python3-pip-whl set on hold.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3 : Conflicts: python3-distutils but 3.10.8-1~22.04 is to be installed
 python3-distutils : Depends: python3:any (< 3.12)
                     Depends: python3-lib2to3 (= 3.10.8-1~22.04) but 3.12.3-0ubuntu1 is to be installed
E: Unable to correct problems, you have held broken packages.
python3-pip set on hold.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-virtualenv : Depends: python3-pip (>= 20.1-1~) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
python3-virtualenv set on hold.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 virtualenv : Depends: python3-virtualenv but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
virtualenv set on hold.
./install_python2_kalite-venv_u2404.sh: line 31: virtualenv: command not found
./install_python2_kalite-venv_u2404.sh: line 33: cd: /usr/local/kalite/venv: No such file or directory
./install_python2_kalite-venv_u2404.sh: line 34: bin/activate: No such file or directory
./install_python2_kalite-venv_u2404.sh: line 35: bin/pip: No such file or directory
```

jvonau, 2024-04-30:
Downgrading is no longer possible. No magic potion available. Move to unpacking a virtual environment created on U22.04 but python2 would still have to be installed but the script would be much less invasive.

jvonau, 2024-04-30:
Not a bug, just evolution with the newest OS.

holta, 2024-04-30:
Disappointing but good to know!

holta, 2024-05-01:
Patch has been applied, so that mainline IIAB installs on Ubuntu 24.04 can now proceed:

- PR #3733

I'm tempted to leave this ticket open for a few weeks, just in case others have other opinions on how to proceed here?