iiab/iiab #3737: error installing on ub 2404 vm

tim-moody, 2024-05-02:
TASK [1-prep : Grab a clean copy of ssh.service] *************************************************************************************
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: If you are using a module and expect the file to exist on the remote, see the remote_src option
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Could not find or access '/usr/share/doc/avahi-daemon/examples/ssh.service' on the Ansible Controller.\nIf you are using a module and expect the file to exist on the remote, see the remote_src option"}

tim-moody, 2024-05-02:
root@ub2204base:/opt/iiab/iiab# ll /usr/share/doc/avahi-daemon/examples/
total 8
drwxr-xr-x 2 root root 4096 Apr  5 10:57 ./
drwxr-xr-x 3 root root 4096 May  2 17:45 ../

holta, 2024-05-02:
`/usr/share/doc/avahi-daemon/examples/ssh.service` is normally installed by apt package `avahi_daemon` which itself is normally installed by:

https://github.com/iiab/iiab/blob/673fff90e2e2eb92fadc1f97009eeb8795124f69/roles/network/tasks/install.yml#L21-L24

Any chance you've set `network_install: False` — or some other reason the above is not being run?

Run `/opt/iiab/iiab/scripts/iiab-diagnostics` to share more context?

tim-moody, 2024-05-02:
I'm rerunning, but it looked like avahi was installed but was missing a file

tim-moody, 2024-05-02:
using stock local vars none

tim-moody, 2024-05-02:
Nope. I built an entirely new vm and got the same error

```
TASK [1-prep : Install Avahi (mDNS, Zeroconf/Bonjour)] *******************************************************************************
included: /opt/iiab/iiab/roles/network/tasks/avahi.yml for 127.0.0.1

TASK [1-prep : Create user 'avahi'] **************************************************************************************************
changed: [127.0.0.1]

TASK [1-prep : Install avahi announce config file /etc/avahi/services/schoolserver.service] ******************************************
changed: [127.0.0.1]

TASK [1-prep : Grab a clean copy of ssh.service] *************************************************************************************
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: If you are using a module and expect the file to exist on the remote, see the remote_src option
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Could not find or access '/usr/share/doc/avahi-daemon/examples/ssh.service' on the Ansible Controller.\nIf you are using a module and expect the file to exist on the remote, see the remote_src option"}

```

tim-moody, 2024-05-02:
http://sprunge.us/06Gtk9

tim-moody, 2024-05-02:
/usr/share/doc/avahi-daemon/examples is just empty

tim-moody, 2024-05-02:
on Deb 12 I see

```
root@iiab-ref:~# ll /usr/share/doc/avahi-daemon/examples
total 20
drwxr-xr-x 2 root root 4096 Mar  1 18:30 .
drwxr-xr-x 3 root root 4096 Mar  1 18:30 ..
-rw-r--r-- 1 root root 1497 Apr  1  2015 example.service
-rw-r--r-- 1 root root 1075 Apr 19  2023 sftp-ssh.service
-rw-r--r-- 1 root root 1071 Apr 19  2023 ssh.service
```

tim-moody, 2024-05-02:
I patched it with ssh.service from deb 12 install and got past the error.

holta, 2024-05-02:
FWIW I cannot reproduce this, e.g. on this 24.04 VM: http://sprunge.us/942sP6?en

> I'm rerunning, but it looked like avahi was installed but was missing a file

Can you create an empty 24.04 VM (do not install anything) and then run `apt install avahi-daemon` ?

For me this is the result:

```
root@u24:~# ll /usr/share/doc/avahi-daemon/examples
total 20
drwxr-xr-x 2 root root 4096 May  2 17:20 ./
drwxr-xr-x 3 root root 4096 May  2 17:20 ../
-rw-r--r-- 1 root root 1497 Apr  1  2015 example.service
-rw-r--r-- 1 root root 1075 Apr  5 06:57 sftp-ssh.service
-rw-r--r-- 1 root root 1071 Apr  5 06:57 ssh.service
```

(Is your result different?  Any idea why if so?)

tim-moody, 2024-05-02:
root@ub2404base:~# ll /usr/share/doc/avahi-daemon/examples
total 8
drwxr-xr-x 2 root root 4096 Apr  5 10:57 ./
drwxr-xr-x 3 root root 4096 May  2 22:28 ../

tim-moody, 2024-05-02:
Wonder if it's my Canadian archive

root@ub2404base:~# cat  /etc/apt/sources.list.d/ubuntu.sources
Types: deb
URIs: http://ca.archive.ubuntu.com/ubuntu/
Suites: noble noble-updates noble-backports
Components: main restricted universe multiverse
Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg

Types: deb
URIs: http://security.ubuntu.com/ubuntu/
Suites: noble-security
Components: main restricted universe multiverse
Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg

holta, 2024-05-02:
Insane!

If apt is failing to deliver legit apt packages a week after 24.04's release, this sounds quite serious. ⚠️

tim-moody, 2024-09-25:
just hit this error again

holta, 2024-09-25:
> Wonder if it's my Canadian archive

Is the above theory still viable?

Or any other theory?

(Over almost 6 months since Ubuntu 24.04's release, this is the only known such bug report.)

holta, 2024-09-25:
> I built an entirely new vm and got the same error

@tim-moody how are you building these VMs?

e.g. With what exact kind of Ubuntu & version, and what image?  (Downloaded from which specific URL?)

(For whatever mysterious reason nobody else appears to have run into this yet, over a full year of hundreds of 24.04 tests of both their pre-releases and post-releases.)

tim-moody, 2024-09-26:
with ubuntu-24.04-live-server-amd64.iso using VBoxManage
(2754981888 Apr 23 12:46 ubuntu-24.04-live-server-amd64.iso )
when it prompted me I think I chose the Canadian archive
```
root@ub2404base:~# cat /etc/apt/sources.list.d/ubuntu.sources
Types: deb
URIs: http://ca.archive.ubuntu.com/ubuntu/
Suites: noble noble-updates noble-backports
Components: main restricted universe multiverse
Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg

Types: deb
URIs: http://security.ubuntu.com/ubuntu/
Suites: noble-security
Components: main restricted universe multiverse
Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg

```
I then apt update and upgrade

holta, 2024-09-26:
Check the new https://releases.ubuntu.com/24.04/ubuntu-24.04.1-live-server-amd64.iso to see if it has the same problem? (2024-08-27, 2.6G)

(If you can!)

tim-moody, 2024-09-30:
build new 24.04.1 vm. also took the default on ubuntu server. (wonder if I selected the minimal server before)

None installed with no complaints, and in particular there were no problems with avahi.

holta, 2025-08-30:
Many thanks @elstevi ! 🙏

(Some context from Google...)

> Using the `unminimize` command-line utility:
>
> This command is specifically designed for Ubuntu Server Minimal installations. It converts a minimal server environment into a more standard server installation by installing additional packages and dependencies that are typically included in a full server installation. This is useful for users who initially deployed a minimal server but later require a broader set of tools and functionalities for interactive use or specific server roles."

holta, 2026-03-31:
> I had this error on a "minimal" 24.04 Ubuntu install. By running `unminimize`, it converted itself to a "full fat" Ubuntu installation in place. IIAB installation succeeded after that.

"The /usr/share/doc/avahi-daemon/examples/ssh.service file is **sometimes missing on Ubuntu images because it is packaged within the avahi-daemon documentation, which is often excluded from minimized**, cloud, or Docker container images to save space."

@westleyd do the above tips help, and solve the problem?

cc: @chapmanjacobd @jvonau