iiab/iiab #3747: "Scale of the Universe" module fails to run without Internet

mattrossman, 2024-05-29:
### Expected behavior and actual behavior:

Expected behavior is that the Scale of the Universe module can function offline.

Actual behavior is that it works as long as my device has an internet connection, but silently fails if I'm connected to IIAB offline (e.g. connected via the server's wifi access point).

The client's browser console shows various network errors where it tries unsuccessfully to pull Internet hosted resources such as Google Fonts and CloudFront.

[image]

### Steps to reproduce the problem:

Install "Scale of the Universe" module (version 2.1)

In my case this was included with the English speaking School 256G Starter

Click Scale of the Universe link on IIAB home page

Select a language and attempt to click "Start" at which point nothing happens. Observe errors in browser console.

### IIAB version, OS version, hardware details:

**Server**
ThinkPad T490
Version: master
Date Installed: 2024-05-28T21:38:07Z
Commit ID: 1733512463dcf50f03cbb572fde3e6c9385ff827

**Client**
2023 MacBook Pro
MacOS Sonoma 14.4.1
Chrome 125.0.6422.76

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

tim-moody, 2024-05-29:
Not sure what to say. This module has deteriorated over the years, but we are not the author. I have considered removing it from the catalog.

Any ideas are welcome.

mattrossman, 2024-05-29:
The source code for the app appears to be here https://github.com/matttt/scale_of_the_universe

Related https://github.com/iiab/iiab/issues/2628

It looks like the whole thing is already built to static files inside `dist`. When I serve those static from local server (e.g. `npx serve dist`), most of the problematic resources (those reaching out to CloudFront) load as desired from localhost.

The only remaining assets I see loading from the Internet are the two Google Font assets which shouldn't be too hard to add local copies for.

I could probably share an updated archive of the site with all assets bundled locally.

Note to @matttt—I am able to run the `build` script but `build:prod` produces errors from `gulp-google-closure-compiler`. Perhaps he could shed light on how to successfully build the production bundle, as newer Node/NPM versions may have introduced incompatibilities in the 3 yrs since the last commit there.

mattrossman, 2024-05-29:
Here you go:
https://github.com/mattrossman/scale_of_the_universe/blob/ef7a07b4b4bb0352dc48182043a18667c1e72a73/dist.zip

Compare changes:
https://github.com/mattrossman/scale_of_the_universe/compare/master..iiab

I basically just did a `wget` on each of the Google Font assets (2 CSS files, 2 TTF files) and updated URLs to point at the local assets. I also deleted a bunch of old assets that no longer seemed to be used to reduce the archive size.

Size could shrink further if images were optimized with modern formats such as WebP.

But ultimately it works offline. I tested by downloading/extracting the new archive, starting a local server to serve the static files (I used `npx serve` inside the dist folder for this) and turning off Internet on my computer. The page loaded and everything functioned properly from what I tested (audio playing, buttons respond, other languages work).

holta, 2024-05-30:
Incredible!

matttt, 2024-05-30:
Thank you @mattrossman ! Sorry you had some trouble with the repo. I'm a bit inundated at the moment but maybe you could try using node/npm versions from 2020?

tim-moody, 2024-05-30:
I'm not clear on whether this runs offline with a web server or is a node app. Is node only needed to build?

The files also need to be hosted somewhere and that location added to our catalog. This will allow installation with the Admin Console by simply checking a box and clicking a button.

mattrossman, 2024-05-30:
@tim-moody These are static HTML/CSS/JS/etc files that can be hosted on any web server, it is not a Node app. Node was (at some point) used to build but I didn't use it for my changes, I just edited the static files manually.

The modified files in my branch are hosted on GitHub Pages here: https://mattrossman.github.io/scale_of_the_universe/

Note I had to make [an additional change](https://github.com/mattrossman/scale_of_the_universe/commit/e1ffd64d403b07d2e2c1ac4e87713844efbd7548) so font resource links use relative instead of absolute paths for GH Pages. I am not sure what your catalog expects in terms of resource paths.

You may wish to point it the catalog at a link that doesn't require trusting me though. For instance raw.githack.com serves the raw files from GitHub pinned to a commit hash, e.g. here is the [index.html](https://rawcdn.githack.com/mattrossman/scale_of_the_universe/e1ffd64d403b07d2e2c1ac4e87713844efbd7548/index.html) file (note that link doesn't run directly in browser due to a cross origin security quirk with githack media, but the files are ultimately the same as downloading them from the repo and would behave fine offline). Or you can host the files wherever else you please.