iiab/iiab #3754: Does `iiab_gateway_enabled: True` still work?

holta, 2024-06-18:
@EMG70 struggled to get "passthrough" to work today, without luck:

- No matter if upstream Internet was provided to the RPi 400 IIAB via an Ethernet cable, or via a 2.4FGz WiFi.

- He opened up UDP and TCP on both ports 80 and 443 (using IIAB's iptables firewall) just in case, but still couldn't get `iiab_gateway_enabled: True` to work.

- His test case: do student devices (connected to IIAB's WiFi hotspot) browse real/live Internet websites?

Nothing has worked so far.  Of course after changing any network settings in /etc/iiab/local_vars.yml, he ran `sudo iiab-network` and then rebooted.

His iiab-diagnostics:

- https://dpaste.com/GMEWXM9PG#line-410
- https://dpaste.com/GMEWXM9PG#line-3056

holta, 2024-06-18:
@EMG70 got it to work... curiously by adding this DNS to local_vars.yml :

```
wan_nameserver: 8.8.8.8
```

EMG70, 2024-06-18:
I can confirm passthrough finally worked for both wifi and ethernet connections when wanserver setting is appended as above.I only tried after googling and found high scores relating to wanserver.