iiab/iiab #3755: Calibre-web on Rpi Bookworm -  502 Bad Gateway

miloshev, 2024-06-25:
### Expected behavior and actual behavior: When i try to start calibre-web from the main menu it goes to a page with nginx 502 Bad Gateway
Everything else is working

### Steps to reproduce the problem:

### IIAB version, OS version, hardware details: Latest version of IIAB, RPi Bookworm, Rpi3B+

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!) 
https://dpaste.com/AFPYPWSV4

holta, 2024-06-25:
Hi @miloshev

Coincidentally this was fixed yesterday!

Please either [upgrade](https://github.com/iiab/calibre-web/wiki#upgrading) or reinstall, Thanks.

Related:

- PR IIAB/calibre-web#201

holta, 2024-06-25:
@miloshev

And please reconfirm that you get Calibre-Web working, if possible?

(This would be very useful 32-bit testing for others, as you are running "armed" as your primary OS architecture, according to your iiab-diagnostics !)

miloshev, 2024-06-25:
I have tried to upgrade, but "git pull https://github.com/iiab/calibre-web --no-rebase --no-edit"
throws an error: "fatal: not a git repository (or any of the parent directories): .git"

holta, 2024-06-25:
> fatal: not a git repository (or any of the parent directories): .git

It sounds like... you're not in the right directory?

Are you able to run all 7 commands in order, one at a time?

```
sudo -i
pipx upgrade --include-injected xklb
cd /usr/local/calibre-web-py3
git pull https://github.com/iiab/calibre-web --no-rebase --no-edit
cd /opt/iiab/iiab
git pull https://github.com/iiab/iiab --no-rebase --no-edit
./runrole --reinstall calibre-web
```

Paste a larger screenshot / photo of the error on your screen maybe?

miloshev, 2024-06-25:
Sorry, my mistake.
Now Calibre works :)

holta, 2024-06-25:
Huge thanks @miloshev, for confirming!

closed 2024-06-25 by holta

EMG70, 2024-06-25:
IIAB -DIAGNOSTICS -https://dpaste.com/78PF7Y8QA
Tested on Raspberry Pi 3B+, IIAB small size installed.
Calibre-web working despite the slow downloads, a Playlist of 16 short videos all downloaded well.
[image]

holta, 2024-06-25:
> Tested on Raspberry Pi 3B+, IIAB small size installed.
Calibre-web working despite the slow downloads, a Playlist of 16 short videos all downloaded well.

Tremendous news:

Thank you @EMG70 for reconfirming!