iiab/iiab #3758: WARNING: Support for installing IIAB onto Ubuntu 22.04 and 23.10 will soon deprecated / desupported — likely by/around early August 2024

holta, 2024-06-25:
Just a reminder that supporting too many OS's lowers software quality.

Keeping OS coverage to a sane number (i.e. only very recent versions of Ubuntu, Mint, Debian, and Raspberry Pi OS) keeps software quality much higher.

So IIAB tries to offer 15 days of transitional support — for both the old and the new version of an LTS OS — and in this case much more than 15 days, as a result of Kolibri not running on Ubuntu 24.04 until last week!

Background context:

- PR #3045
- PR #3415
- #3416
- PR #3621
- #3672
- PR #3691

Related:

- PR #3751

tim-moody, 2024-06-25:
23.10 is EOL next month. 22.04 not until Apr, 2027.

More time could be saved by not supporting an OS until it is released, like 24.10.

holta, 2024-06-25:
> More time could be saved by not supporting an OS until it is released

Many in the Debian community and other distro communities respectfully disagree (the most important work and most efficient community and infra work, often happens prior to OS release).  But re-litigating this argument 5 years in a row definitely isn't healthy or constructive.  The aforementioned tickets (and #3027) explain in detail reasons behind the carefully considered decision to use about 15 days of transition time (old LTS version and new LTS version of OS's), with sincere & due respect for all.

holta, 2024-08-13:
Just FYI "Ubuntu 24.04.1 LTS was scheduled to ship this week but has now been delayed to the end of August" [likely August 29th]

https://phoronix.com/news/Ubuntu-24.04.1-LTS-Delay

closed 2025-07-02 by holta