iiab/iiab #3781: "after a while, browsing to IIAB home page fails, and requires IIAB to be rebooted, somewhat regularly"

holta, 2024-08-05:
Question from @nzola:

> cellphones can't open the browser. But is it is rebooting iiab that repairs the issue. This situation is still coming back and I have to reboot the system to bring up the front page. connecting to iiab wifi is not a problem.

> sometimes I get the following message:

```
The home page menu-json file is broken and could not be patched
```

> [Is the problem caused by me changing the hostname to lokole-RDC from box.lan???]

@tim-moody might have ideas — as to what other diagnostics @nzola should look out for — and experiments he can hopefully try?

Thanks to @nzola, iiab-diagnostics for both Raspberry Pi 4 machines (each 4GB RAM, LARGE-sized IIAB installs) are here:

- hostname lokole-RDC : https://dpaste.com/75N82CTWM
- hostname lokole-FR: https://dpaste.com/8T9EXV4YH

Is this hypothetically/perhaps related to the issues below?

- #3487
- #3602

holta, 2024-08-05:
> The home page menu-json file is broken and could not be patched

@nzola must mean:

```
The Home Page menu.json file is broken and could not be patched.
```

...likely arising from either of these:

- https://github.com/iiab/iiab-admin-console/blob/4f4399c373c98776431fcb42b60ac8b41c137792/roles/console/files/js/js_menu_functions.js#L143-L150
- https://github.com/iiab/iiab-admin-console/blob/4f4399c373c98776431fcb42b60ac8b41c137792/roles/js-menu/files/menu-files/js/js-menu.js#L182-L190

holta, 2024-08-05:
FYI both iiab-diagnostics below will disappear from dpaste.com (free pastebin service) in 30 days...

> - hostname lokole-RDC : https://dpaste.com/75N82CTWM
> - hostname lokole-FR: https://dpaste.com/8T9EXV4YH

...so let's paste any suspicious / relevant details onto this ticket itself if possible, to be sure!?

tim-moody, 2024-08-05:
I don't see anything wrong with menu.json in https://dpaste.com/75N82CTWM

tim-moody, 2024-08-05:
nor in https://dpaste.com/8T9EXV4YH

tim-moody, 2024-08-05:
maybe try to get a screen shot and copy of menu.json prior to any reboot

nzola, 2024-08-06:
> maybe try to get a screen shot and copy of menu.json prior to any reboot

[image]

admin@lokole-RDC:/library/www/html/home $ less menu.json
{
  "debug": false,
  "menu_lang": "fr",
  "force_mobile_layout": false,
  "mobile_header_font": "Open Sans",
  "mobile_incl_description": false,
  "mobile_incl_extra_description": false,
  "mobile_incl_extra_html": false,
  "mobile_incl_footnote": false,
  "desktop_header_font": "Finger Paint",
  "desktop_incl_description": true,
  "desktop_incl_extra_description": true,
  "desktop_incl_extra_html": true,
  "desktop_incl_footnote": true,
  "allow_kiwix_search": true,
  "allow_poweroff": false,
  "poweroff_prompt": "Power Off",
  "autoupdate_menu": true,
  "allow_server_time_update": true,
  "multi_tab": false,
  "tab_names": [
    "Main"
  ],
  "menu_items_1": [
    "fr-wikipedia_fr_all_maxi",
    "fr-osm_viewer_v2",
    "fr-kalite_en",
    "fr-vikidia_fr_all_maxi",
    "fr-les-fondamentaux_fr_all",
    "fr-wiktionary_fr_all_maxi",
    "fr-wikiquote_fr_all_maxi",
    "fr-phet_fr",
    "fr-wikiversity_fr_all_maxi",
    "fr-wikivoyage_fr_all_maxi",
    "fr-biologie-tout-compris_fr_all",
    "fr-wikibooks_fr_all_maxi",
    "fr-gutenberg_fr_all",
    "fr-jaimelire_fr",
    "fr-japprendsalire_fr",
    "fr-deus-ex-silicium_fr_all",
    "fr-dirtybiology_fr_all",
    "fr-e-penser_fr_all",
    "fr-wikimooc_fr_all",
    "fr-usb",
    "en-cups",
    "en-lokole",
    "en-mediawiki",
    "en-nodered",
    "en-nextcloud",
    "en-wordpress",
    "en-kolibri",
    "en-moodle",
    "en-sugarizer",
    "en-internetarchive",
    "en-calibreweb",
    "fr-women-in-african-history",
    "fr-afrique-marie-wabbes",
    "fr-phet-haiti",
    "fr-catdogbooks",
    "fr-basic_elecricity_fr_all_min",
    "fr-voa_fr_all",
    "fr-ubongo_fr_all",
    "fr-haitifutur",
    "fr-wikihow_fr_maxi",
    "fr-ebooksgratuits",
    "fr-cest-pas-sorcier_fr_all",
    "en-credits"
  ]
}

tim-moody, 2024-08-06:
I don't see any problem with menu.json. Looks like a connectivity problem. How many people are connected?

/usr/sbin/hostapd_cli all_sta |grep :

might show you. sometimes people get kicked off if they are idle and another user tries to connect

nzola, 2024-08-06:
Only one person connecting. Just me, testing.

nzola, 2024-08-09:
> My gut feel is that there is too many components running on PI 4 with 4GB of RAM. 4GB is probably not enough. Try running on 8GB model if possible.

The iiab I have been having browsing trouble is an 4 GB and has a lot of material on it, more than 300 GB. I will install a basic one and test browsing.

nzola, 2024-08-11:
> I don't see any problem with menu.json. Looks like a connectivity problem. How many people are connected?
> 
> /usr/sbin/hostapd_cli all_sta |grep :
> 
> might show you. sometimes people get kicked off if they are idle and another user tries to connect

@tim-moody  How do I apply this command line? I tried sudo less /usr/sbin/hostapd_cli all_sta |grep : It didn't work.

tim-moody, 2024-08-11:
You don't want the 'less'

nzola, 2024-08-26:
> My gut feel is that there is too many components running on PI 4 with 4GB of RAM. 4GB is probably not enough. Try running on 8GB model if possible.

I installed a new iiab on pi3 about 2 weeks ago with minimum components, but it still need reboot to browse after sitting for a while without activities.

nzola, 2024-08-26:
This issue is happening with the following devices: android Samsung and Motorolla, ipad, iphone and 
Beelink SER5 MAX Mini PC. 
But not with Windows 10 on lenovo and HP Pavillon. These devices browse without the need of restarting iiab pi4 and pi zero.

nzola, 2024-08-26:
For the last two weeks, I have been testing and comparing the iiab pi4 and pi3 new install with iiab pi4 old install (more than 3 years ago)
I noticed the old iiab pi4 does not require any reboot. It is always ready to browse on android Samsung and Motorolla, ipad, iphone, Beelink SER5 MAX Mini PC, Windows 10 on lenovo and HP Pavillon without any issues.

This issue concerns me because in Africa Android cellphones are pre-dominant over laptops. I would like to make sure cellphones browse without any problems.

**Should I install a Cron job to start the iiab web server at specific time interval?**

holta, 2024-08-26:
A few more details on @nzola's ongoing struggle:

> [Nzola writes:] I would like to know about the people you mentioned [some time] ago, who had browsing problems, whether they are using iiab on pi4 server and using cellphones to access the server?
Because I still have this issue to restart iiab p4 server every 2 hours to keep it alive. I tested a more than 3 years old iiab, no issues. I noticed the issue is only with android and apple products, not pc windows browser.

> [Response from @EMG70:] Hi Nzola.The few people who were struggling to access the content were using old Android phones Version 6 mostly. I am using linux 23.10 on a lenovo server so not sure performance on raspberry pi ,but i have not had issues with my test kit here(Android 7 phone) and Raspberry pi 4/400/3b+.

> [Nzola writes:] Ok. I did extensive tests between different iiab pi server and noticed newer intall has this sleeping issue. The old one works all the time.

holta, 2024-08-26:
> This issue is happening with the following devices: android Samsung and Motorolla, ipad, iphone and Beelink SER5 MAX Mini PC.

> But not with Windows 10 on lenovo and HP Pavillon. These devices browse without the need of restarting iiab pi4 and pi zero.

@nzola what exact browser(s) are you using?

- if browser on Windows 11 on Beelink SER5 MAX Mini PC consistently leads to problems, can you please (also) send screenshots of menu.json BEFORE rebooting??

- If browser on Windows 10 on Lenovo or HP Pavilion never leads to problems, can you confirm exact browser & version number?

nzola, 2024-08-27:
> > This issue is happening with the following devices: android Samsung and Motorolla, ipad, iphone and Beelink SER5 MAX Mini PC.
> 
> > But not with Windows 10 on lenovo and HP Pavillon. These devices browse without the need of restarting iiab pi4 and pi zero.
> 
> @nzola what exact browser(s) are you using?
> 
> * If browser on Windows 11 on Beelink SER5 MAX Mini PC consistently leads to problems, can you please (also) send screenshots of menu.json BEFORE rebooting Raspberry Pi??
> * If browser on Windows 10 on Lenovo or HP Pavilion never leads to problems, can you (also) confirm their exact browser & version numbers?

For lenovo X220, I am using: 
OS Windows 10 Pro
Browser Edge Version 127.0.2651.98 (Official build) (64-bit)

For HP Pavillon 23, I am using:
OS Windows 10 Home
Browser Edge Version 127.0.2651.98 (Official build) (64-bit)

For Beelink SER5 MAX Mini PC, I am using:
OS Windows 11 Pro
Browser Edge Version 128.0.2739.42 (Official build) (64-bit)

For iphone, I am using browsers Google Chrome and Safary

For Androids (Samsung and Motorolla) using browsers Google Chrome and Internet

holta, 2024-08-27:
Another example of the same problem but from a different group, in a different country (Zimbabwe).

Report + screenshot thanks to @EMG70:

> This screenshot was sent to me last week

> There was low battery after running server overnight so I asked them to power down and the problem has not been reported since.

[image]

holta, 2024-08-27:
@nzola
Can `sudo iiab-diagnostics` be run immediately — as soon as you see `menu.json file is broken` — _and BEFORE the IIAB is rebooted?!_

@EMG70
If the IIAB is offline and `menu.json file is broken` appears, try to take a photo of the broken file — by running:
```
cat -v /library/www/html/home/menu.json
```

holta, 2024-08-27:
@tim-moody

- Would it be possible to make the pop-up error message contain a couple key diagnostics?

- So that people photographing this `menu.json file is broken` pop-up are sending us genuine insights or at least indicators to help us narrow the cause?

EMG70, 2024-08-27:
> @nzola Can `sudo iiab-diagnostics` be run immediately — as soon as you see `menu.json file is broken` — _and BEFORE the IIAB is rebooted?!_
>
> @EMG70 If the IIAB is offline and `menu.json file is [...]

I will ask users on the ground to run suggested command when the error occurs again.

EMG70, 2024-08-27:
SUDO IIAB DIAGNOSTICS - https://paste.centos.org/view/e76b938d
The broken menu.json message has just appeared in my IIAB server built over the weekend.The messages (mine)  may be misleading as i have been creating own sub menus recently including on the server sent to Zimbabwe.
I must admit it is work in progress and suspect wrong indention or missing comma or braces but so happened to Nzola also recently.
Although new menus were working yesterday i had still not downloaded modules via admin-console.It was just menu-defs,images, and /library/www/html/home/menu.json.
Please see attached cat -v /library/www/html/home/menu.json 
[menu.json](https://github.com/user-attachments/files/16761574/menu.json)

holta, 2024-08-27:
@tim-moody's suggestion for @nzola:

> To help isolate the problem can  you try restarting nginx instead of rebooting.
>
> This sounds like an idle timeout, and the fact that it affects different clients differently makes me suspicious that the problem is in the client.

In other words, @nzola please try tunning `sudo systemctl restart nginx` to see if that might help? (But do not reboot!)

nzola, 2024-08-27:
> @nzola Can `sudo iiab-diagnostics` be run immediately — as soon as you see `menu.json file is broken` — _and BEFORE the IIAB is rebooted?!_
>
> @EMG70 If the IIAB is offline and `menu.json file is [...]

I will run sudo iiab-diagnostics when I return home

nzola, 2024-08-28:
Here is my screen shot from Beelink SER5 MAX Mini PC, This happened after 2 hours of inactive.
[image]

@holta This is the answer to your request. This command is run immediately as soon as menu.json is broken: PUBLISHING TO URL... https://paste.centos.org/view/6b00490f

@tim-moody As suggested, I did not run normal reboot, but run this command:  **sudo systemctl restart nginx** to retart the iiab pi4 (lokole-rdc). This did did not fix the problem.
I run sudo reboot, and this fixed the problem.

tim-moody, 2024-08-28:
When a menu, home or sub, is invoked or refreshed, javascript tries to load menu.json and convert the json into an object with all the menu items. This is a try/catch, which on exception tries to fix up problems with the json like illegal characters and if that fails reports the error you are getting. The cause could be missing commas, unmatched curly braces, or maybe the file is empty, or maybe the server was unreachable.

Note that I don't think anything tries to fix the file on reboot, so once corrupt and unfixable it should remain in that state regardless of reboot. The actual state of the file at the point of reboot remains unknown as so far no one has  shown us the file contents prior to rebooting, but if it is OK after reboot, it was probably OK before reboot.

tim-moody, 2024-08-28:
I think the problem is that the server is unreachable and the error message is misleading.

if you were able to issue systemctl restart nginx you must be connected to the server.

is this a server where you are directly logged in, accessing via ssh over ethernet, or via the hotspot?

tim-moody, 2024-08-28:
I also don't see anything wrong with the menu.json reported in  iiab-diagnostics.

tim-moody, 2024-08-28:
When you reboot, the hotspot will be turned off, so all devices will lose connection and have to reassociate with the hotspot. What happens if you simply have a device with the error message reassociate with the server?

nzola, 2024-08-28:
> When a menu, home or sub, is invoked or refreshed, javascript tries to load menu.json and convert the json into an object with all the menu items. This is a try/catch, which on exception tries to [...]

@tim-moody I agree to what you've said. Though I said reboot fixed the problem, but it is not fixed.

nzola, 2024-08-28:
> When you reboot, the hotspot will be turned off, so all devices will lose connection and have to reassociate with the hotspot. What happens if you simply have a device with the error message reassociate with the server?

@tim-moody Do you suggest to reboot the device that is showing the problem instead of iiab pi4?

nzola, 2024-08-28:
> I think the problem is that the server is unreachable and the error message is misleading.
> 
> if you were able to issue systemctl restart nginx you must be connected to the server.
> 
> is this a server where you are directly logged in, accessing via ssh over ethernet, or via the hotspot?

@tim-moody I am both, directly logged in via hotspot, and connected to my home LAN to ssh it.

holta, 2024-08-28:
@tim-moody wrote:
> What happens if you simply have a device with the error message reassociate with the server?

1. @nzola please try @tim-moody's above suggestion — next time you see error message `menu.json file is broken` :
   - disconnect your client/student device from IIAB's Wi-Fi hotspot (if it's still connected?)
   - reconnect your client/student device to IIAB's Wi-Fi hotspot
   - try again to browse IIAB
   - also try with a completely different client/phone/tablet + different browser if possible?

2. If that really doesn't help, you might also try:
   - `sudo systemctl restart hostapd`
   - reconnect your client/student device to IIAB's Wi-Fi hotspot
   - try again to browse IIAB
   - also try with a completely different client/phone/tablet + different browser if possible?

tim-moody, 2024-08-28:
I'm wondering if I should have the client make a request to the server every 10 seconds or so in order not to timeout on idle.

nzola, 2024-08-28:
What is very interesting, The iiab pi4 (lokole) I build over 3 years ago doesn't have this problem, but only the newer build.
I am at work right now. This morning, before coming to work, I powered up 2 iiab pi4, 1 iiab pi zero and one 3 years+ iiab pi4.
I will return home at 7 pm. I will test the connection to each and see what happens.

I just thought about something else with regards to the hardware. The iiab box I build have an internal battery with a board to recharge the battery. I am think to power the iiab pi4 directly and see what happens.

This is the internal battery I use:
https://www.aliexpress.com/item/1005006018365329.html?src=google&src=google&albch=shopping&acnt=631-313-3945&slnk=&plac=&mtctp=&albbt=Google_7_shopping&gclsrc=aw.ds&albagn=888888&ds_e_adid=&ds_e_matchtype=&ds_e_device=c&ds_e_network=x&ds_e_product_group_id=&ds_e_product_id=en1005006018365329&ds_e_product_merchant_id=586275750&ds_e_product_country=CA&ds_e_product_language=en&ds_e_product_channel=online&ds_e_product_store_id=&ds_url_v=2&albcp=19373658437&albag=&isSmbAutoCall=false&needSmbHouyi=false&gad_source=1&gclid=CjwKCAjwlbu2BhA3EiwA3yXyu3vx52N04rtJbPkPTWUsIYcb77p8SBRASiE7gUFGCsU4yZT64GqmHhoCW90QAvD_BwE&aff_fcid=88518a9813d84a9da12cff48036270ed-1724860562692-04555-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=88518a9813d84a9da12cff48036270ed-1724860562692-04555-UneMJZVf&terminal_id=a62436d0bddb4a619c91cfc20e65e89c&afSmartRedirect=n

nzola, 2024-08-28:
@holta @tim-moody **I will try Tim's suggestions, before trying mine**

nzola, 2024-08-30:
> I'm wondering if I should have the client make a request to the server every 10 seconds or so in order not to timeout on idle.

This solution will work for a set of clients in a lab. But not good the general public who wants to connect to the server.
How about having the server auto run **sudo systemctl restart hostapd** ?? I actually find the script to do this. I implemented it last night, it seems to be working. But I do not know the consequences. I will do more tests today.
Here is the script:
**sudo nano /usr/local/bin/restart_hostapd.sh

#!/bin/bash

while true; do
    sudo systemctl restart hostapd
    sleep 600  # 600 seconds = 10 minutes
done

sudo chmod +x /usr/local/bin/restart_hostapd.sh
sudo nano /etc/systemd/system/restart_hostapd.service

[Unit]
Description=Restart hostapd every 10 seconds
After=network.target

[Service]
ExecStart=/usr/local/bin/restart_hostapd.sh
Restart=always

[Install]
WantedBy=multi-user.target

sudo systemctl daemon-reload
sudo systemctl enable restart_hostapd.service
sudo systemctl start restart_hostapd.service
sudo systemctl status restart_hostapd.service**

tim-moody, 2024-08-30:
@nzola can you test as follows:

wait until a client sees the error message or other evidence that the connection to the server is broken
now run systemctl status hostapd to see if hostapd knows it is not working and report the result
report whether you have an ssh connection over ethernet or wifi or both
see if any other client is still connected to the server over wifi
now restart hostapd

nzola, 2024-08-30:
> @nzola can you test as follows:
>
> wait until a client sees the error message or other evidence that the connection to the server is broken now run systemctl status hostapd to see if hostapd knows it [...]

@tim-moody Ok. I will wait until I see any errors, then I will report as you've just said.

nzola, 2024-09-02:
> @tim-moody Ok. I will wait until I see any errors, then I will report as you've just said.

I did a lot of tests changing the hostapd restard time and discovered the connections broke from the server at the hostapd restart time. 
Here is the report when the hostapd restart time was at every hour (3600 seconds).
 ● restart_hostapd.service - Restart hostapd every 1 hours
     Loaded: loaded (/etc/systemd/system/restart_hostapd.service; enabled; preset: enabled)
     Active: active (running) since Sun 2024-09-01 23:13:59 PDT; 1h 14min ago
   Main PID: 864 (restart_hostapd)
      Tasks: 2 (limit: 3910)
        CPU: 70ms
     CGroup: /system.slice/restart_hostapd.service
             ├─  864 /bin/bash /usr/local/bin/restart_hostapd.sh
             └─10796 sleep 3600

Sep 01 23:13:59 lokole-RDC sudo[866]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/systemctl restart hostapd
Sep 01 23:13:59 lokole-RDC sudo[866]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Sep 01 23:14:00 lokole-RDC sudo[866]: pam_unix(sudo:session): session closed for user root
Sep 01 23:14:03 lokole-RDC systemd[1]: /etc/systemd/system/restart_hostapd.service:11: Missing '=', ignoring line.
Sep 01 23:14:05 lokole-RDC systemd[1]: /etc/systemd/system/restart_hostapd.service:11: Missing '=', ignoring line.
Sep 01 23:14:08 lokole-RDC systemd[1]: /etc/systemd/system/restart_hostapd.service:11: Missing '=', ignoring line.
Sep 01 23:21:59 lokole-RDC systemd[1]: /etc/systemd/system/restart_hostapd.service:11: Missing '=', ignoring line.
Sep 02 00:14:18 lokole-RDC sudo[10785]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/systemctl restart hostapd
Sep 02 00:14:18 lokole-RDC sudo[10785]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Sep 02 00:14:19 lokole-RDC sudo[10785]: pam_unix(sudo:session): session closed for user root
~
I was monitoring the real time and found time the videos I was watching stopped exctly at the hostapd start time.
For this tests I was running long videos downloaded from Calibe-Web on 5 different devices; Mottorola and Samsung cellphone stopped and connections would not auto reconnect to the server. After I reconnect them manually to the server, the videos resumed. 2 windows computers did not loose connections, the videos continued to the end without stopping. 1 windows video stopped did not reconnect to server.
SSH still worked on WIFI and home LAN

nzola, 2024-09-08:
@tim-moody @holta I resolved this situation by clicking on **Forgot All WIFI Connections** in the admin console under control and wifi control. I think this stops internal wifi ???
My understanding is that we cannot run both hotspot and internal wifi simultaneously. Is this correct?

holta, 2024-09-08:
> My understanding is that we cannot run both hotspot and internal wifi simultaneously. Is this correct?

You can:

https://github.com/iiab/iiab/blob/d1bd41bf48f7c570ba49b24a486b1031ef8d9a09/vars/local_vars_small.yml#L94-L95

But I personally don't recommend it — as it tends to lead to a lot of problems (e.g. WiFi runs more slowly, and the WiFi firmware [for the internal WiFi chip] can fail completely in some cases, depending on your IIAB hardware).

A few more details: https://github.com/iiab/iiab/wiki/IIAB-Networking#2-apsta-mode

tim-moody, 2024-09-24:
from an Adm Cons point of view the bug is that the wrong error message is displayed: corrupted file should be no connection to server.

holta, 2024-09-26:
While different from hostapd.service repeatedly & silently failing to serve student/client WiFi devices (after about an hour or so) @nzola also described a different but also very serious problem: (confusingly the very same error message is often shown!)

After extensive testing, @nzola believes that `hostapd_secure: True` is colliding with `wifi_up_down: True` — i.e. as soon as the IIAB/RPi connects to a home router a.k.a. upstream "Internet Access WiFi router" — this damages if not destroys client/student WiFi connections to IIAB's hotspot.

RESULT:

- From that point onwards, many client/student machines (older Windows machines especially, but also some phones) claim they can still connect to the IIAB hotspot (typically RPi hotspot) — but their browsers suddenly start displaying the message "The home page menu-json file is broken and could not be patched" _(that's not really what's happening; reality seems to be that student/client devices have partially or completely lost the ability to connect to IIAB's hotspot!)_

- From that point onwards, many other client/student devices (recent model phones and computers especially?) can no longer connect to IIAB's hotspot at all!

POSSIBLE SUGGESTIONS, ESPECIALLY IF ABOVE IS CONFIRMED:

- The error message "The home page menu-json file is broken and could not be patched" should (if possible!) be made more accurate.  Ideally, if this proves possible, the error message would be tailored to provide more precise context at to what's really happening in each case?  e.g. if the true problem is "older machines" believing they're connect to IIAB/RPi's internal hotspot (when in reality they are not!?) perhaps some hints could be provided, to explain what really/likely happening?

- @nzola also suggests that `wifi_up_down: True` really should be changed to default to `wifi_up_down: False` going forward.  To immediately help a lot of people avoid this and many similar problems with "Dual Wifi".   It's true that a few of these problems are hitting more people in 2024 because of the increased prominence of http://box/admin > Control > Wifi Control > "Setup Internet Access WiFi Connection" (blue button) causing them to fall for temptation.  But in any case, the problems with `wifi_up_down: True` are real and growing, for a number of different reasons (wifi firmware, etc, etc) so all these painful setup/configuration challenges (arising from Dual WiFi, that very few deployments want) really need to be "opt in" rather than "opt out".

tim-moody, 2024-09-30:
Added https://github.com/iiab/iiab-admin-console/pull/585.
No idea if it helps as I can not reproduce the problem, so can't test.

jvonau, 2025-01-19:
>> [@tim-moody](https://github.com/tim-moody) [@holta](https://github.com/holta) I resolved this situation by clicking on **Forgot All WIFI Connections** in the admin console under control and wifi control. 

>After extensive testing, @nzola believes that hostapd_secure: True is colliding with wifi_up_down: True — i.e. as soon as the IIAB/RPi connects to a home router a.k.a. upstream "Internet Access WiFi router" — this damages if not destroys client/student WiFi connections to IIAB's hotspot.

Sounds like the upstream WiFi connection was 5GHz. Is anybody else having issues with using passwords with hostapd?

>I think this stops internal wifi ??? 

Yes wlan0 is for upstream, ap0 is for hostapd but only one channel can be used for both, limit of the firmware. The choice was made to only support 2.4GHz for hostapd.  
>My understanding is that we cannot run both hotspot and internal wifi simultaneously. Is this correct?

No, both can be used at the same time but only on 2.4GHz even try to keep the channels matched up. Older install that used [dhcpcd](https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/50-hostapd#L17-L44) needed a reboot, might have better results now with [sysd-net-dis](https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/50-hostapd#L17-L44) being used with RasPiOS.

jvonau, 2025-01-19:
Sample logging
>COMMAND: /usr/bin/sudo journalctl -b 0 -u networkd-dispatcher    # networkd-dispatcher log
 
Jan 16 10:51:06 box systemd[1]: Starting networkd-dispatcher.service - Dispatcher daemon for systemd-networkd...
Jan 16 10:51:12 box networkd-dispatcher[741]: NET-DISP-unmanaged lo carrier
Jan 16 10:51:13 box networkd-dispatcher[765]: NET-DISP-unmanaged eth0 off
Jan 16 10:51:15 box networkd-dispatcher[776]: NET-DISP-unmanaged wlan0 off
Jan 16 10:51:16 box networkd-dispatcher[779]: NET-DISP-configured br0 no-carrier
Jan 16 10:51:17 box networkd-dispatcher[836]: NET-DISP-unmanaged ap0 no-carrier
Jan 16 10:51:17 box networkd-dispatcher[841]: NET-DISP-unmanaged eth0 no-carrier
Jan 16 10:51:17 box networkd-dispatcher[845]: NET-DISP-unmanaged wlan0 no-carrier
Jan 16 10:51:17 box networkd-dispatcher[846]: NET-DISP-WiFi wlan0 no-carrier
Jan 16 10:51:17 box networkd-dispatcher[850]: NET-DISP-unmanaged wlan0 dormant
Jan 16 10:51:16 box systemd[1]: Started networkd-dispatcher.service - Dispatcher daemon for systemd-networkd.
Jan 16 10:51:17 box networkd-dispatcher[853]: NET-DISP-unmanaged wlan0 carrier
Jan 16 10:51:17 box networkd-dispatcher[857]: NET-DISP-WiFi wlan0 carrier
Jan 16 10:51:17 box networkd-dispatcher[503]: WARNING:Unknown index 6 seen, reloading interface list
Jan 16 10:51:18 box networkd-dispatcher[909]: NET-DISP-unmanaged tailscale0 off
Jan 16 10:51:18 box networkd-dispatcher[914]: NET-DISP-unmanaged tailscale0 carrier
Jan 16 10:51:18 box networkd-dispatcher[954]: NET-DISP-unmanaged wlan0 degraded
Jan 16 10:51:18 box networkd-dispatcher[996]: NET-DISP-unmanaged wlan0 routable
Jan 16 10:51:19 box networkd-dispatcher[997]: NET-DISP-WiFi wlan0 routable
Jan 16 10:51:19 box networkd-dispatcher[997]: Using channel 6 for carrier
Jan 16 10:51:19 box networkd-dispatcher[997]: Hostapd set for 6
Jan 16 10:51:19 box networkd-dispatcher[997]: Upstream Channel greater than 13 or is the same - not changing hostapd.conf

tim-moody, 2025-01-19:
>Is anybody else having issues with using passwords with hostapd?

I often set a password to protect my home network without issue.

nzola, 2025-01-20:
I took time the last weekend to test iiab wifi and hotspot both running, and this is the results:

1) installed iiab-rpi4 destop server (with monitor) through wifi to my home router, 2.4 GHz and 5 GHz were enabled.
	a) iiab operated very well direct on the rpi4 server without any problem.
	b) iiab wifi hotspot name did not show up on the client HP Pavillon desktop wifi list.
	c) iiab wifi hotspot name did show up on Motorola cellphone wifi list, but refused to connect on the network.
	d) iiab wifi hotspot name did show up and connected to the network on Lenovo laptop, but browsing on iiab webpage was intermittent stopped several time and generated "can't reach this page" error.
	e) iiab wifi hotspot name did show up on Samsung cellphone wifi list, but would not open iiab browser.

2) Kept the same install as in number (1) here above, but disabled 5 GHz on my router.
	a) iiab operated very well direct on the rpi4 server without any problem.
	b) iiab wifi hotspot name did show up and connected on the client HP Pavillon desktop wifi list, but browsing on iiab webpage was intermittent stopped several time and generated "can't reach this page" error.
	c) iiab wifi hotspot name did show up on Motorola cellphone wifi list, but refused to connect on the network.
	d) iiab wifi hotspot name did show up and connected to the network on Lenovo laptop, but browsing on iiab webpage was intermittent stopped several time and generated "can't reach this page" error.
	e) iiab wifi hotspot name did show up and connected on Samsung cellphone wifi list, but browsing on iiab webpage was intermittent stopped several time and generated "can't reach this page" error.

3) ran new install, iiab-rpi4 destop server (with monitor) through wifi to my home router, 5 GHz was disabled. (clients operation the same as on number (2) here above.
	a) iiab operated very well direct on the rpi4 server without any problem.
	b) iiab wifi hotspot name did show up and connected on the client HP Pavillon desktop wifi list, but browsing on iiab webpage was intermittent stopped several time and generated "can't reach this page" error.
	c) iiab wifi hotspot name did show up on Motorola cellphone wifi list, but refused to connect on the network.
	d) iiab wifi hotspot name did show up and connected to the network on Lenovo laptop, but browsing on iiab webpage was intermittent stopped several time and generated "can't reach this page" error.
	e) iiab wifi hotspot name did show up and connected on Samsung cellphone wifi list, but browsing on iiab webpage was intermittent stopped several time and generated "can't reach this page" error.

4) kept the same install as on number (3) here above, but clicked on FORGET WIFI ble button iiab admin console->wifi control server-> wifi control.
	- all the above clients worked without any trouble.

NOTE: I experienced this exact situation about 6 months ago, and the solution was to turn off iiab wifi by clicking FORGET WIFI in the admin console. I did not need to disable 5 GHz on my router.

Conclusion: For my experience, the problem is running iiab rpi4 server with both wifi and hotspot are enabled. For those who say it works, I will recommend to not just open the iiab browser, but to also run a minimum 30 minutes video and see if it goes thru.

tim-moody, 2025-01-20:
>3. ran new install, iiab-rpi4 destop server
- This was the most recent raspios?
- There was no wired ethernet connection?
- You used the raspios desktop to create the wifi connection to your router?

>kept the same install as on number (3) here above, but clicked on FORGET WIFI ble button iiab admin console->wifi control server-> wifi control.
- The setting of wifi_up_down was not changed?
- Did the desktop show no connection to the router?
- Do you know what the setting was/is for rpi3bplus_rpi4_wifi_firmware?

>all the above clients worked without any trouble.
- All of the tested clients could access content on the server without problems?
- But additional content could not be added to the server with Admin Console?

>For those who say it works, I will recommend to not just open the iiab browser, but to also run a minimum 30 minutes video and see if it goes thru.
- Good point.

nzola, 2025-01-20:
> > 3. ran new install, iiab-rpi4 destop server
> 
> * This was the most recent raspios? **YES**
> * There was no wired ethernet connection? **NOT ALL**
> * You used the raspios desktop to create the wifi connection to your router? **YES**
> 
> > kept the same install as on number (3) here above, but clicked on FORGET WIFI ble button iiab admin console->wifi control server-> wifi control. **YES CORRECT**
> 
> * The setting of wifi_up_down was not changed? **NO**
> * Did the desktop show no connection to the router? **NO**
> * Do you know what the setting was/is for rpi3bplus_rpi4_wifi_firmware? NO. I DID NOT CHECK
> 
> > all the above clients worked without any trouble.
> 
> * All of the tested clients could access content on the server without problems? YES. AFTER I CLICKED FORGET WIFI
> * But additional content could not be added to the server with Admin Console? **I DID NOT VERIFY NOW, but I know from the past experience when FORGET WIFI is cclicked, iiab sever would not be able to add more contents, I had to use ethernet cable connected to the router. I wll verify tonight to make sure.** 
> 
> > For those who say it works, I will recommend to not just open the iiab browser, but to also run a minimum 30 minutes video and see if it goes thru.
> 
> * Good point.

jvonau, 2025-01-21:
Perhaps there was some channel hopping by the upstream router because of a crowed wifi environment? That might cause hostapd to be restarted as the channels are changed or outright break hostapd. I designed wifi_up_down specifically for desktops to be able play nice **during installs** and not totally disable upstream wifi capabilities in the future as in the case of using just the bare wifi (wlan0) interface for hostapd. If you use just wlan0 for hostapd all upstream wifi info must be removed and hostapd would need to be totally disabled should you want to use upstream wifi. As I suggested earlier and many years ago it highly recommended to disable upstream wifi connection info before shipping for use. Guess that never made it into the docs as I'm often discounted and ignored.

nzola, 2025-01-21:
> * This was the most recent raspios?
> * There was no wired ethernet connection?
> * You used the raspios desktop to create the wifi connection to your router?
>
>
> * The setting of wifi_up_down was not [...]

Do you know what the setting was/is for rpi3bplus_rpi4_wifi_firmware? **I don't know how to check this. Maybe you can see it on this iiab-diagnostics I created before clicking on FORGOT ALL WIFI**
PUBLISHING TO URL...
https://paste.centos.org/view/ba332399

But additional content could not be added to the server with Admin Console? **CORRECT. after clicking FORGET ALL WIFI the iiab-sever loses its wifi connection to my home router. No new contents can be downloaded. **

jvonau, 2025-01-21:
Thank you for ba332399 interesting points below for discussion. Day job prevents me from posting my complete thoughts all at once but I will edit this post from time to time in the next couple of days as time permits.
>COMMAND: /usr/bin/journalctl -b | grep wpa_supplicant    # wpa_supplicant log since boot
 Jan 20 06:09:08 box systemd[1]: Starting wpa_supplicant.service - WPA supplicant...
Jan 20 06:09:09 box wpa_supplicant[687]: Successfully initialized wpa_supplicant
Jan 20 06:09:09 box systemd[1]: Started wpa_supplicant.service - WPA supplicant.
Jan 20 06:09:19 box wpa_supplicant[687]: wlan0: WPS-CANCEL
Jan 20 06:09:19 box wpa_supplicant[687]: wlan0: Trying to associate with SSID '1155'
Jan 20 06:09:19 box wpa_supplicant[687]: wlan0: Associated with 3e:82:c0:fe:10:d2
Jan 20 06:09:19 box wpa_supplicant[687]: wlan0: CTRL-EVENT-CONNECTED - Connection to 3e:82:c0:fe:10:d2 completed [id=0 id_str=]
Jan 20 06:09:19 box wpa_supplicant[687]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Jan 20 06:09:19 box wpa_supplicant[687]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=CA

> country CA: DFS-FCC

> COMMAND: /usr/bin/dmesg | grep Firmware    # (Wi-Fi) firmware boot diagnostics

Boot time change of country code but the cmline options passed from cmdline.txt were not captured with the recent changes from #3899 #3901, this is a step backwards. 

> COMMAND: /usr/bin/journalctl -b | grep NetworkManager | head -100    # NetworkManager log since boot
 Jan 20 06:09:08 box systemd[1]: Starting NetworkManager.service - Network Manager...
Jan 20 06:09:10 box NetworkManager[714]: <info>  [1737349750.7171] NetworkManager (version 1.42.4) is starting... (boot:618d8c21-f996-47a5-90de-1d7a025efc7e)
Jan 20 06:09:10 box NetworkManager[714]: <info>  [1737349750.7172] Read config: /etc/NetworkManager/NetworkManager.conf (lib: no-mac-addr-change.conf) (etc: ap0-manage.conf)
Jan 20 06:09:10 box systemd[1]: Started NetworkManager.service - Network Manager.
Jan 20 06:09:10 box NetworkManager[714]: <info>  [1737349750.7376] bus-manager: acquired D-Bus service "org.freedesktop.NetworkManager"
Jan 20 06:09:10 box systemd[1]: Starting NetworkManager-wait-online.service - Network Manager Wait Online...
Jan 20 06:09:10 box NetworkManager[714]: <info>  [1737349750.9150] manager[0x55bb6b7000]: monitoring kernel firmware directory '/lib/firmware'.
Jan 20 06:09:10 box NetworkManager[714]: <info>  [1737349750.9152] monitoring ifupdown state file '/run/network/ifstate'.
Jan 20 06:09:11 box dbus-daemon[520]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.11' (uid=0 pid=714 comm="/usr/sbin/NetworkManager --no-daemon")
Jan 20 06:09:11 box iiab-test-wifi[755]: NetworkManager
Jan 20 06:09:11 box NetworkManager[714]: <info>  [1737349751.5019] hostname: hostname: using hostnamed
Jan 20 06:09:11 box NetworkManager[714]: <info>  [1737349751.5020] hostname: static hostname changed from (none) to "box"
Jan 20 06:09:11 box NetworkManager[714]: <info>  [1737349751.5030] dns-mgr: init: dns=systemd-resolved rc-manager=unmanaged (auto), plugin=systemd-resolved
Jan 20 06:09:11 box NetworkManager[714]: <info>  [1737349751.5057] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
Jan 20 06:09:11 box NetworkManager[714]: <info>  [1737349751.5062] manager[0x55bb6b7000]: rfkill: Wi-Fi hardware radio set enabled
Jan 20 06:09:11 box NetworkManager[714]: <info>  [1737349751.5062] manager[0x55bb6b7000]: rfkill: WWAN hardware radio set enabled
Jan 20 06:09:12 box NetworkManager[714]: <info>  [1737349752.4765] Loaded device plugin: NMBluezManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-bluetooth.so)
Jan 20 06:09:12 box NetworkManager[714]: <info>  [1737349752.6716] Loaded device plugin: NMWifiFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wifi.so)
Jan 20 06:09:12 box NetworkManager[714]: <info>  [1737349752.7914] Loaded device plugin: NMAtmManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-adsl.so)
Jan 20 06:09:13 box NetworkManager[714]: <info>  [1737349753.0117] Loaded device plugin: NMTeamFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-team.so)
Jan 20 06:09:13 box NetworkManager[714]: <info>  [1737349753.1166] Loaded device plugin: NMWwanFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wwan.so)
Jan 20 06:09:13 box NetworkManager[714]: <info>  [1737349753.1173] manager: rfkill: Wi-Fi enabled by radio killswitch; enabled by state file
Jan 20 06:09:13 box NetworkManager[714]: <info>  [1737349753.1175] manager: rfkill: WWAN enabled by radio killswitch; enabled by state file
Jan 20 06:09:13 box NetworkManager[714]: <info>  [1737349753.1176] manager: Networking is enabled by state file
Jan 20 06:09:13 box dbus-daemon[520]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.11' (uid=0 pid=714 comm="/usr/sbin/NetworkManager --no-daemon")
Jan 20 06:09:13 box systemd[1]: Starting NetworkManager-dispatcher.service - Network Manager Script Dispatcher Service...
Jan 20 06:09:13 box systemd[1]: Started NetworkManager-dispatcher.service - Network Manager Script Dispatcher Service.
[... cut ...]
ap0            wifi      unmanaged               --            
 
NetworkManager is aware of systemd-network (connected (externally))

>COMMAND: /usr/bin/networkctl    # systemd-networkd status
 IDX LINK       TYPE     OPERATIONAL SETUP
  1 lo         loopback carrier     unmanaged
  2 eth0       ether    no-carrier  unmanaged
  3 wlan0      wlan     routable    unmanaged
  4 br0        bridge   routable    configured
  5 ap0        wlan     enslaved    unmanaged
  6 tailscale0 none     carrier     unmanaged

NM has not taken control of br0 and ap0

nzola, 2025-01-21:
I have thought this might be important to know for your troubleshooting; when both wifi and hotspot are running on the iiab server, it only affects the clients connecting to the server through wifi, **but not those connecting through the server ip address assigned by the router.**

tim-moody, 2025-01-21:
How about access to Admin Console via the router issued IP, does that eliminate the "can't reach this page" errors?

tim-moody, 2025-01-21:
So I think my problem is that I was looking for a software bug, and this is a hardware problem. I haven't tested on rpi4 for some time and of course never in your environment. You have persuaded me that an upstream wifi connection and hotspot cannot be used at the same time in this environment. You have found a good way to turn off the upstream connection with Admin Console. (I am not sure it will work if your upstream connection is 5GHz, but then the hotspot is probably not on either.)  You can probably also turn the hotspot off and on there as well. So an admin can probably find a work around for this problem as long as they are aware of how to do it. Having a desktop will also help in not getting locked out.

nzola, 2025-01-22:
> How about access to Admin Console via the router issued IP, does that eliminate the "can't reach this page" error?
Yes. It does eliminate "Can't reach this page"