iiab/iiab #3789: new maps repository not responding but old one does.

Abdulrahman392011, 2024-08-16:
I am pretty sure this isn't a problem with my device.

I running ubuntu24 on 2015 8gigs of ram dell laptop.

no respond when I try to download maps entire world or any part.

I looked into the script and took the download links and sure enough, they aren't responding.
but then i did the same for the 2019 legacy script and took the download link and i downloaded the 75GB file.

please validate the new links.

Abdulrahman392011, 2024-08-31:
Hi, i will send you the links that work.

But basically the maps times out every time I try to download it or even try to download the individual files of the regions using the links that are up to date

Abdulrahman392011, 2024-08-31:
https://github.com/iiab/iiab/wiki/IIAB-Maps

so if you scroll down in the link above you'll find [new catalog, legacy catalog, older catalog]
press on each one and try to find a link that you can copy and do so. you'll find that the new catalog direct you to a not found page while the legacy and older catalogs start downloading normally.

Abdulrahman392011, 2024-08-31:
it's worth noting that in links that work if you try to download the 75 gigabytes world file it will time out at around 45 to 50 gigabytes that is if you're downloading at speed of 3.5 megabytes a second.

for that i try to download the regions separately in hopes that it won't time out but still the matter of the installation files for maps in iiab itself as it uses the up to date links which aren't working to install the software that runs the regions. so even if I had the regions it won't work.

tim-moody, 2024-08-31:
@Abdulrahman392011 unfortunately there have been a number of hands on this functionality with detritus left behind. 

Am I right in assuming you are trying to download tiles directly rather than with Admin Console? I don't recommend or support that. Can you comment on whether the downloads using Admin Console work and if not which regions don't and what errors you encounter. I last did a world download a couple of months ago without a problem.

Abdulrahman392011, 2024-08-31:
Hi Tim, so here is the full story, concisely. 

I download IIAB without configuring it. used the default settings which don't automatically download the maps.

after the installation of IIAB new user is add to ubuntu and that is the one it automatically login to. (the admin console user isn't automatically logged in. but I can see it if I logged out)

then I have to enable the service from configure in http://127.0.0.1/admin/

that last part fails. it tries to download something but fails everytime. (I can see it's progress in utilities-->display job status)

I believe it's trying to download the installation files for Maps which in turn can read the regions files.

I get kinda intrigued and want to see what's wrong with it. and so I do the above message: 

> https://github.com/iiab/iiab/wiki/IIAB-Maps
>
> so if you scroll down in the link above you'll find [new catalog, legacy catalog, older catalog] press on each one and try to find a link that you can [...]

tim-moody, 2024-08-31:
@Abdulrahman392011 can you send me the job status for the failed map download. I'll try to have a look, but I will be travelling and away from communications. It is possible that we have a problem with Wasabi or one of the S3 providers. My memory is that the catalog links are misleading to say the least.

Abdulrahman392011, 2024-08-31:
correction! so when I check the IIAB maps in the configure tab in the admin console, It does succeed and no error is displayed in the job status. 
However when I go to (install content)-->(get maps addons) it tells me that the base maps isn't installed and I should install them from (get maps regions)

now when I go to (get maps regions) and only have the Required Base Files checked:
Planet to Zoom 10 1.96 GB
Satellite to Zoom 9 1.19 GB

then click (install selected region) that's where it starts four jobs:
scripts/osm-vect_v2_finish_install.py satellite_z0-z9_2020.mbtiles
scripts/osm-vect_v2_finish_install.py osm-planet_z0-z10_2020.mbtiles
scripts/osm-vect_v2_install_step2.sh satellite_z0-z9_2020.mbtiles
scripts/osm-vect_v2_install_step2.sh osm-planet_z0-z10_2020.mbtiles

while monitoring my network I find nothing is being downloaded and also monitoring the cpu and find that it's not doing anything. after sometime it eventually time out and give an error. 
it takes sometime for it to give the error it remains stuck on SCHEDULED for a few hours.
I just made a fresh job and I will wait till it eventually give me the error and send it to you.

Abdulrahman392011, 2024-08-31:
now there is another two jobs that appeared from no where under the four jobs I mentioned above. I have enunciate that they appeared under not above them. like they started first but there was a delay in the job status report or something.

the two jobs are named:
rclone -P copy contabo:iiab-maps/osm-planet_z0-z10_2020.mbtiles /library/working/maps/
rclone -P copy contabo:iiab-maps/satellite_z0-z9_2020.mbtiles /library/working/maps/

tim-moody, 2024-08-31:
When cmdsrv comes up it tries to restart any incomplete jobs. I'm wondering if somehow cmdsrv didn't sense that maps were installed. Maps download would be blocked until maps service is marked installed.

Abdulrahman392011, 2024-08-31:
that's what I got wrong the first time. I was talking from memory and I mistakenly said that the service change is failing but no the jobs succeed when I check the IIAB maps in the configure tab and then press (install configured options). it give up two jobs both succeed.

the issue comes when I try to download the base maps that required (get maps regions)

here's a screenshot of my job status at the moment:
[image]

tim-moody, 2024-08-31:
So possibly problem with contabo

I have been thinking about another host but haven't gotten around to it.

holta, 2024-08-31:
@Abdulrahman392011

Hypothetically, is there any possibility the `rclone` downloads might be blocked by your ISP?

```
rclone -P copy contabo:iiab-maps/osm-planet_z0-z10_2020.mbtiles /library/working/maps/

rclone -P copy contabo:iiab-maps/satellite_z0-z9_2020.mbtiles /library/working/maps/
```

(Certainly the above commands have been working well for others in recent weeks.)

Abdulrahman392011, 2024-08-31:
I am in Egypt could it be that the links aren't responding to me because of that. but I thought of that and used a vpn to direct me germany or england but the same issue occurs.
are you certain that the links are valid.

Abdulrahman392011, 2024-08-31:
it's not as big of a deal as it seems. just have a rollback for those that have the issue happen with them. the old links respond just fine with me. I am guessing the same host contabo. so this roles out that it's because of my location.

Abdulrahman392011, 2024-08-31:
sometimes people do mistakes on the server side. maybe someone accidentally deleted the files or some type of bad sector on the ssd. you never know.

holta, 2024-08-31:
> are you certain that the links are valid.

I just ran the command below, and the download worked, for me in the US anyway!

```
rclone -P copy contabo:iiab-maps/satellite_z0-z9_2020.mbtiles /library/working/maps/
```

@Abdulrahman392011 I'm interested in helping you work around this with VPN options if you have time to write me privately?

holt@unleashkids.org

(As many countries now have ISP-level blocks, sometimes even by accident, and it's important to help educators understand their options!)

Abdulrahman392011, 2024-08-31:
alright so, I tried to run the same command above and here is what I got:

2024/08/31 11:31:11 NOTICE: Config file "/home/abdelrahman/.config/rclone/rclone.conf" not found - using defaults
2024/08/31 11:31:11 Failed to create file system for "contabo:iiab-maps/satellite_z0-z9_2020.mbtiles": didn't find section in config file

holta, 2024-08-31:
@Abdulrahman392011 what happens if you run the `rclone` command as root?

(Run `sudo -i` to become root?)

Abdulrahman392011, 2024-08-31:
it looks like there is a bug. I ran (sudo apt install rclone) and it actually downloaded it. it wasn't downloaded before. but in all fairness it output of the command above is the same before installing rclone and after installing it. once again it seems that my old laptop is buggy.  

when I try to run the command using sudo or sudo -i it keeps loading but nothing is going on with the cpu or the network in the (system monitor). it just keeps loading and nothing happens

Abdulrahman392011, 2024-08-31:
well it's not a big deal. it works generally speaking. my laptop is too old. I am planning on getting a new one in the future. also the maps was not needed at the moment for me. so it's not worth the time and effort. 

I am working on a way of giving the AI model long term memory so it becomes useful for local use. when I am done with it will share it with you guys.

holta, 2024-08-31:
@Abdulrahman392011 we install rclone **without** using `apt` to get a better version, so your laptop may be acting properly 😀 

e.g. `curl -s https://rclone.org/install.sh | bash -` installs `/usr/bin/rclone`

As you can see here:
https://github.com/iiab/iiab-admin-console/blob/61d93594dc9084e68f702055dfda4ce994e7b14d/roles/cmdsrv/tasks/main.yml#L40-L43

➡️ Regardless, definitely please write me if you get the chance (holt@unleashkids.org).

As there are concrete & practical ways to solve such issues with VPN-like techniques.

Abdulrahman392011, 2024-08-31:
sure thing, I will be contacting you as soon as possible. 

grateful for your time.

tim-moody, 2024-08-31:
I'm not clear how the rclone config got created but rclone didn't get installed.

Abdulrahman392011, 2024-08-31:
so basically the way curl installation is done installs a newer version but when I try to install it using apt it tries to install an older version and somehow misses the newer version like it's not even installed. that's the current conciseness. funny but it's the best we got. it's an old laptop and have bad rep  in being buggy as hell. if I were to think about every bug this laptop had I would have been in a mental asylum by now , haha.

Abdulrahman392011, 2024-09-06:
[memory.zip](https://github.com/user-attachments/files/16902511/memory.zip)

ok, so this is a working version of what I mean by long-term memory. it uses sqlite as a relational database for the entities and their related thoughts that the model encounters. it then augment the user's prompt by putting relevant memories in the prompt before it process it. for development sake I made print the entities and thoughts it detects as it register it in the database. tell me what you guys think. this is different than what they use normally which is basic rag on the chat history which have the flaw of missing the memory if there is no obvious reason to conduct a search for it. the relational database method always puts related thoughts because of the detection of entities. so it always remembers without consuming any compute resources. tell me what you think?

closed 2024-09-07 by Abdulrahman392011