iiab/iiab #3790: older versions of IIAB are not available

Abdulrahman392011, 2024-08-16:
I run the IIAB on ubuntu24 native vs run IIAB on ubuntu24 in VM running on ubuntu20.

there is a big difference in download speed. 500KB vs 3.5MB.

I'm guessing that my 2015 laptop has some old drivers that are available in the host ubuntu20 and that's why when runs in a VM on ubuntu20 the drivers optimize the speed not to mention that it maintains the connection much better.

the point of my story is that you guys should keep the support for older OS available. as it might make a difference for many of us. most of us have old machines and so this isn't exactly a isolated incidence.

as for me I simply download them from the virtual machine, extract them, put them on a external hard drive and then put them again in the library of the native OS. well, at least I backup the library. haha

holta, 2024-08-16:
> 500KB vs 3.5MB

Hi @Abdulrahman392011

Could you perhaps try Ubuntu Server "24.04.1" which is is less than 1GB to download at https://cdimage.ubuntu.com/ubuntu-server/noble/daily-preinstalled/current/ ?

Example: https://cdimage.ubuntu.com/ubuntu-server/noble/daily-preinstalled/current/noble-preinstalled-server-amd64.img.xz (969 MB)

FYI many IIAB apps have proven to be _nearly impossible_ to make work on older OS's, as software maintenance costs are in fact very large, and we are all volunteers working hard doing our best, which is sometimes often not obvious to everyone :)

Abdulrahman392011, 2024-08-16:
I sure will give it a try, but I think it's a matter of decision on ubuntu's side. they must have made the decision to make the OS work on some new architecture for better performance or security and they just dropped the old drivers that don't support that. they too have limited resources being a non profit.

I think running it on a VM is the most reliable way so far. keep it in mind if any one else ask for a solution for slow downloads. but tell them to extract it from the VDI file with an extracting app like P7Zip. it extract with speed of 200MB instead of the VM copying speed of 10MB or something stupid like that. they can navigate to the library file with the app and extract only the file they need without running the VM. doing it from the host is much faster and then copy it to native ubuntu24 that they have installed on a another drive on their machine. so they are better copying it from ubuntu20 to the drive of ubuntu24. if they couldn't do that then they will have to use an external drive but ubuntu 24 have some issues with mounting ntfs drive so just search the internet of how to mount ntfs drives on ubuntu24. they might get problems with symlinks  afterwards if they didn't do it correctly. nothing that can't be fixed. it's a bit of a hassle but it works eventually on old machines.

Abdulrahman392011, 2024-08-21:
ok, it's been a couple of days. I downloaded ubuntu server live the 2.5 gigabytes iso file, installed it then installed ubuntu-desktop which was like another 2.5 gigabytes. not exactly what you told me to do, but close enough. I try to download IIAB and it gives me an error in step 3.

TASK [nginx : Enable & (Re)Start 'nginx' systemd service, if nginx_enabled] ****************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service nginx: Job for nginx.service failed because the control process exited with error code.\nSee \"systemctl status nginx.service\" and \"journalctl -xeu nginx.service\" for details.\n"}

PLAY RECAP *********************************************************************************************************************
127.0.0.1                  : ok=32   changed=2    unreachable=0    failed=1    skipped=7    rescued=0    ignored=0   

now I am not sure if the internet speed issue that I was having was solved. but it does feel like it. the speed of downloading in the first two steps was reading 2 and 3 megabytes while in the ubuntu 24 desktop version was reading 500 and 600 kilobytes.

Abdulrahman392011, 2024-08-21:
I am not sure why I am getting an error in installing IIAB in ubuntu24 server version even though I didn't have this issue while installing it in the ubuntu24 desktop version. something isn't adding up. but it's not that big of a deal, I can just revert back to the desktop version I got installed on the machine. 

on a separate note I was working on enhancing the search ability on IIAB using AI. I would like to share the python file with you and see if you have any thoughts. it's not finished yet but it works as a prototype kinda thing.

Abdulrahman392011, 2024-08-21:
[links_test3.zip](https://github.com/user-attachments/files/16686253/links_test3.zip)

holta, 2024-08-21:
> TASK [nginx : Enable & (Re)Start 'nginx' systemd service, if nginx_enabled] ****************************************************
> fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to [...]

Did you run `systemctl status nginx.service` as the error advises?  Please reply with that output?

Also run `journalctl -u nginx` if necessary?

Also please reply with... the URL output by:

`sudo iiab-diagnostics`

Abdulrahman392011, 2024-08-21:
× nginx.service - A high performance web server and a reverse proxy server
     Loaded: loaded (/usr/lib/systemd/system/nginx.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Wed 2024-08-21 05:12:17 UTC; 4min 38s ago
       Docs: man:nginx(8)
    Process: 33019 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
    Process: 33085 ExecStart=/usr/sbin/nginx -g daemon on; master_process on; (code=exited, status=1/FAILURE)
        CPU: 99ms

Aug 21 05:12:15 box nginx[33019]: 2024/08/21 05:12:15 [info] 33019#33019: Using 131072KiB of shared memory for nchan in /etc/ng>
Aug 21 05:12:15 box nginx[33085]: 2024/08/21 05:12:15 [emerg] 33085#33085: bind() to 0.0.0.0:80 failed (98: Address already in >
Aug 21 05:12:15 box nginx[33085]: 2024/08/21 05:12:15 [emerg] 33085#33085: bind() to 0.0.0.0:80 failed (98: Address already in >
Aug 21 05:12:16 box nginx[33085]: 2024/08/21 05:12:15 [emerg] 33085#33085: bind() to 0.0.0.0:80 failed (98: Address already in >
Aug 21 05:12:16 box nginx[33085]: 2024/08/21 05:12:15 [emerg] 33085#33085: bind() to 0.0.0.0:80 failed (98: Address already in >
Aug 21 05:12:17 box nginx[33085]: 2024/08/21 05:12:15 [emerg] 33085#33085: bind() to 0.0.0.0:80 failed (98: Address already in >
Aug 21 05:12:17 box nginx[33085]: 2024/08/21 05:12:15 [emerg] 33085#33085: still could not bind()
Aug 21 05:12:17 box systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Aug 21 05:12:17 box systemd[1]: nginx.service: Failed with result 'exit-code'.
Aug 21 05:12:17 box systemd[1]: Failed to start nginx.service - A high performance web server and a reverse proxy server.
~
~

holta, 2024-08-21:
> on a separate note I was working on enhancing the search ability on IIAB using AI. I would like to share the python file with you and see if you have any thoughts.

Sure!  Not an easy task but important!

holta, 2024-08-21:
> Aug 21 05:12:15 box nginx[33085]: 2024/08/21 05:12:15 [emerg] 33085#33085: bind() to 0.0.0.0:80 failed (98: Address already in >

Any chance the OS installed Apache (or another web server) which is already using port 80?

https://stackoverflow.com/questions/14972792/nginx-nginx-emerg-bind-to-80-failed-98-address-already-in-use/16798023

Abdulrahman392011, 2024-08-21:
this is how what I done on the python code.
I used a generative model (gemma2:2b) 1.6 gigabytes to extract the search items and name them.

then I used an embedding model (mxbai-embed-large) 700 megabytes to look at the names of the books that came back in the search in the 70 gigabytes gutenberg library of books and choose the top 2 that might be of relevance to the topic of the search.

this were we need to pause and think cause it takes a lot of compute if we just put the content of those 2 books for the larger embedding model.

then I used a smaller model (nomic-embed-text) 270 megabytes to filter the content of those two books and choose the top 50 chunks that are of relevance to the search topic but it still takes too much compute.

I wanted to use the (snowflake-arctic-embed:22m) 50 megabytes to filter the content and then give out like 10 pages for the larger embedding model.

then the larger embedding model should give the generative model then the reply.

Abdulrahman392011, 2024-08-21:
the issue is for some reason the (snowflake-arctic-embed:22m) 50 megabytes model isn't working on my machine. it gives an error. I am taking with ollama about it and it seem to be a bug of some sort that is related to my machine. they confirmed it works fine on their machines.

Abdulrahman392011, 2024-08-21:
you're right they were listening to the same port. I got past that now
funny the other things that were listening was nginx itself but the default and the defaul:# 
weired!

Abdulrahman392011, 2024-08-21:
if you want to try the python file yourself, you'll need to install ollama and chromadb and requests and ..... typically all the import modules at the top of the file. also you will need to pull the models from ollama. the main models I mentioned in the previous comments. I am still working on the efficiency thing. the idea is simple use a small model on a larger content and choose the relevant information then give it (medium content) to a medium model, then (larger model) with small content. spitting out 1 or half a page to the generative model to handle. this way it will not use a lot of compute and will be possible for a small machine with 8 gigabytes of ram to handle the request in less than a minute. hopefully 15 to 30 seconds.

Abdulrahman392011, 2024-08-21:
I'm running into another error at stage 8 with installing mariadb-server. I am not sure what is wrong but I've been at it for like an hour now and I just can't get it to install.

perhaps I should just revert to the ubuntu24 desktop version that I have IIAB installed on and focus on the python code till I get to work correctly. it seems like a better investment of energy since I already have a way of downloading IIAB content and it's the bugs are isolated incidence and related to my old machine. it feels like it's not worth all our time for now. maybe I will get back to it later out of curiosity. but for now I will get back to ollama's team and see if I can get the small model to work. fingers crossed.

holta, 2024-08-21:
> error at stage 8 with installing mariadb-server.

Please send the URL output by `sudo iiab-diagnostics`

> spitting out 1 or half a page to the generative model to handle. this way it will not use a lot of compute and will be possible for a small machine with 8 gigabytes of ram to handle the request in less than a minute. hopefully 15 to 30 seconds.

Of course 15-to-30 seconds is too much for most students (!)

But some researchers/librarians/teachers/parents would be very interested if search relevance ~ quality appears very high (?)

Abdulrahman392011, 2024-08-21:
well, I understand that 15-30 seconds is long but I am planning to make it a parallel search system. I won't put it in series. meaning the model will reply instantly and do the research i the background. then it will get back to this point when search results are out. 

it takes patience not time. but eventually  it will work better than just getting lost in a large library of books. once I get it to work with gutenberg library then I will make it work with Wikipedia and wiki-how and stackoverflow. 

in the mean time I will have to replace Ollama api with hugging face transformers library that works with pytorch or tensorflow. this will help bypass the limiation of Ollama's model collection.

patience not time.

Abdulrahman392011, 2024-08-21:
https://paste.centos.org/view/6f51470d

Abdulrahman392011, 2024-08-21:
this is the diagnostics 

https://paste.centos.org/view/6f51470d

holta, 2024-08-21:
> https://paste.centos.org/view/6f51470d

Question: Why is `apt install mariadb-server` failing on Line 2219 of the above pastebin?

IIAB installations require a bare OS — without NGINX and MariaDB (MySQL).

(Is it possible you might accidentally be trying to install IIAB onto an OS that already has these components/services?)

Abdulrahman392011, 2024-08-22:
when i try to update it tells me that there are dependencies missing and when i try to download the missing dependencies it tells me that other dependencies missing and when i try to install those it tells me that other .......... etc. 

mainly the python installation is flawed and there are missing components all over.

by the way Ollama team gave another small model that works, so there is no need to use Pytorch. this saves a lot of work. I'll just carry on with the same python code just replace the model names.

Abdulrahman392011, 2024-08-22:
alright I am done. it's well tuned to give one very good result for every search query. you will find it a big improvement from the regular search. it still takes 15-30 seconds but this performance is on my 9 year old laptop from 2015 and it's running on CPU only. ollama inference does support GPU on new machines; 3.7 compute capability and above from nvidia. They also support amd gpu. but not NPU from new machines, they default to GPU. on GPU it should take about 5 seconds on avarage. NPU will be almost instantly but that will be in a few years or so till it become widely supported. it's already included in new machines' chips but the software still does not support it.

Abdulrahman392011, 2024-08-22:
[search.zip](https://github.com/user-attachments/files/16708443/search.zip)

holta, 2024-08-22:
@Abdulrahman392011 can you summarize install tips here on GitHub, if possible, for those wanting to try it out?

Abdulrahman392011, 2024-08-22:
I added a readme.txt file to the search.zip file above. it have all the details and copy and paste commands to have it running in half an hour or so.

Abdulrahman392011, 2024-08-22:
I'm working on making it work with Wikipedia, Wiki-how and Stack-overflow as well. I made it chat a little about the search result at the end but the above version doesn't do that. it just provide the name of the book and the part that it thinks relevant to what you're searching for.

holta, 2024-08-23:
> I added a readme.txt file to the search.zip file above. it have all the details and copy and paste commands to have it running in half an hour or so.

search.zip's current readme.txt pasted in below — to help others understand:

```
assuming that you have IIAB and the zims file of Gutenberg library installed

############################################ requirments ############################################
copy and paste these in your terminal wait till each one is finished before pasting the other:
sudo apt install python3
sudo apt install python3-pip
pip install chromadb beautifulsoup4 requests langchain-text-splitters
curl -fsSL https://ollama.com/install.sh | sh
ollama pull gemma2:2b
ollama pull nomic-embed-text
ollama pull mxbai-embed-large:latest
ollama pull all-minilm:22m
#####################################################################################################

########################################## usage of the program ########################################
after you done installing the requirments above then run the links_test3.py file:
cd into the directory and copy and paste this command into the terminal:

python3 ./links_test3.py

wait until the model is loaded. it will press enter automatically and put you on a new line. 
then enter your search query.
########################################################################################################

############################################## enhancing performance ###########################################
if you would like to enhance the performance assuming that you have 8 gigabytes of RAM or more in your machine.
do the following:-

copy and paste this in your terminal:
sudo nano /etc/systemd/system/ollama.service

then under the [service] section write:
Environment="OLLAMA_MAX_LOADED_MODELS=5"

######################################### these are commands that will have to be repeated after every boot:

after you done the above, copy and paste these one by one in the terminal:-
curl http://localhost:11434/api/generate -d '{"model": "gemma2:2b", "keep_alive": -1}'

curl http://localhost:11434/api/embed -d '{"model": "mxbai-embed-large:latest", "keep_alive": -1}'

curl http://localhost:11434/api/embed -d '{"model": "nomic-embed-text", "keep_alive": -1}'

curl http://localhost:11434/api/embed -d '{"model": "all-minilm:22m", "keep_alive": -1}'

ps. you can make them automatically load on boot startup but it will be a waste of RAM as they clear themselves
from memory after 5 minutes. the above lines prevents that and keep them in memory till reboot.
##################################################################################################################
```

Abdulrahman392011, 2024-08-24:
hi again. I made it work with Wikipedia, Wiki-how, Guetnberg library and Stack-overflow.
I also made it give a link to the article or question or book it's talking about. and talk a little at the end about the content and organize the information related to the search that was found in it. I modified the readme.txt as the caveat of this version is that it will require the zims files for:

*wikipedia_en_all_maxi_2024-01

*gutenberg_en_all_2023-08

*wikihow_en_maxi_2023-03

*stackoverflow.com_en_all_2023-11

if any of them isn't installed the code will break when it tries to search in the missing source.
it will work if the search wasn't conducted in the missing zims file.
the reason that's a big deal is because these four zims files are a little over 300 gigabytes.
but they provide a comprehensive library for the Language model to be useful despite of it's 
small context window of 8000 tokens. it's response time is also a delayed beyond the 15-30 seconds window. now it's more 45-60 seconds but it doesn't feel longer as the link that it provide keeps you engaged till it reads the content and provide it's opinion. 

also minor enhancement instructions was add at the end of the readme.txt file, not obligatory by any means.

Abdulrahman392011, 2024-08-24:
[search 4 sources.zip](https://github.com/user-attachments/files/16735450/search.4.sources.zip)

tim-moody, 2024-08-24:
How long does the training take in case one of the zims is a newer version?

Abdulrahman392011, 2024-08-31:
No training is done. It's using the data from the zims file to augmente its response (RAG).

Technically speaking, it can work with all the zims files, but the code has to be modified for it to do so. 

read the code. It's pretty self-explanatory.

holta, 2024-08-31:
> No training is done. It's using the data from the zims file to augmente its response (RAG).

https://en.wikipedia.org/wiki/Retrieval-augmented_generation !

closed 2024-09-07 by Abdulrahman392011