iiab/iiab #3792: Test BalenaEtcher on M1 or M2 to clone IIAB microSD card

avni, 2024-08-23:
From @holta 

Can you try BalenaEtcher on M1 or M2 to see if it will clone an IIAB microSD card without too much hassle?

Or if not, suggest any amendment to FAQ #47 here?

https://wiki.iiab.io/index.php?title=FAQ&diff=793&oldid=792

FAQ #47 is:

"How do I back up, shrink & copy IIAB microSD cards?"

Requested date: 8/31

holta, 2024-08-23:
@EMG70 can you likewise take a look at BalenaEtcher on Windows, if you have time, this month ideally?

To see if it now offers a decent IIAB cloning approach for newcomers + newbies! 🙏

holta, 2024-08-23:
`FAQ #47` slightly cleaned up:

https://wiki.iiab.io/index.php?title=FAQ&diff=795&oldid=792

(If it needs to be corrected or cleaned up in coming weeks with a screenshot-or-otherwise LMK, Thanks!)

EMG70, 2024-08-27:
Steps for cloning an  SD of same size and model (Sandisk 128GB) using   BalenaEtcher on Windows 11 and SD card adapter (Ugreen  model used).
IIAB size 1 was built on the SD card.The process for cloning took 10 minutes.Both cards tested and working OK.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

The two SD cards are shown below after successful cloning as /dev/mmcblk/0p2 and /dev/sdd2
[image]

holta, 2024-08-31:
@avni has confirmed likewise (successfully duplicating / cloning an IIAB microSD card) using the macOS version of balenaEtcher !

holta, 2024-08-31:
Many thanks to @EMG70 and @avni for the hard work of verifying this!

closed 2024-08-31 by holta

avni, 2024-08-31:
It only took 4 adaptors to do the test on the M1 MacOS! So glad we were able to figure it out and make it work! ✅