iiab/iiab #3794: [Review requested] New FAQ: Can I install IIAB apps on an existing IIAB instance?

avni, 2024-08-26:
@holta Making you aware and requesting review of New FAQ: "_Can I install IIAB apps on an existing IIAB instance?_":
https://wiki.iiab.io/go/FAQ#Can_I_install_IIAB_apps_on_an_existing_IIAB_instance? (Currently `FAQ #30`)

(Context: My goal was to install Nextcloud on an existing IIAB instance. I saw the FAQ on "_Can I upgrade or reinstall IIAB apps?_" and tried executing runrole with the reinstall flag `(./runrole --reinstall nextcloud).` It failed and suggested trying again w/o the reinstall flag, which worked. I am thinking it will be helpful to have an FAQ for those who are not familiar with runrole, may not know which apps they want installed on IIAB from the get go, and end up incrementally installing apps.) 

Thank you! 🙏

holta, 2024-08-26:
Further context added:

https://wiki.iiab.io/index.php?title=FAQ&diff=806&oldid=804

(Is that good enough?)

avni, 2024-08-26:
TY! I didn’t know one could install apps through admin console. 

> Yes, if your IIAB is online! 

Does the IIAB have to be online to do #2 via the command line? 

The link to “What services (IIAB apps) are suggested during installation?” doesn’t work for me (the first “See <X>” link). I can fix it tomorrow.

holta, 2024-08-26:
> Does the IIAB have to be online to do 2. via the command line?

Yes.

> link to “What services (IIAB apps) are suggested during installation?” doesn’t work

Fixed: https://wiki.iiab.io/index.php?title=FAQ&diff=807&oldid=806

ASIDE: URL-encoding of punctuation is useful in other contexts (WhatsApp breaks URLs that end with a '?' for example) I've removed them in this case, as they don't really add anything here.

closed 2024-08-26 by holta

avni, 2024-08-26:
> Does the IIAB have to be online to do 2. via the command line?
> Yes.

Oh yes, because a shutdown of IIAB from Admin Console, shuts down the whole server not just the web application. Thank you!