iiab/iiab #3795: Does IIAB currently install on the new Edubuntu 24.04.1 ?

holta, 2024-08-29:
@EMG70 can you take a look in the coming week?

(If you have time to compare this to Mint 22?!)

- Ubuntu 24.04.1 LTS released, and Edubuntu 24.04.1 too:
  https://lists.ubuntu.com/archives/ubuntu-announce/2024-August/000304.html

- Edubuntu:
  https://edubuntu.org

holta, 2024-08-29:
@EMG70 it sounds like @avni too will give this a shot, so compare notes with her over coming days — to see how Edubuntu 24.04.1 compares to Mint 22 (both just released!)

Especially through the eyes of low-income schools / students / teachers / parents who lack IT assistance❓

holta, 2024-08-29:
Clarif from https://discourse.ubuntu.com/t/edubuntu-24-04-lts-released/44455 that Edubuntu 24.04.1's downloadable final release images is/are here:

- https://cdimages.ubuntu.com/edubuntu/releases/24.04/release/
- https://cdimages.ubuntu.com/edubuntu/releases/24.04/release/edubuntu-24.04.1-desktop-amd64.iso 5.8G

EMG70, 2024-08-30:
Edubuntu 24.04 LTS installed 
IIAB installed,I cannot see any issues so far,operates as quick as Linux Mint 22.

SUDO IIAB-DIAGNOSTICS - https://paste.centos.org/view/e746a6c3

$ cat /etc/os-release
PRETTY_NAME="Ubuntu 24.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.1 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
ed@box:~$

holta, 2024-08-30:
Really great news @EMG70! 🎸

1. Can you please run `more /etc/debian_version /etc/issue /etc/lsb-release /etc/os-release` (then hit [SPACE BAR] 3 times) to help us compare to Ubuntu 24.04.1?

```
$ more /etc/debian_version /etc/issue /etc/lsb-release /etc/os-release
::::::::::::::
/etc/debian_version
::::::::::::::
trixie/sid
::::::::::::::
/etc/issue
::::::::::::::
Ubuntu 24.04.1 LTS \n \l

::::::::::::::
/etc/lsb-release
::::::::::::::
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=24.04
DISTRIB_CODENAME=noble
DISTRIB_DESCRIPTION="Ubuntu 24.04.1 LTS"
::::::::::::::
/etc/os-release
::::::::::::::
PRETTY_NAME="Ubuntu 24.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.1 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
```

2. Paste your results onto this ticket, [fenced within triple-backticks](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks) if possible :pray:

EMG70, 2024-08-30:
''' $ more /etc/debian_version /etc/issue /etc/lsb-release /etc/os-release
::::::::::::::
/etc/debian_version
::::::::::::::
trixie/sid
::::::::::::::
/etc/issue
::::::::::::::
Ubuntu 24.04.1 LTS \n \l

::::::::::::::
/etc/lsb-release
::::::::::::::
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=24.04
DISTRIB_CODENAME=noble
DISTRIB_DESCRIPTION="Ubuntu 24.04.1 LTS"
::::::::::::::
/etc/os-release
::::::::::::::
PRETTY_NAME="Ubuntu 24.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.1 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-poli
cy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo '''

holta, 2024-08-30:
Ok, until we know more, Edubuntu is now mentioned as a recommended OS within http://FAQ.IIAB.IO :

https://wiki.iiab.io/index.php?title=FAQ&diff=814&oldid=813

PS If @avni and @EMG70 find that Edubuntu is not as well polished as Linux Mint 22, we can diminish the visibility of Edubuntu later.

EMG70, 2024-08-30:
From a users view, there are good interactive programs like Kalzium (Interactive Periodic table) but only limited to online servers ,it eventually  refers user to Wikipidea (only works when online) eg when one clicks  Atomic mass for Lithium. Not very helpful for offline servers.

[image]
[image]
[image]

holta, 2024-08-30:
In the end, shout if anyone sees any very substantial UX/UI advantages of Edubuntu 24.04.1 over Mint 22 for teachers/learners ??

Linux Mint 22 has a quick fiercely loyal user base of course, so few will likely run away unless Edubuntu offers very distinct advantages! 😆

Best Case If All 3-to-4 Prove Sustainable: "COOPetition" will help literally everybody, allowing ALL 3/4 together {Edubuntu, Mint, Raspberry PI OS, Ubermix} to now improve & build on each other 🙏

avni, 2024-08-31:
Amazing! I installed Edubuntu 24.04.1 today on a Beelink Mini PC. Comes out of the gate with a lot of kid-friendly apps - good place to start for more novice teachers. 

I installed Mint 22 (Wilma) on a multipass VM on the Beelink. Need to install the kid-friendly apps to be able to do an apples to apples comparison - that's the next step.

holta, 2024-08-31:
> I installed Mint 22 (Wilma) on a multipass VM

I think you mean Mint 22 in VirtualBox 7.1 Beta 2 VM 👍

(Multipass doesn't really allow VM's with graphical desktops!)

avni, 2024-08-31:
> I think you mean Mint 22 in a VirtualBox 7.1 Beta 2 VM 👍

Oy. 🤦‍♀️Yes, of course. VirtualBox 7.1 Beta 2 VM. Thanks.