iiab/iiab #3801: Unable to make menu item edits display on home page and unable to make removed menu items remain removed

stenevang, 2024-09-20:
### Expected
1) That edited menu items (edited text etc) which display edits in the Admin Console should also display the same edits in the Home page
2) That menu items removed by drag-drop should remain removed from the menu and not re-appear at each time the Admin console is entered. 

### Observed
1) Menu item edits (title text etc) show in the Admin console but will not show in the Home page. Pressing Save Menu and/or Update Home Menu makes no difference
2) Menu items removed by drag drop keep popping back into the menu in the Admin console. No change in terms of menu item removal is never seen in the Home Page. However, reordering of menu items works and is reflected in the Home page.

### iiab-diagnostics output:
https://paste.centos.org/view/3eb63d12

jvonau, 2024-09-20:
@tim-moody

holta, 2024-09-20:
Thanks @stenevang for writing.  @tim-moody should respond in coming days, if he hopefully has time.

tim-moody, 2024-09-24:
The only time I have seen this was when browser caching was involved. When menu definitions are changed the browser must be forced to reload the underlying json, not just the html. The reboot does this.

stenevang, 2024-09-25:
I have finally succeeded in making the menu edits apply. It may be that a very specific procedure must be followed, which could be this:
1. Admin Console > Content Menus > Edit Content Menus: Press "Load Menu".
2. Go to subtab Content Item List
3. Drag-drop menu items you do not want in the menu from the left list "Menu Items in Current Menu", to the right list, "Available Menu Item Definitions"
4. Press Save Menu
5. Press Update Home Menu.