iiab/iiab #3805: Ubuntu 24.10 removed legacy apt package `wireless-tools`

holta, 2024-09-23:
We should probably reconfirm in coming days, that Ubuntu Oracular Oriole has indeed stuck with its decision to permanently remove the `wireless-tools` apt package.

Meantime, folks should be able to remove it by commenting out Line 35 of...

```
sudo nano /opt/iiab/iiab/roles/network/tasks/install.yml
```

https://github.com/iiab/iiab/blob/master/roles/network/tasks/install.yml#L35

Thank you to @EMG70 !

jvonau, 2024-09-23:
Looks like the library was renamed to [libiw30t64](https://launchpad.net/ubuntu/+source/wireless-tools/30~pre9-17ubuntu2 )

holta, 2024-09-23:
Interesting.

A name like `libiw30t64` is (presumably) not going to be standardized/available on all distros in coming years however.

jvonau, 2024-09-23:
This is not my problem to correct, but you could start with some conditional to exclude the package on U24.10 so people do not have to hand edit files.

holta, 2024-09-23:
Good idea.

Aything along those lines probably makes sense prior to Ubuntu 24.10's release on October 10th.

closed 2024-10-03 by holta