iiab/iiab #3806: Changed name and/or password  of Wi-Fi Access Point has no effect

stenevang, 2024-09-24:
### Steps to reproduce
Go to Admin console > Configure > Internal Wi-Fi AP and change the Internal Wi-Fi Access Point name to something different from default. Also set an Internal Wi-Fi Access Point Password. Verify that the Wi-Fi is visible from another device and verify that you can connect to the Wi-Fi access point of IIAB with the Wi-Fi password set. Next, go back to Admin Console and make the following changes, one at a time:
- Disable password requirement for Wi-Fi
- Rename the Internal Wi-Fi Access Point Name
Save the configuration, and reboot the IIAB system

### Expected
- That the IIAB Wi-Fi should be available without password
- That the new Wi-Fi access point name should be visible in the list of available Wi-Fi networks

### Observed
- Wi-Fi password is still required
- The previous Wi-Fi name is still visible, not the most recent name set
- In summary, the configuration changes are reflected in the Admin Console, but they are not applied in practice

### iiab-diagnostics output:
https://paste.centos.org/view/3eb63d12

holta, 2024-09-24:
Thanks @stenevang for the report!

@tim-moody is this a small bug, do you know?

tim-moody, 2024-09-24:
What surprises me is that the first part works not that the second doesn't.  Saving the config and not running Install Configured Options should have no effect, either turning on wifi password and changing name or turning it off and changing the name.

It is possible that there is an unintended side effect that if changes are made to wifi under Control they propagate to full changes, but you don't mention doing that.

stenevang, 2024-09-25:
Update:

I have finally been able to make it work, and the changes I made compared to before were the following:

1. In Admin Console >Control > WiFi Control, press "Forget all WiFi Connections", because I noticed the system tray wifi icon in the Raspberry PI OS was constantly blinking, as if indicating a problem state
2. In Admin Console > Configure > Network Parameters, change from LAN Controller to Gateway (the Raspberry Pi has a wired connection), and also check the box "Check to use the internal WiFi adapter as two devices"
3. In Admin Console > Configure > Internal Wi-Fi AP, enable Internal Wi-Fi Access Point Password
Then Save Configuration, then Install Configured Options, then Reboot server.

Since before, I also had checked "Outbound - check to Enable traffic from LAN to WAN".

The result of this is that finally, my new Wi-Fi AP name is broadcasted, and I can connect to the IIAB from a remote device.
HOWEVER: Under Network Parameters > Network Discovered at Last Boot, it says; "Gateway: Not Found", and I cannot reach the internet from a remote device via the IIAB WiFi AP (the Raspberry Pi has a wired connection and internet connection is working from the local system).

holta, 2024-09-25:
I noticed you have set `iiab_gateway_enabled: True` in `/etc/iiab/local_vars.yml` which is positive (https://paste.centos.org/view/3eb63d12).

(Towards trying to get client machines to have "passthrough" access to WAN/Internet.)

tim-moody, 2024-09-25:
>also check the box "Check to use the internal WiFi adapter as two devices"

at this point you potentially have 2 gateways, wired and wireless. If the wired connection to the Internet is expected to be permanent I would uncheck using internal WiFi as two devices.

stenevang, 2024-09-25:
Regarding the Gateway not found, I am pretty sure that it is a system level problem. I read that Raspberry Pi users have had similar issues, unrelated to IIAB.