iiab/iiab #3807: The Home Page menu.json file is broken and could not be patched

stenevang, 2024-09-25:
Following admin console config changes to Wi-Fi name, where I pressed Save Config, Install Configured Options and then Rebooted the system, I get a message (in the browser on the local machine) saying:

The Home Page menu.json file is broken and could not be patched
and the home page is blank (background color showing but no menu items)

However, after waiting a minute and reloading the home page URL, the menu looks like before.

tim-moody, 2024-09-25:
>The Home Page menu.json file is broken and could not be patched

There is a bug that issues this message when it is really true that the connection to the server has been lost.

holta, 2024-09-25:
Many Thanks @stenevang !  Closely related issues here:

- #3487
- #3602
- #3781

We'll try to fix this ASAP 🪜

PS Please keep in touch.  It's an honor to receive your thorough & precise reports.  Also please don't hesitate to connect over email by writing holt@unleashkids.org anytime, e.g. if you have private suggestions or questions of any kind!

tim-moody, 2025-03-11:
>There is a bug that issues this message when it is really true that the connection to the server has been lost.

fixed