iiab/iiab #3812: Admin Console download of 5.5GB Ubongo Videos won't start, or is it stuck, and Kiwix catalog cannot be refreshed?

holta, 2024-09-29:
Trying to download the 5.5GB Ubongo Videos (full set) leads to "SCHEDULED" (screenshot further below) but it never starts:

[image]

[image]

Canceling individual jobs ("Cancel Checked Jobs") was attempted and did not work (and really should 😉).

At a deeper level, https://download.kiwix.org/zim/videos/ does not show the above `ubongo_sw_playlist-*` files being available anymore, likely due to Kiwix housekeeping that began August 8th:

- openzim/zim-requests#1132

"Refresh Kiwix Catalog" was attempted (repeatedly) and is not working: 

[image]

It sounds like they're rebooted and tried a lot of different things to get unstuck, without success.

Any suggestions to help them get going again?  (I'm not sure if they have command-line experience, but if necessary I could ask them to delete specific-or-stuck files &/or to try to run things like `sudo iiab-get-kiwix-cat` more forcefully, etc?)

tim-moody, 2024-09-29:
Please verify that Kiwix is installed

tim-moody, 2024-09-30:
Had a chance to look at this some more and I see that indeed Kiwix looks to be installed.

First, for me get kiwix catalog works.

There's no doubt that scheduling a download of a zim that is then yanked out of the catalog and removed from the server could easily cause problems, but looking at the code I don't see why it would not be able to be cancelled, especially if it hadn't started. If there was a reboot that moved it from scheduled to restarted there might be a problem, but I don't see how that happened either.

I also note that the 2024 Ubongo is much larger than 2023 so not sure if that broke anything.

Hard to diagnose without access to the server.

holta, 2024-10-02:
He [person asking the original question] replied:

> Now, I get a 502 Bad Gateway error for most of the content I've downloaded.  Maps work, great books work, Khan Academy works, etc.   Ubongo, most of Wiki (except Medical which works), English Wiki, and most everything else I get 502 Bad Gateway error.  I've tried rebooting several times with no success.  Hmmm.  Any suggestions?

> BTW, I did rerun the re-index KWIX and it was successful.

holta, 2024-10-02:
He was able to log in to the IIAB (RPi) but the iiab-diagnostics appear too large to easily extract.  I'll ask if he can try https://temp.sh which sometimes works, much like a pastebin, but for larger files.

holta, 2024-10-02:
Tangentially here's a kiwix-serve bug fix to watch for soon, in upcoming or nightly releases:

- kiwix/libkiwix#1132

tim-moody, 2024-10-02:
running /opt/iiab/kiwix/bin/kiwix-serve --version might give us some info

ls -lav /library/zims/content/ would tell us what zims

holta, 2024-10-05:
kiwix-serve is unable to start, apparently due to a problem with `/library/zims/library.xml` :

[image]

holta, 2024-10-05:
Does the failure of `ls -lav /library/zims/content` mean something potentially more serious, like a disk issue ??

tim-moody, 2024-10-05:
a lot of this still makes me wonder if kiwix installed completely

holta, 2024-10-05:
> a lot of this still makes me wonder if kiwix installed completely

Yes, as he's apparently trying to add more content to an IIAB that was originally installed in June 2023.