iiab/iiab #3821: 403 Forbidden

mnmed, 2024-10-03:
### Expected behavior and actual behavior:
I had just installed IIAB via the curl command, then tried to access http://box.lan and http://box. Both returned a 403 Forbidden page. Rebooting did not help.

### Steps to reproduce the problem: Stated above

### IIAB version, OS version, hardware details:
IIAB 8.2
Linux Mint 22
i5-3320M 8GB DDR3

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
https://paste.centos.org/view/8016a533

mnmed, 2024-10-04:
Thanks for the help, @holta ! IIAB is now running and I'm adding content to it. I did just run iiab-diagnostics and it seems that the error above is still there.

https://paste.centos.org/view/764367a3

I hope you don't mind knowing more about my context so that you/someone could help me with networking. I'm building offline homeschooling servers for my family so that we have material even if the submarine cables get cut (knock on wood). Here's what I'm trying to set up:
1. ThinkPad as an IIAB server on home LAN - Our home network is composed of 3 desktop PCs and a few laptops. Some are connected via Ethernet but most are on the 2.4ghz network. This ThinkPad server is the one I asked for help with earlier. My assumption is that this server will be in use only if the electrical grid is up, which means my router will still be on, and our devices will be connected to that. I won't need the ThinkPad to function as an access point.
2. Raspberry Pi 4 (or 5, if I manage to buy one) as a traveling IIAB server - This is for a scenario where we'd need to relocate. I imagine this portable server would allow me to serve others in my community as well. I'm thinking of putting it in a Pelican case, power it using power banks, and hopefully pack in a decent solar charger too. I expect that devices will be connecting directly to the Raspberry Pi.

My dream is to have complete K12 resources on these servers, ready to continue educating during difficult times.

For now, I want to be able to use the ThinkPad as an IIAB server on my LAN through the 2.4ghz band. I'm currently able to access http://box.lan locally, but not through the other devices. I use 192.168.1.14 to access IIAB using other devices in the network. Is that how it should work? Using the IP address is totally fine with me but, if the server should be accessible on other devices through http://box.lan, I hope you don't mind helping me with the config. I'm a bit afraid to mess with the networking settings because I know so little about it.

tim-moody, 2024-10-04:
box.lan will only get resolved by other devices if your router does it. for me it is a mixed bag as to when it works and when it doesn't. for example if you have two box.lan machines the router will not know what to do.

if you can access with ip address then the server is working.

jvonau, 2025-01-28:
> Finally, please send screenshot(s) or photo(s) if errors arise, with another iiab-diagnostics at that time, Thanks! I don't know why, but FYI this is where the network phase failed for you: (does this happen every time you run `iiab-network` as root — or is the error below no longer happening?)
> 
> > 2024-10-03 20:19:57,689 p=7953 u=root n=ansible | TASK [network : Restart hostapd when WiFi is present but not when using WiFi as gateway with wifi_up_down False] ***
> > 2024-10-03 20:20:01,675 p=7953 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service hostapd: Job for hostapd.service failed because the control process exited with error code.\nSee "systemctl status hostapd.service" and "journalctl -xeu hostapd.service" for details.\n"}

There is a desktop that comes along with Mint, that is controlling wlan0 behavior and with wifi_up_down False there is no ap0 available, thus hostapd is trying to use wlan0 at the sametime as the desktop.