iiab/iiab #3822: urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='api.github.com', port=443): Max retries exceeded with url: /repos/iiab-share/js-menu-files/contents/menu-defs/

holta, 2024-10-04:
@tim-moody have you seen this before?

> 2024-10-03 23:05:19,524 p=7857 u=root n=ansible | TASK [js-menu : Post process the downloaded menu defs] ********************************************************************************
2024-10-03 23:05:36,767 p=7857 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/opt/admin/cmdsrv/scripts/install_menu_defs.py", "delta": "0:00:15.472978", "end": "2024-10-03 23:05:36.407040", "msg": "non-zero return code", "rc": 1, "start": "2024-10-03 23:05:20.934062", "stderr": "Traceback (most recent call last):\n  File \"/usr/lib/python3/dist-packages/urllib3/connection.py\", line 169, in _new_conn\n    conn = connection.create_connection(\n  File \"/usr/lib/python3/dist-packages/urllib3/util/connection.py\", line 73, in create_connection\n    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):\n  File \"/usr/lib/python3.10/socket.py\", line 955, in getaddrinfo\n    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):\nsocket.gaierror: [Errno -2] Name or service not known\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n  File \"/usr/lib/python3/dist-packages/urllib3/connectionpool.py\", line 700, in urlopen\n    httplib_response = self._make_request(\n  File \"/usr/lib/python3/dist-packages/urllib3/connectionpool.py\", line 383, in _make_request\n    self._validate_conn(conn)\n  File \"/usr/lib/python3/dist-packages/urllib3/connectionpool.py\", line 1017, in _validate_conn\n    conn.connect()\n  File \"/usr/lib/python3/dist-packages/urllib3/connection.py\", line 353, in connect\n    conn = self._new_conn()\n  File \"/usr/lib/python3/dist-packages/urllib3/connection.py\", line 181, in _new_conn\n    raise NewConnectionError(\nurllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x730822143910>: Failed to establish a new connection: [Errno -2] Name or service not known\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n  File \"/usr/lib/python3/dist-packages/requests/adapters.py\", line 439, in send\n    resp = conn.urlopen(\n  File \"/usr/lib/python3/dist-packages/urllib3/connectionpool.py\", line 756, in urlopen\n    retries = retries.increment(\n  File \"/usr/lib/python3/dist-packages/urllib3/util/retry.py\", line 574, in increment\n    raise MaxRetryError(_pool, url, error or ResponseError(cause))\nurllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='api.github.com', port=443): Max retries exceeded with url: /repos/iiab-share/js-menu-files/contents/menu-defs/ (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x730822143910>: Failed to establish a new connection: [Errno -2] Name or service not known'))\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n  File \"/opt/admin/cmdsrv/scripts/install_menu_defs.py\", line 22, in <module>\n    repo_menu_item_defs = adm.get_repo_menu_item_defs() # returns dict\n  File \"/usr/lib/python3/dist-packages/iiab/adm_lib.py\", line 148, in get_repo_menu_item_defs\n    response = requests.get(CONST.menu_def_base_url + 'contents/' + CONST.menu_def_path, headers=headers)\n  File \"/usr/lib/python3/dist-packages/requests/api.py\", line 76, in get\n    return request('get', url, params=params, **kwargs)\n  File \"/usr/lib/python3/dist-packages/requests/api.py\", line 61, in request\n    return session.request(method=method, url=url, **kwargs)\n  File \"/usr/lib/python3/dist-packages/requests/sessions.py\", line 544, in request\n    resp = self.send(prep, **send_kwargs)\n  File \"/usr/lib/python3/dist-packages/requests/sessions.py\", line 657, in send\n    r = adapter.send(request, **kwargs)\n  File \"/usr/lib/python3/dist-packages/requests/adapters.py\", line 516, in send\n    raise ConnectionError(e, request=request)\nrequests.exceptions.ConnectionError: HTTPSConnectionPool(host='api.github.com', port=443): Max retries exceeded with url: /repos/iiab-share/js-menu-files/contents/menu-defs/ (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x730822143910>: Failed to establish a new connection: [Errno -2] Name or service not known'))", "stderr_lines": ["Traceback (most recent call last):", "  File \"/usr/lib/python3/dist-packages/urllib3/connection.py\", line 169, in _new_conn", "    conn = connection.create_connection(", "  File \"/usr/lib/python3/dist-packages/urllib3/util/connection.py\", line 73, in create_connection", "    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):", "  File \"/usr/lib/python3.10/socket.py\", line 955, in getaddrinfo", "    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):", "socket.gaierror: [Errno -2] Name or service not known", "", "During handling of the above exception, another exception occurred:", "", "Traceback (most recent call last):", "  File \"/usr/lib/python3/dist-packages/urllib3/connectionpool.py\", line 700, in urlopen", "    httplib_response = self._make_request(", "  File \"/usr/lib/python3/dist-packages/urllib3/connectionpool.py\", line 383, in _make_request", "    self._validate_conn(conn)", "  File \"/usr/lib/python3/dist-packages/urllib3/connectionpool.py\", line 1017, in _validate_conn", "    conn.connect()", "  File \"/usr/lib/python3/dist-packages/urllib3/connection.py\", line 353, in connect", "    conn = self._new_conn()", "  File \"/usr/lib/python3/dist-packages/urllib3/connection.py\", line 181, in _new_conn", "    raise NewConnectionError(", "urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x730822143910>: Failed to establish a new connection: [Errno -2] Name or service not known", "", "During handling of the above exception, another exception occurred:", "", "Traceback (most recent call last):", "  File \"/usr/lib/python3/dist-packages/requests/adapters.py\", line 439, in send", "    resp = conn.urlopen(", "  File \"/usr/lib/python3/dist-packages/urllib3/connectionpool.py\", line 756, in urlopen", "    retries = retries.increment(", "  File \"/usr/lib/python3/dist-packages/urllib3/util/retry.py\", line 574, in increment", "    raise MaxRetryError(_pool, url, error or ResponseError(cause))", "urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='api.github.com', port=443): Max retries exceeded with url: /repos/iiab-share/js-menu-files/contents/menu-defs/ (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x730822143910>: Failed to establish a new connection: [Errno -2] Name or service not known'))", "", "During handling of the above exception, another exception occurred:", "", "Traceback (most recent call last):", "  File \"/opt/admin/cmdsrv/scripts/install_menu_defs.py\", line 22, in <module>", "    repo_menu_item_defs = adm.get_repo_menu_item_defs() # returns dict", "  File \"/usr/lib/python3/dist-packages/iiab/adm_lib.py\", line 148, in get_repo_menu_item_defs", "    response = requests.get(CONST.menu_def_base_url + 'contents/' + CONST.menu_def_path, headers=headers)", "  File \"/usr/lib/python3/dist-packages/requests/api.py\", line 76, in get", "    return request('get', url, params=params, **kwargs)", "  File \"/usr/lib/python3/dist-packages/requests/api.py\", line 61, in request", "    return session.request(method=method, url=url, **kwargs)", "  File \"/usr/lib/python3/dist-packages/requests/sessions.py\", line 544, in request", "    resp = self.send(prep, **send_kwargs)", "  File \"/usr/lib/python3/dist-packages/requests/sessions.py\", line 657, in send", "    r = adapter.send(request, **kwargs)", "  File \"/usr/lib/python3/dist-packages/requests/adapters.py\", line 516, in send", "    raise ConnectionError(e, request=request)", "requests.exceptions.ConnectionError: HTTPSConnectionPool(host='api.github.com', port=443): Max retries exceeded with url: /repos/iiab-share/js-menu-files/contents/menu-defs/ (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x730822143910>: Failed to establish a new connection: [Errno -2] Name or service not known'))"], "stdout": "", "stdout_lines": []}
2024-10-03 23:05:36,786 p=7857 u=root n=ansible | PLAY RECAP ****************************************************************************************************************************
2024-10-03 23:05:36,791 p=7857 u=root n=ansible | 127.0.0.1                  : ok=67   changed=54   unreachable=0    failed=1    skipped=4    rescued=0    ignored=0

tim-moody, 2024-10-04:
no. is it repeatable or was the service actually unavailable at the time?

holta, 2024-10-04:
Next attempt an hour later, took 24 minutes (for just this one single step, "Post process the downloaded menu defs") and finally completed.

Mysterious.

tim-moody, 2024-10-04:
there is a path through the code that does a lot of computation; sorry don't remember the details. but github could be slow too, especially the api. we should watch for any changes to the api or use of token keys that could be interfering