iiab/iiab #3823: Scale of the Universe fails to install

scarolan, 2024-10-04:
Here's the error.  This is on a fresh installation.  I've tried twice.  Everything else in the 'large' option installed fine.

```
stdout, stderr = process.communicate(input, timeout=timeout)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3.11/subprocess.py", line 1207, in communicate
stdout, stderr = self._communicate(input, endtime, timeout)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3.11/subprocess.py", line 2060, in _communicate
self._check_timeout(endtime, orig_timeout, stdout, stderr)
File "/usr/lib/python3.11/subprocess.py", line 1251, in _check_timeout
raise TimeoutExpired(
subprocess.TimeoutExpired: Command '['iiab-update-menus']' timed out after 55.0 seconds
```

holta, 2024-10-04:
Thank you @scarolan for reporting.

We definitely need to automate this fix soon:

- #3747

scarolan, 2024-10-04:
Oddly it seems to work?  I was able to click on it in the Web UI and zoom in and out.

scarolan, 2025-03-18:
Closing issue.

closed 2025-03-18 by scarolan