iiab/iiab #3832: cant login on wifi, job status says scheduled never starts

Hansolokhalifa, 2024-10-13:
### Expected behavior and actual behavior:Login with http://box.lan/, http://schoolserver.lan/, http://localhost/ or http://10.10.10.10/. Scheduled installs not starting

### Steps to reproduce the problem:login with wifi, install vims from kiwix

### IIAB version, OS version, hardware details:Version:	master
Date Installed:	2024-10-13T08:12:17 Raspberrypi 4

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)Gathers IIAB diagnostics into 1 file, to accelerate troubleshooting.  USAGE:

   iiab-diagnostics
   sudo iiab-diagnostics                                # USE 'sudo' FOR MORE
   sudo iiab-diagnostics PATH/FILE1 PATH/FILE2 ...      # COMPLETE RESULTS !!

Can you provide a short public nickname: (no spaces!) Hannibal

Compiling diagnostics...

  0. HW + SW Quick Summary
     ls -ltr /etc/iiab/install-flags  # IIAB install flags

  1. Files Specially Requested: (from "iiab-diagnostics PATH/FILE1 PATH/FILE2")

  2. Regular Files:

     /.iiab-image
     /etc/iiab/iiab.env
     /etc/iiab/iiab.ini
     /etc/iiab/local_vars.yml
     /etc/iiab/iiab_state.yml
     /etc/resolv.conf
     /etc/network/interfaces
     /etc/hostapd/hostapd.conf
     /etc/wpa_supplicant/wpa_supplicant.conf
sed: can't read /etc/wpa_supplicant/wpa_supplicant.conf: Permission denied
     /library/www/html/home/menu.json

  3. Content of Directories: (1-level deep)

     /etc/network/interfaces.d
     /etc/systemd/network
     /etc/NetworkManager/system-connections
     /etc/netplan

  4. Output of Commands:

     uname -a  # Linux kernel
     free  # RAM memory
     lscpu  # CPU details
     df -h  # Disk usage
     df -ah  # Disk usage detail
df: /run/user/1000/doc: Operation not permitted
     lsblk  # Partition mount points
     blkid  # Mount point details
     /etc/fstab
     lshw -C network  # Network hardware/interfaces
WARNING: you should run this program as super-user.
WARNING: output may be incomplete or inaccurate, you should run this program as super-user.
     ip addr  # Network interfaces
     ifconfig  # Network interfaces (old view)
     ip route  # Routing table
     netstat -rn  # Routing table (old view)
     bridge -d link  # Bridge for LAN side
     sudo netstat -natp  # Ports/Services in use
     systemctl status dnsmasq  # Is dnsmasq running?
     sudo journalctl -b 0 -u dnsmasq  # dnsmasq log
     networkctl  # systemd-networkd status
     nmcli d  # NetworkManager status
     sudo journalctl -b 0 -u networkd-dispatcher  # networkd-dispatcher log
     rfkill list  # Are WiFi and Bluetooth interfaces blocked?
     iw reg get  # Detected WiFi country code / legal frequencies
     iw dev  # List wireless interfaces
     iw list  # List capabilities of all wireless devices
     systemctl status hostapd  # Downstream Wi-Fi: Is hostapd running?
     ls -l /etc/wpa_supplicant  # Upstream Wi-Fi
     ps -AH  # Process hierarchy: staging of hostapd & wpa_supplicant?
     dmesg | grep -i -e 80211 -e 802\.11 -e wireless -e wifi -e wlan -e broadcom -e brcm -e bcm -e realtek | head -100  # Wi-Fi firmware/driver msgs
     lspci -nn  # Devices on PCI buses
     ls -l /lib/firmware/cypress/*43430*  # RPi Zero W & 3 WiFi firmware
     ls -l /lib/firmware/cypress/*43455*  # RPi 3 B+ & 4 WiFi firmware
     env  # Environment variables
     node -v  # Node.js version
     npm -v  # npm version
     /opt/iiab/kiwix/bin/kiwix-serve --version  # kiwix-tools
     cd /usr/local/calibre-web-py3; sudo git log --graph --oneline --decorate | head -50  # Calibre-Web version
     sudo lb --version  # xklb version
     sudo yt-dlp --version  # yt-dlp version
     systemctl status calibre-web  # Is Calibre-Web running?
     journalctl -u calibre-web | tail -100  # Calibre-Web systemd log
     /var/log/calibre-web.log
     /var/log/xklb.log
     journalctl -t IIAB-CMDSRV  # Admin Console CMDSRV log

  5. Firewall Rules:

     sudo iptables-save  # Firewall rules
     sudo ufw status verbose  # Firewall status & rules
sudo: ufw: command not found

  6. Log Files: (e.g. last 100 lines of each)

     grep -B2 "SEE ERROR ABOVE" /opt/iiab/iiab/*.log  # for skip_role_on_error
     /opt/iiab/iiab/iiab-install.log
     /opt/iiab/iiab/iiab-configure.log
     /opt/iiab/iiab/iiab-debug.log
     /opt/iiab/iiab/iiab-network.log
     /opt/iiab/iiab-admin-console/admin-install.log
     /var/log/messages
     /var/log/syslog

holta, 2024-10-13:
Hi @Hansolokhalifa !

1) What student/client devices are you using to browse to Internet-in-a-Box ?

2) Did you also try...

http://192.168.0.44
http://192.168.0.44/admin

3) Do not use HTTPS (remove the 's' !)

Hansolokhalifa, 2024-10-14:
> Hi @Hansolokhalifa !
> 
>     1. What student/client devices are you using to browse to Internet-in-a-Box ?
> 
>     2. Did you also try...
>        http://192.168.0.44
>        http://192.168.0.44/admin
> 
>     3. Do not use HTTPS (remove the 's' !)

using an iphone. I have tried your suggestions and no change. I dont understand what is going on.

holta, 2024-10-14:
First you would need to...

(1) Connect your iPhone to your home's Wi-Fi.  THEN try http://192.168.0.44 (in your iPhone's browser).

Or,

(2) Connect your iPhone to the "Internet in a Box" Wi-Fi signal broadcast by the Raspberry Pi 4.  THEN try http://BOX.LAN and https://10.10.10.10 (in your iPhone's browser).

Does that help?

Hansolokhalifa, 2024-10-14:
connecting to home works when going to the site on wifi, nothing when connecting to internet in a box itself.

holta, 2024-10-14:
Which iPhone and which iOS and which browser?

1. Reboot the Raspberry Pi 4.

2. Wait more than 1 minute.

3. Try connecting **other** student/client devices laptop to the "Internet in a Box" Wi-Fi signal broadcast by the Raspberry Pi 4?

4. THEN try http://BOX.LAN and https://10.10.10.10

5. On some mobile phones, you might have to TURN OFF "Mobile Data" and follow these instructions: https://wiki.iiab.io/go/FAQ#My_Android_device_says_%22Connected,_no_Internet%22_and_won't_browse_http://box

6. Reply with photo(s) or screenshot(s) of your browser, if truly nothing works.

Hansolokhalifa, 2024-10-14:
> Which iPhone and which iOS and which browser?
> 
>     1. Reboot the Raspberry Pi 4.
> 
>     2. Wait more than 1 minute.
> 
>     3. Try connecting **other** student/client devices (e.g. a laptop) to the "Internet in a Box" Wi-Fi signal broadcast by the Raspberry Pi 4?
> 
>     4. THEN try http://BOX.LAN and https://10.10.10.10
> 
>     5. On some mobile phones, you might have to TURN OFF "Mobile Data" and follow these instructions: https://wiki.iiab.io/go/FAQ#My_Android_device_says_%22Connected,_no_Internet%22_and_won't_browse_http://box
> 
>     6. Reply with photo(s) or screenshot(s) of your browser, if truly nothing works.

Iphone 15 plus, unplugged and plugged back in, also have tried turning off mobile data on browser firefox

Hansolokhalifa, 2024-10-14:
> Which iPhone and which iOS and which browser?
>
>     1. Reboot the Raspberry Pi 4.
>
>     2. Wait more than 1 minute.
>
>     3. Try connecting **other** student/client devices (e.g. a laptop) to the [...]

I am able to login with firefox from computers and from android but not iphone. Also having issues logging in as admin and installing more items. login and pass are correct its just not responding at all

holta, 2024-10-14:
> having issues logging in as admin and installing more items. login and pass are correct its just not responding at all

You're not explaining:

1. What exact username
2. What exact URL
3. Screenshot / Photo showing the error

Please understand that you need to be precise if you expect assistance. 😄

tim-moody, 2024-10-14:
>connecting to home works when going to the site on wifi, nothing when connecting to internet in a box itself.

I don't understand the distinction.

>Also having issues logging in as admin and installing more items. login and pass are correct its just not responding at all

The log shows the Admin Console startup sequence and many ZIMs installed successfully, so not clear what the issue is.

Hansolokhalifa, 2024-10-14:
> You're not explaining:
>
>     1. What exact username
>
>     2. What exact URL
>
>     3. Screenshot / Photo showing the error
>
>
> Please understand that you need to be precise if you expect assistance. 😄

Always one of you whether its Discord or Reddit lol if you cant understand then dont answer me my guy, Whats so hard to understand about me not being able to login as admin lol.  I am using iiab-admin and 192.168.0.55/1dmin. Please understand that if you cant understand the question you shouldnt be trying to "help"

holta, 2024-10-14:
@Hansolokhalifa can you send a photo or screenshot of iiab-admin failing to log into http://192.168.0.44/admin showing all on-screen errors?

Hansolokhalifa, 2024-10-14:
> @Hansolokhalifa can you send a photo or screenshot of iiab-admin failing to log into http://192.168.0.44/admin showing all on-screen errors?

[image]
It wont do anything at all. If I restart the server by unplugging, it will let me login but I cannot click anything such as configure, install content and what not. I can access the server by joining from my laptop, and android phones and I can browse the information I have installed for the most part. Should I just reinstall all over?

Hansolokhalifa, 2024-10-15:
> This should either log you in or give a message invalid login. Pressing enter should work, but try clicking Login as well. Seeing all the tabs for Kali Linux makes me wonder what OS you're on and whether your browser has restrictions on Javascript. You could go to the Javascript console and look for errors.

Thanks for the reply. I am using Ubuntu now, uninstalled Kali for other reasons. I click login and just nothing happens at all. I will do as you instructed and report back.

Hansolokhalifa, 2024-10-16:
decided to just reinstall. should I be accessing iiab through ssh from my home network to install content or should I connect to iiab wifi and install content that way?

tim-moody, 2024-10-16:
I recommend installing content using the Admin Console. How you connect to IIAB doesn't matter, but IIAB must have a connection to the Internet in order to download content.

Hansolokhalifa, 2024-10-16:
> I recommend installing content using the Admin Console. How you connect to IIAB doesn't matter, but IIAB must have a connection to the Internet in order to download content.

I got it up and running. Is there a way to install vims directly to usb, my micro sd is only 256gb and I want to install the survivor vim to a 512gb usb I have.