iiab/iiab #3833: Question

eriad25, 2024-10-16:
I would like to know what software do you use to host websites with content avaliable on IIAB? From what i see IIAB is essentialy multi purpouse server with everything you'll ever need so what is it that IIAB uses to host avaliable services if it is not a secret because i'd like try and make IIAB by my self for personal use with sites of my choice.

holta, 2024-10-16:
Hi @eriad25,

IIAB's code is open source if you want to examine it, learn from it, modify it and/or contribute to our global volunteer community.

You might start by looking over IIAB's ~30 Apps here?

https://github.com/iiab/iiab/tree/master/roles

And in addition, you might also want to look over our Technical Contributors Guide here:

https://github.com/iiab/iiab/wiki/Technical-Contributors-Guide

eriad25, 2024-10-16:
I checked everything as you said and from what i see underneath everything is is webserver to host all web services and apps coming with it. I do have an idea how to make something identical to IIAB that will actualy have stuff specific for me. So thanks for response, you can close issue now.

closed 2024-10-16 by holta