iiab/iiab #3836: Could not open input file: /library/www/nextcloud/occ

Nemoron, 2024-10-26:
### Expected behavior and actual behavior:
Nextcloud gets installed.
### Steps to reproduce the problem:
unknown
### IIAB version, OS version, hardware details:
IIAB 8.2 
Raspberry Pi OS Lite
Release date: July 4th 2024
System: 64-bit
Raspberry Pi 5 8Gb
### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: https://paste.centos.org/view/0b471efe

holta, 2024-10-26:
@Nemoron thanks for reporting!

We'll try to reproduce this in coming days & offer a patch as necessary.

holta, 2024-10-27:
Hi @Nemoron,

- Where exactly are you seeing the error?

- Can you please paste in a screenshot (or photo) and explain?

(FWIW Nextcloud 30.0.1 installs cleanly with the latest IIAB on Ubuntu 24.04.1, so it's not clear how to reproduce what you mention.)

holta, 2024-10-27:
Ok, at least I know see Lines 1852-1853 of your iiab-diagnostics here: (and likewise on Lines 1734-1752!)

> 2024-10-26 17:04:48,375 p=10613 u=root n=ansible | TASK [nextcloud : Allow Nextcloud access from all hosts and IP addresses -- SEE ALSO THE OLD WAY: https://github.com/iiab/iiab/blob/master/roles/nextcloud/templates/nextcloud.conf.j2.unused] ***
2024-10-26 17:04:48,627 p=10613 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["php", "/library/www/nextcloud/occ", "config:system:set", "trusted_domains", "1", "--value=*"], "delta": "0:00:00.041901", "end": "2024-10-26 17:04:48.596681", "msg": "non-zero return code", "rc": 1, "start": "2024-10-26 17:04:48.554780", "stderr": "", "stderr_lines": [], "stdout": "Could not open input file: /library/www/nextcloud/occ", "stdout_lines": ["Could not open input file: /library/www/nextcloud/occ"]}

So if I or someone can get access to a Raspberry Pi 5 in the coming week, we can try to reproduce there.

QUESTION for @Nemoron: Have you tried another clean/fresh/new install of IIAB to see if this is fully reproducible for you?

Nemoron, 2024-10-27:
Yes without activating phpmyadmin and without installing firmware for the cooling and hat, everything worked fine. When researching the issue, I found a post with a similar problem where one comment stated, that the permissions look weird. When I checked the owner of the next cloud folder and everything inside it was correct, but the owner did not have the rights to execute anything. I don't know if that is normal. ^^ 
But when I gave the owner executive permissions, the error persisted.

Nemoron, 2024-10-27:
> Hi @Nemoron,
> 
> - Where exactly are you seeing the error?
> 
> - Can you please paste in a screenshot (or photo) and explain?
> 
> (FWIW Nextcloud 30.0.1 installs cleanly with the latest IIAB on Ubuntu 24.04.1, so it's not clear how to reproduce what you mention.)

I saw the error when trying to install iiab with the one line installer. I also tried restarting the script and rebooting the pi, but the error persisted.

Nemoron, 2024-10-27:
[image]
Just checked the permissions in my new install. The executive permissions don't seem to be the problem.

Nemoron, 2024-11-09:
Tried something again: 
Neither Phpmyadmin nor the firmware is the cause.
Some things I have changed in the local_vars.yml
-host_country_code to de
-rpi3bplus_rpi4_wifi_firmware: os
-squid_install: True
-lokole_install: True

holta, 2024-11-09:
> -lokole_install: True

I don't think that will work (to be avoided!)

Reason: Lokole is not being maintained in recent years (the original developer moved on).

Nemoron, 2024-11-09:
> > lokole_install: True
> 
> I don't think that will work (to be avoided!)
> 
> Reason: Lokole is not being maintained in recent years (the original developer moved on).

That should be added to the local_vars.yml!

closed 2024-11-09 by Nemoron