iiab/iiab #3839: downloading zim files but they arent showing up on internal storage

Hansolokhalifa, 2024-10-28:
### Expected behavior and actual behavior: Download zim, it downkloads to internal storage

### Steps to reproduce the problem:Download any ted talk, doesnt show up

### IIAB version, OS version, hardware details: version says master

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!) cant run this, password is not being accepted

tim-moody, 2024-10-29:
I tried to duplicate this. I added a TED zim and a wiki ZIM. Both because scheduled, but never started.

systemctl status kiwix-serve shows kiwix not running. I restarted it but it failed.

tim-moody, 2024-10-29:
This is on bare metal Debian 12.

tim-moody, 2024-10-29:
running iiab-install --reinstall

tim-moody, 2024-10-29:
```
root@iiab-ref:/opt/iiab/iiab# systemctl status kiwix-serve
× kiwix-serve.service - Provides the Kiwix ZIM File Server
     Loaded: loaded (/etc/systemd/system/kiwix-serve.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Tue 2024-10-29 17:08:28 EDT; 1min 38s ago
   Duration: 5ms
   Main PID: 101052 (code=exited, status=1/FAILURE)
        CPU: 60ms

Oct 29 17:08:28 iiab-ref systemd[1]: Starting kiwix-serve.service - Provides the Kiwix ZIM File Server...
Oct 29 17:08:28 iiab-ref kiwix-serve[101051]: Loading the library from the following files:
Oct 29 17:08:28 iiab-ref kiwix-serve[101051]:         /library/zims/library.xml
Oct 29 17:08:28 iiab-ref kiwix-serve[101051]: The library was successfully loaded.
Oct 29 17:08:28 iiab-ref systemd[1]: Started kiwix-serve.service - Provides the Kiwix ZIM File Server.
Oct 29 17:08:28 iiab-ref kiwix-serve[101052]: Unable to instantiate the HTTP daemon. The port 3000 is maybe already occupied or need >
Oct 29 17:08:28 iiab-ref systemd[1]: kiwix-serve.service: Main process exited, code=exited, status=1/FAILURE
Oct 29 17:08:28 iiab-ref systemd[1]: kiwix-serve.service: Failed with result 'exit-code'.

```

holta, 2024-10-29:
FWIW Lines 1390 and 1992 of https://paste.centos.org/view/369239a9 show kiwix-serve.service apparently running ok: (on @Hansolokhalifa's Raspberry Pi 4)

```
COMMAND: /usr/bin/sudo netstat -natp    # Ports/Services in use
...
tcp        0      0 0.0.0.0:3000            0.0.0.0:*               LISTEN      933/kiwix-serve    `
```

```
COMMAND: /usr/bin/ps -AH    # Process hierarchy: staging of hostapd & wpa_supplicant?
...
    933 ?        00:00:00   kiwix-serve
```

Hansolokhalifa, 2024-10-30:
> I rebooted and kiwix started and the zim download jobs completed.
> 
> Both zims were automatically added to the home page and are accessible, but the browser cache must be refreshed.

Thank you I will restart and see what I can do.  Thank you.

Hansolokhalifa, 2024-10-30:
> FWIW Lines 1390 and 1992 of https://paste.centos.org/view/369239a9 show kiwix-serve.service apparently running ok: (on @Hansolokhalifa's Raspberry Pi 4)
>
> ```
> COMMAND: /usr/bin/sudo netstat -natp [...]

I have no idea why it wont add zim to my internal storage. Downloaded all of Ted talks 3 times already and they never showed up.

tim-moody, 2024-10-30:
>Downloaded all of Ted talks 3 times already

by what means and to what location?

>it wont add zim to my internal storage

if you downloaded it where did it go?