iiab/iiab #3841: still unable to login to IIAB admin console

Hansolokhalifa, 2024-11-05:
### Expected behavior and actual behavior: Tried to login but never can over wifi with Arch Linux and Chrome

### Steps to reproduce the problem: go to web address, try to login, no action after clocking login

### IIAB version, OS version, hardware details: Raspbian 8.2, Raspberry pi 4 

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!) Unable to read from: /etc/iiab/diag/8.2_raspbian-12_2024-11-05_06:49:59_PST_iiab
error

tim-moody, 2024-11-05:
Please have a look at https://wiki.iiab.io/go/FAQ#What_OS_should_I_use?

Arch Linux is not supported.

Hansolokhalifa, 2024-11-05:
> Please have a look at https://wiki.iiab.io/go/FAQ#What_OS_should_I_use?
> 
> Arch Linux is not supported.

misunderstanding. I am trying to access from arch linux not install IIAB to it. I am trying a fresh install to Raspbian again, the latest update.

tim-moody, 2024-11-05:
sorry, I see that now.

I can't guess what the problem is as there is very little to go on.

Hansolokhalifa, 2024-11-05:
> sorry, I see that now.
> 
> I can't guess what the problem is as there is very little to go on.

trying to reinstall I got this error. I am.....almost over IIAB and looking for alternatives. 

ASK [moodle : Execute /opt/iiab/moodle/moodle_installer IF /opt/iiab/moodle/config.php doesn't yet exist -- REQUIRES 'max_input_vars = 5000' (or higher) in /etc/php/8.2/cli/php.ini IF PHP 8+ (as set up by www_options/tasks/main.yml) -- WHEREAS LATER ON Moodle uses /etc/php/8.2/fpm/php.ini during regular operation] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/opt/iiab/moodle/moodle_installer", "delta": "0:00:00.745752", "end": "2024-11-05 19:04:35.254817", "msg": "non-zero return code", "rc": 1, "start": "2024-11-05 19:04:34.509065", "stderr": "+ sudo -u www-data /usr/bin/php /opt/iiab/moodle/admin/cli/install.php --wwwroot=http://box.lan/moodle --dataroot=/library/moodle --dbtype=pgsql --dbname=moodle --dbuser=Admin --dbpass=changeme --fullname=Your_School --shortname=School --adminuser=admin --adminpass=changeme --non-interactive --agree-license --allow-unstable\nPHP Warning:  Invalid \"upload_max_filesize\" setting. Invalid quantity \"10000M\": value is out of range, using overflow result for backwards compatibility in Unknown on line 0\nPHP Warning:  Invalid \"post_max_size\" setting. Invalid quantity \"10000M\": value is out of range, using overflow result for backwards compatibility in Unknown on line 0\nPHP Deprecated:  Using ${var} in strings is deprecated, use {$var} instead in /opt/iiab/moodle/lib/phpminimumversionlib.php on line 56\nPHP Deprecated:  Using ${var} in strings is deprecated, use {$var} instead in /opt/iiab/moodle/lib/phpminimumversionlib.php on line 56", "stderr_lines": ["+ sudo -u www-data /usr/bin/php /opt/iiab/moodle/admin/cli/install.php --wwwroot=http://box.lan/moodle --dataroot=/library/moodle --dbtype=pgsql --dbname=moodle --dbuser=Admin --dbpass=changeme --fullname=Your_School --shortname=School --adminuser=admin --adminpass=changeme --non-interactive --agree-license --allow-unstable", "PHP Warning:  Invalid \"upload_max_filesize\" setting. Invalid quantity \"10000M\": value is out of range, using overflow result for backwards compatibility in Unknown on line 0", "PHP Warning:  Invalid \"post_max_size\" setting. Invalid quantity \"10000M\": value is out of range, using overflow result for backwards compatibility in Unknown on line 0", "PHP Deprecated:  Using ${var} in strings is deprecated, use {$var} instead in /opt/iiab/moodle/lib/phpminimumversionlib.php on line 56", "PHP Deprecated:  Using ${var} in strings is deprecated, use {$var} instead in /opt/iiab/moodle/lib/phpminimumversionlib.php on line 56"], "stdout": "\nDeprecated: Using ${var} in strings is deprecated, use {$var} instead in /opt/iiab/moodle/lib/phpminimumversionlib.php on line 56\n\nDeprecated: Using ${var} in strings is deprecated, use {$var} instead in /opt/iiab/moodle/lib/phpminimumversionlib.php on line 56\n                                 .-..-.       \n   _____                         | || |       \n  /____/-.---_  .---.  .---.  .-.| || | .---. \n  | |  _   _  |/  _  \\/  _  \\/  _  || |/  __ \\\n  * | | | | | || |_| || |_| || |_| || || |___/\n    |_| |_| |_|\\_____/\\_____/\\_____||_|\\_____)\n\nMoodle 4.1.14+ (Build: 20241101) command line installation program\n== Environment ==\n!! php !!\n[System] version 7.4.0 is required and you are running 8.2.7 - PHP version 8.2 and higher are not supported.", "stdout_lines": ["", "Deprecated: Using ${var} in strings is deprecated, use {$var} instead in /opt/iiab/moodle/lib/phpminimumversionlib.php on line 56", "", "Deprecated: Using ${var} in strings is deprecated, use {$var} instead in /opt/iiab/moodle/lib/phpminimumversionlib.php on line 56", "                                 .-..-.       ", "   _____                         | || |       ", "  /____/-.---_  .---.  .---.  .-.| || | .---. ", "  | |  _   _  |/  _  \\/  _  \\/  _  || |/  __ \\", "  * | | | | | || |_| || |_| || |_| || || |___/", "    |_| |_| |_|\\_____/\\_____/\\_____||_|\\_____)", "", "Moodle 4.1.14+ (Build: 20241101) command line installation program", "== Environment ==", "!! php !!", "[System] version 7.4.0 is required and you are running 8.2.7 - PHP version 8.2 and higher are not supported."]}

TASK [moodle : SEE ERROR ABOVE (skip_role_on_error: False)] *******************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": ""}

PLAY RECAP ********************************************************************************************************************
127.0.0.1                  : ok=85   changed=17   unreachable=0    failed=1    skipped=24   rescued=1    ignored=0

tim-moody, 2024-11-09:
I'm sorry I still have no answer for you. I recommend that you start with only the services you need and then possibly add more later. If you don't need moodle, then leave it for now. In terms of accessing box.lan/admin, the only thing that I can come up with is lack of javascript support in the browser. Maybe it is turned off or you have ublock origin or something else.

holta, 2024-11-09:
@Hansolokhalifa would you be able to re-run `sudo iiab-diagnostics` and post the full RESULTING URL?

(Or if you cannot generate a working URL, can you perhaps post the actual ~2000 line diagnostic file from within /etc/iiab/diag/ to GitHub here?)

Hansolokhalifa, 2024-11-10:
> @Hansolokhalifa would you be able to re-run `sudo iiab-diagnostics` and post the full RESULTING URL?
> 
> (Or if you cannot generate a working URL, can you perhaps post the actual ~2000 line diagnostic file from within /etc/iiab/diag/ to GitHub here?)

I am able to login now, but a persistent issue is me being able to download all of the TEd talks  videos. Theh job status says complete, it takes about 44 hours to complete each time yet they are not on my internal storage or my usb external storage. I have literally downloaded them 5 times now and I am losing my patience with this project.

holta, 2024-11-10:
Thanks @Hansolokhalifa.  Tim can answer you in more detail, but just a small point that might help you regarding .zim file downloading slowly over HTTPS from Kiwix.org mirrors (having nothing to do with Internet-in-a-Box!)  You are the 2nd person in recent weeks to report this very serious issue.

So, until that's resolved by Kiwix.org, just FYI a very pragmatic workaround it to torrent individual ZIM files that you need, to immediately solve the downloading slowness problem, if you want to go this (manual!) route, e.g. if you're comfortable at the Linux command-line example:

1. As shown in the following example, you would just need to add `.torrent` to the right side of any original .zim file's URL:

   - https://download.kiwix.org/zim/ted/ted_mul_education_2024-10.zim
   - https://download.kiwix.org/zim/ted/ted_mul_education_2024-10.zim.mirrorlist
   - https://download.kiwix.org/zim/ted/ted_mul_education_2024-10.zim.torrent

2. Then, after downloading the file to your IIAB, you would place it is in folder: `/library/zims/content`

3. Then, you would run: `sudo iiab-make-kiwix-lib`

4. Then you would clean up any needed Content Pack description(s) on your IIAB main page:
   - Log into IIAB Admin Console
   - Click on "Content Menus" tab.
   - Either click on "Content Item List" to rearrange the order of Content Pack descriptions.
   - Or click on "Edit Menu Items" (left side) to refine the wording of any Content Pack descriptions.

Hansolokhalifa, 2024-11-10:
> Thanks @Hansolokhalifa. Tim can answer you in more detail, but just a small point that might help you regarding .zim files downloading very slowly over HTTPS from Kiwix.org mirrors (having nothing to do with Internet-in-a-Box!) You are the 2nd person in recent weeks to report this very serious issue.
> 
> So, until that's resolved by Kiwix.org, just FYI a very pragmatic workaround it to torrent individual ZIM files that you need, to immediately solve the downloading slowness problem, if you want to go this (manual!) route, e.g. if you're comfortable at the Linux command-line example:
> 
>     1. As shown in the following example, you would just need to add `.torrent` to the right side of any original .zim file's URL:
>        
>        * https://download.kiwix.org/zim/ted/ted_mul_education_2024-10.zim
>        * https://download.kiwix.org/zim/ted/ted_mul_education_2024-10.zim.mirrorlist
>        * https://download.kiwix.org/zim/ted/ted_mul_education_2024-10.zim.torrent
> 
>     2. Then, after downloading the file to your IIAB, you would place it is in folder: `/library/zims/content`
> 
>     3. Then, you would run: `sudo iiab-make-kiwix-lib`
> 
>     4. Then you would clean up any needed Content Pack description(s) on your IIAB main page:
>        
>        * Log into IIAB Admin Console
>        * Click on "Content Menus" tab.
>        * Either click on "Content Item List" to rearrange the order of Content Pack descriptions.
>        * Or click on "Edit Menu Items" (left side) to refine the wording of any Content Pack descriptions.

Ok I will try it

Hansolokhalifa, 2024-11-10:
> This is a repeat of #3839
> 
> As I said there, I encountered this problem when Kiwix was not running and had to do a reinstall of Kiwix after which TED zims downloaded.
> 
> > I  get this about the ted talks videos Finished making Kiwix library.xml
> > Updating home page menues
> > Waiting to run iiab_update_menus.py
> > Now running iiab_update_menus.py
> > Updating Home Page Menu
> > Updating kiwix menu items
> > Adding en-ted_mul_all to Menu
> > Updating iiab installed services' menu items
> > Updating installed OER2Go menu items
> > Finished running iiab_update_menus.py
> 
> Looks right. Do you refresh your browser's cache after all this? The home menu in the cache will not contain new content. Do a full refresh so all json files are also refreshed.

Will try that now thank you

Hansolokhalifa, 2024-11-10:
> This is a repeat of #3839
>
> As I said there, I encountered this problem when Kiwix was not running and had to do a reinstall of Kiwix after which TED zims downloaded.
>
>
> Looks right. Do you refresh [...]

refreshed and no change. I have a 256gb micro sd installed and it says that 116gb is being used but I donot know for what since there are no zim files installed.

tim-moody, 2024-11-10:
is the sdcard a brand name? there are cards that report one amount of storage but actually have less.

see https://www.grc.com/validrive.htm

holta, 2024-11-10:
@Hansolokhalifa can you run the following and send us the output?

```
sudo ls -lSh /library/zims/content/
```

(And better yet, send the full output below if you can!)

```
sudo apt install tree
sudo tree --du /library/zims
```

Hansolokhalifa, 2024-11-10:
> @Hansolokhalifa can you run the following and send us the output?
> 
> ```
> sudo ls -lSh /library/zims/content/
> ```
> 
> (And better yet, send the full output below if you can!)
> 
> ```
> sudo apt install tree
> sudo tree --du /library/zims
> ```

for the first command I get this total 
74G
74G -rw-r--r-- 1 root root 74G Jul 18 09:09 ted_mul_all_2024-07.zim
56K -rw-r--r-- 1 root root 53K Nov  7 14:39 test.zim

second command I get this 
78866161517]  /library/zims
├── [78866081224]  content
│   ├── [78866023771]  ted_mul_all_2024-07.zim
│   └── [      53357]  test.zim
├── [       4096]  index
└── [      72101]  library.xml

 157732234549 bytes used in 3 directories, 3 files
The brand of sd is Sandisk

holta, 2024-11-10:
> I have a 256gb micro sd installed and it says that 116gb is being used

@Hansolokhalifa where exactly are you seeing 116GB?  (That's probably a misunderstanding, that we can better clarify I hope, please paste in a screenshot or photo if possible!!)

FYI your disk [i.e. your microSD card] looks to be in good shape: (at least according to https://paste.centos.org/view/09ac5bd7 here!)

```
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2  235G   87G  144G  38% /      ZIMs: 2 OER2Go: 1 Apps2B: 0
```